osrf/homebrew-simulation #36: Failing to build gazebo 5.0.1 on 10.10.2

francesco-romano, 2015-02-05:
~~~
[ 57%] Built target gazebo_physics_simbody
/usr/local/Cellar/cmake/3.1.1_1/bin/cmake -E cmake_progress_report /tmp/gazebo5-cSqea9/gazebo-5.0.1/build/CMakeFiles 
[ 57%] Building CXX object gazebo/physics/ode/CMakeFiles/gazebo_physics_ode.dir/ODEUniversalJoint.cc.o
cd /tmp/gazebo5-cSqea9/gazebo-5.0.1/build/gazebo/physics/ode && /usr/local/Library/ENV/4.3/clang++   -DLIBBULLET_VERSION=2.82 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_ode_EXPORTS -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -std=c++11 -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -fPIC -I/tmp/gazebo5-cSqea9/gazebo-5.0.1/test/gtest/include -I/tmp/gazebo5-cSqea9/gazebo-5.0.1 -I/tmp/gazebo5-cSqea9/gazebo-5.0.1/build -isystem /tmp/gazebo5-cSqea9/gazebo-5.0.1/deps/opende/include -isystem /usr/local/include -iframework /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Paging -isystem /usr/local/Cellar/sdformat/2.2.1/include/sdformat-2.2    -fPIC -o CMakeFiles/gazebo_physics_ode.dir/ODEUniversalJoint.cc.o -c /tmp/gazebo5-cSqea9/gazebo-5.0.1/gazebo/physics/ode/ODEUniversalJoint.cc
Linking CXX shared library libgazebo_physics_ode.dylib
cd /tmp/gazebo5-cSqea9/gazebo-5.0.1/build/gazebo/physics/ode && /usr/local/Cellar/cmake/3.1.1_1/bin/cmake -E cmake_link_script CMakeFiles/gazebo_physics_ode.dir/link.txt --verbose=1
/usr/local/Library/ENV/4.3/clang++    -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -std=c++11 -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -dynamiclib -Wl,-headerpad_max_install_names   -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 5.0.0 -current_version 5.0.1 -o libgazebo_physics_ode.5.0.1.dylib -install_name /tmp/gazebo5-cSqea9/gazebo-5.0.1/build/gazebo/physics/ode/libgazebo_physics_ode.5.dylib CMakeFiles/gazebo_physics_ode.dir/ODEBallJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODECollision.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEGearboxJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEHeightmapShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEHinge2Joint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEHingeJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODELink.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEMesh.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEMeshShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEMultiRayShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEPhysics.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEPolylineShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODERayShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEScrewJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODESliderJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODESurfaceParams.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEUniversalJoint.cc.o  -L/tmp/gazebo5-cSqea9/gazebo-5.0.1/build/test  -L/usr/local/Cellar/ogre/1.7.4/lib  -L/usr/local/Cellar/bullet/2.82/lib ../../../deps/opende/libgazebo_ode.5.0.1.dylib ../../../deps/opende/OPCODE/libgazebo_opcode.5.0.1.dylib ../../common/libgazebo_common.5.0.1.dylib /usr/local/lib/libtbb.dylib ../../../deps/opende/GIMPACT/libgazebo_gimpact.5.0.1.dylib ../../../deps/opende/ou/libgazebo_opende_ou.5.0.1.dylib ../../../deps/libccd/libgazebo_ccd.5.0.1.dylib -lm ../../math/libgazebo_math.5.0.1.dylib /usr/lib/libdl.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libfreeimage.dylib -ltinyxml -lcurl /usr/local/lib/libtar.dylib /usr/local/lib/libtbb.dylib /usr/local/Cellar/sdformat/2.2.1/lib/libsdformat.dylib -lpthread -framework OpenAL -Wl,-rpath,/tmp/gazebo5-cSqea9/gazebo-5.0.1/build/test -Wl,-rpath,/usr/local/Cellar/ogre/1.7.4/lib -Wl,-rpath,/usr/local/Cellar/bullet/2.82/lib 
cd /tmp/gazebo5-cSqea9/gazebo-5.0.1/build/gazebo/physics/ode && /usr/local/Cellar/cmake/3.1.1_1/bin/cmake -E cmake_symlink_library libgazebo_physics_ode.5.0.1.dylib libgazebo_physics_ode.5.dylib libgazebo_physics_ode.dylib
/usr/local/Cellar/cmake/3.1.1_1/bin/cmake -E cmake_progress_report /tmp/gazebo5-cSqea9/gazebo-5.0.1/build/CMakeFiles  74 75
[ 57%] Built target gazebo_physics_ode
make: *** [all] Error 2
==> Formula
Tap: osrf/homebrew-simulation
Path: /usr/local/Library/Taps/osrf/homebrew-simulation/gazebo5.rb
==> Configuration
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew
HEAD: 32a4621930ec0a4952c548c17d02bdfa57f57967
Last commit: 6 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
CPU: quad-core 64-bit haswell
OS X: 10.10.2-x86_64
Xcode: 6.1.1
CLT: N/A
Clang: 6.0 build 600
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.8_1/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby
Java: 1.6.0_65
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j4
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/libxml2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/apache2:/opt/X11/include:/opt/X11/include/freetype2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /opt/X11/lib:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/qt/Frameworks:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/libzzip/lib/pkgconfig:/usr/local/opt/ogre/lib/pkgconfig:/usr/local/opt/protobuf/lib/pkgconfig:/usr/local/opt/protobuf-c/lib/pkgconfig:/usr/local/opt/qt/lib/pkgconfig:/usr/local/opt/console_bridge/lib/pkgconfig:/usr/local/opt/urdfdom_headers/lib/pkgconfig:/usr/local/opt/urdfdom/lib/pkgconfig:/usr/local/opt/tinyxml/lib/pkgconfig:/usr/local/opt/sdformat/lib/pkgconfig:/usr/local/opt/bullet/lib/pkgconfig:/usr/local/opt/simbody/lib/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Library/ENV/pkgconfig/10.10:/opt/X11/lib/pkgconfig:/opt/X11/share/pkgconfig
HOMEBREW_SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk
ACLOCAL_PATH: /usr/local/share/aclocal:/opt/X11/share/aclocal
PATH: /usr/local/Library/ENV/4.3:/usr/local/opt/cmake/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/doxygen/bin:/usr/local/opt/libtar/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/libzzip/bin:/usr/local/opt/ogre/bin:/usr/local/opt/protobuf/bin:/usr/local/opt/protobuf-c/bin:/usr/local/opt/qt/bin:/usr/local/opt/urdfdom/bin:/opt/X11/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: gazebo5 5.0.1 did not build
~~~

Gazebo 5.0.0 successfully built.

scpeters, 2015-02-05:
Can you paste the entire output with `--verbose`? This snippet didn't actually include the compilation error. Thanks!

francesco-romano, 2015-02-05:
The problem is that the snippet is obtained with --verbose. 
Also log files do no show the real problem

scpeters, 2015-02-05:
Can you paste the entire output then to https://gist.github.com ? Your snippet only started at 57%, and I think the error happened before that. I usually search build logs for `***` to find error locations.

francesco-romano, 2015-02-06:
This is the gist
https://gist.github.com/francesco-romano/afe92351c64d06c73ff2

scpeters, 2015-02-06:
I see a linking error related to dartsim:
~~~
make[2]: *** No rule to make target `/usr/X11R6/lib/libglut.dylib', needed by `gazebo/physics/dart/libgazebo_physics_dart.5.0.1.dylib'.  Stop.
~~~
When you installed the dart simulator, did you install it with `brew install dartsim --core-only`?

francesco-romano, 2015-02-06:
I'm very sorry! I did not see that point. And because everything worked when I installed `gazebo5` I had not thought of dependencies problems. 
I compiled again `dartsim` with that option and everything worked fine.

Thanks

closed 2015-02-06 by francesco-romano

scpeters, 2015-02-06:
It's not a problem! Thanks for the report. I initially recommended installing with `--core-only` because it installs fewer dependencies. I didn't know that the build would actually fail. I'll report this upstream to the dartsim developers.