osrf/homebrew-simulation #37: ogre-1.9 fails to build on Yosemite 10.10.3

fferri, 2015-04-17:
Brand new machine. I started installing ROS Indigo according to https://gist.github.com/mikepurvis/9837958.

Stopping at ogre, with this error:

$ HOMEBREW_MAKE_JOBS=1 VERBOSE=1 brew install https://github.com/osrf/homebrew-simulation/raw/ogre_1_9/ogre-1.9.rb
[...]
[ 45%] Building CXX object OgreMain/CMakeFiles/OgreMain.dir/src/OSX/OgreConfigDialog.mm.o
cd /tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/build/OgreMain && /usr/local/Library/ENV/4.3/clang++   -DBOOST_ALL_NO_LIB -DBOOST_THREAD_VERSION=4 -DFREEIMAGE_LIB -DOGRE_NONCLIENT_BUILD -DOgreMain_EXPORTS -D_MT -D_USRDLL -msse -Wall -Winit-self -Wno-overloaded-virtual -Wcast-qual -Wwrite-strings -Wextra -Wno-unused-parameter -Wshadow -Wno-missing-field-initializers -Wno-long-long -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -fPIC -I/tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/include -I/tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/build/include -I/tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/include/OSX -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include/freetype2 -I/tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667 -I/tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/include/Threading -I/tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/src/nedmalloc    -pthread -DOGRE_GCC_VISIBILITY -fvisibility=hidden -o CMakeFiles/OgreMain.dir/src/OSX/OgreConfigDialog.mm.o -c /tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/src/OSX/OgreConfigDialog.mm
In file included from /tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/src/OSX/OgreConfigDialog.mm:30:
In file included from /tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/include/OgreConfigDialog.h:50:
In file included from /tmp/ogre-1.920150417-67612-ltm5kl/sinbad-ogre-dd30349ea667/OgreMain/include/OSX/OgreConfigDialogImp.h:38:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/Cocoa.framework/Headers/Cocoa.h:12:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/Foundation.framework/Headers/Foundation.h:80:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/Foundation.framework/Headers/NSURLError.h:12:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/CoreServices.framework/Headers/CoreServices.h:55:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/FSEvents.framework/Headers/FSEvents.h:486:40: error: missing ',' between enumerators
  kFSEventStreamEventFlagItemIsHardlink __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_9_0) = 0x00100000,
                                       ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/FSEvents.framework/Headers/FSEvents.h:491:44: error: missing ',' between enumerators
  kFSEventStreamEventFlagItemIsLastHardlink __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_9_0) = 0x00200000,
                                           ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/FSEvents.framework/Headers/FSEvents.h:491:45: error: redefinition of enumerator '__AVAILABILITY_INTERNAL__IPHONE_9_0'
  kFSEventStreamEventFlagItemIsLastHardlink __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_9_0) = 0x00200000,
                                            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/Availability.h:156:50: note: expanded from macro '__OSX_AVAILABLE_STARTING'
    #define __OSX_AVAILABLE_STARTING(_osx, _ios) __AVAILABILITY_INTERNAL##_ios
                                                 ^
<scratch space>:86:1: note: expanded from here
__AVAILABILITY_INTERNAL__IPHONE_9_0
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/FSEvents.framework/Headers/FSEvents.h:486:41: note: previous definition is here
  kFSEventStreamEventFlagItemIsHardlink __OSX_AVAILABLE_STARTING(__MAC_10_10, __IPHONE_9_0) = 0x00100000,
                                        ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/Availability.h:156:50: note: expanded from macro '__OSX_AVAILABLE_STARTING'
    #define __OSX_AVAILABLE_STARTING(_osx, _ios) __AVAILABILITY_INTERNAL##_ios
                                                 ^
<scratch space>:85:1: note: expanded from here
__AVAILABILITY_INTERNAL__IPHONE_9_0
^
3 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OSX/OgreConfigDialog.mm.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2
==> Formula
Path: /Library/Caches/Homebrew/Formula/ogre-1.9.rb
==> Configuration
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew
HEAD: a2c8949bb00680b52241b8208ff51eacd127be34
Last commit: 4 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
CPU: quad-core 64-bit dunno
OS X: 10.10.3-x86_64
Xcode: 6.3
CLT: N/A
Clang: 6.1 build 602
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby
Java: N/A
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j1
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/libxml2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/usr/include/apache2:/opt/X11/include:/opt/X11/include/freetype2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /opt/X11/lib:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/libzzip/lib/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Library/ENV/pkgconfig/10.10:/opt/X11/lib/pkgconfig:/opt/X11/share/pkgconfig
HOMEBREW_MAKE_JOBS: 1
HOMEBREW_SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk
ACLOCAL_PATH: /usr/local/share/aclocal:/opt/X11/share/aclocal
PATH: /usr/local/Library/ENV/4.3:/usr/local/opt/cmake/bin:/usr/local/opt/doxygen/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/libzzip/bin:/opt/X11/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: ogre-1.9 1.9.0 did not build
Logs:
     /Users/federico/Library/Logs/Homebrew/ogre-1.9/01.cmake
     /Users/federico/Library/Logs/Homebrew/ogre-1.9/01.cmake.cc
     /Users/federico/Library/Logs/Homebrew/ogre-1.9/02.make
     /Users/federico/Library/Logs/Homebrew/ogre-1.9/02.make.cc

fferri, 2015-04-18:
Note that I managed to install ogre-1.9 using "brew install https://raw.githubusercontent.com/NikolausDemmel/homebrew-simulation/ogre-fixes/ogre1.9.rb --devel"

Polonium19, 2015-10-29:
I have the same problem. Did you fix the issue?

closed 2017-08-29 by scpeters