osrf/homebrew-simulation #38: Player build failure on OS X 10.10 (Yosemite)

scpeters, 2015-04-28:
I've posted the [detailed console output in a gist](https://gist.github.com/scpeters/d7fcd2f756f9c748c755). @gerkey

gerkey, 2015-04-29:
I can't reproduce this failure on my 10.10 machine (see below).  Any tips?

~~~
Brians-Air:~ gerkey$ brew install player
Warning: player-3.0.2 already installed
Brians-Air:~ gerkey$ brew uninstall player
Uninstalling /usr/local/Cellar/player/3.0.2... (165 files, 4.8M)
Brians-Air:~ gerkey$ brew install player
==> Installing player from osrf/homebrew-simulation
==> Installing player dependency: cmake
==> Downloading https://homebrew.bintray.com/bottles/cmake-3.2.2.yosemite.bottle.3.tar.gz
######################################################################## 100.0%
==> Pouring cmake-3.2.2.yosemite.bottle.3.tar.gz
==> Caveats
Bash completion has been installed to:
  /usr/local/etc/bash_completion.d
==> Summary
🍺  /usr/local/Cellar/cmake/3.2.2: 1850 files, 32M
==> Installing player
==> Downloading https://downloads.sourceforge.net/project/playerstage/Player/3.0.2/player-3.0.2.tar.gz
######################################################################## 100.0%
==> Patching
patching file client_libs/libplayerc++/playerclient.cc
patching file server/drivers/mixed/rflex/rflex.cc
patching file client_libs/libplayerc++/CMakeLists.txt
==> cmake . -DCMAKE_C_FLAGS_RELEASE= -DCMAKE_CXX_FLAGS_RELEASE= -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/player/3.0.2 -DCMAKE_BUILD_TYPE=Release -
==> make install
🍺  /usr/local/Cellar/player/3.0.2: 165 files, 4.5M, built in 2.2 minutes
~~~

scpeters, 2015-04-29:
Maybe we have different Xcode versions? Here's my `brew config`:

~~~
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew.git
HEAD: 4cb6a9a9886b40c27c2d4f729ae66682717e27d4
Last commit: 28 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
CPU: 8-core 64-bit ivybridge
OS X: 10.10.3-x86_64
Xcode: 6.3.1
CLT: 6.3.1.0.1.1429149539
Clang: 6.1 build 602
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby
Java: N/A
~~~

gerkey, 2015-04-30:
Ah, I have:

~~~
OS X: 10.10.2-x86_64
Xcode: 6.1
CLT: N/A
GCC-4.0: build 0
Clang: 6.0 build 600
~~~

Probably increased compiler strictness.  Unfortunately, I need to free up space on this machine before I can update Xcode.

scpeters, 2015-04-30:
I'm planning to upgrade our mac mini to 10.10 this week. You can wait to use that for testing if you prefer.

gerkey, 2015-10-15:
Better late than never.  The build now works on my 10.10 machine.  I can't promise that I didn't break the build for older compilers, but I believe that everything should be OK.

PR #53 (2015-10-15): Fix build on 10.10
Add const qualifier to comparison operators in a class that gets used in a queue, to comply with new compiler strictness.

Fixes #38 .
closed 2015-10-15 by gerkey