osrf/homebrew-simulation #43: Build error for Gazebo 6

jslee02, 2015-07-28:
I got following errors on installing gazebo6:
```
Jeongseoks-MacBook-Pro:Downloads jeongseoklee$ brew install gazebo6
==> Installing gazebo6 from osrf/homebrew-simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo6-prere
Already downloaded: /Library/Caches/Homebrew/gazebo6-6.0.6prerelease.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe59558
Already downloaded: /Library/Caches/Homebrew/gazebo6--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
==> Patching
patching file deps/opende/CMakeLists.txt
==> cmake .. -DCMAKE_C_FLAGS_RELEASE= -DCMAKE_CXX_FLAGS_RELEASE= -DCMAKE_INSTALL

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo620150728-61545-1hkfmp8/gazebo-6.0.6/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/gazebo620150728-61545-1hkfmp8/gazebo-6.0.6/build/CMakeFiles/CMakeError.log".

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/homebrew):
  https://github.com/osrf/homebrew-simulation/issues
```

I believe this is an temporary issue but am reporting just in case.

scpeters, 2015-07-28:
I think it is fixed by d3d8f2b8e680993267d83a33df5952c7eab9d73d. Please reopen if the issue persists

closed 2015-07-28 by scpeters

jslee02, 2015-07-29:
It's fixed for me too. Thanks!