osrf/homebrew-simulation #44: Options for version checkout

ndhir, 2015-08-10:
Hey guys,

Not sure if this is the right place to raise this issue. But I would like to tap simbody version 3.5.1 and not the default option, which is the master version (3.5.3) which is currently the one being used if one taps the git repo osrf/homebrew-simulation. 

Naturally I could get around this by simply installing via git, but automated install and build-process is very handy indeed. 

Just thought I'd add a little memo, to see if this has been addressed before.

Thanks

scpeters, 2015-08-10:
Homebrew used to have a command called `versions` that would let you install old versions of the formula. This was removed recently, though people still ask about it (https://github.com/Homebrew/homebrew/issues/42036 for example).

The short version is that the maintainers prefer that formulae with older versions be maintained separately (ie. as `simbody351.rb`). If you want to make a pull request with a new formula for `simbody351.rb`, I'll merge it in.

If hosting separate formulae for different versions becomes a maintenance burden, I reserve the right to change this policy in the future.

ndhir, 2015-08-10:
If you could merge it in that would be very useful, thank you. 

I completely understand; I cannot speak for the entire community who use simbody in this form. But my impression is that many have not switched from 3.3.1 to 3.5.3 yet, but will do so in due course. In the meantime, having the 3.5.1 would be useful.

Thanks

scpeters, 2015-08-10:
Just to be clear; I'll depend on you making the pull request with the new formula. It should be an easy copy and paste of the current `simbody.rb` to `simbody351.rb` with a change to the tarball name and checksum.

chrisdembia, 2015-08-11:
@ndhir I'm confused; 3.3.1 or 3.5.1? If you meant 3.3.1, this might require some thought since the installation is not compatible with UNIX.

scpeters, 2016-03-11:
The discussion of this issue seems to have stopped. I'll close this for now. Please re-open if the issue persists.

closed 2016-03-11 by scpeters