osrf/homebrew-simulation #45: Memory leak?

dpo, 2015-08-12:
A user submitted a Homebrew formula for `simbody` to Homebrew/Science (probably unaware that there was one here). The user provided a test which caused a memory error for me on 10.9. I replaced the user's test with the pendulum example given in the user's manual and I'm still getting the memory error. For reference, here's the formula: https://gist.github.com/8320f4e14e981dd2e4d7

And this is what happens:
```
$ brew test -v ./simbody.rb 
Testing simbody
==> /usr/bin/clang++ test.cpp -I/usr/local/Cellar/simbody/3.5.3/include/simbody -L/usr/local/Cellar/simbody/3.5.3/lib -lSimTKsimbody -lSimTKmath -lSimTKcommon -llapack -lblas -lpthread -ldl -o test
==> ./test
freeglut (simbody-visualizer):  ERROR:  Function <glutPostRedisplay> called with no current window defined.
simbody-visualizer(16550,0x104638000) malloc: *** error for object 0x7fcbf4031e00: pointer being freed was not allocated
*** set a breakpoint in malloc_error_break to debug
Error: simbody: failed
```

```
$ brew --config
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew.git
HEAD: 5a98bb21e5caafc7a2d5a5141b097b66ffaff70d
Last commit: 5 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: 8-core 64-bit ivybridge
OS X: 10.9.5-x86_64
Xcode: 6.2
CLT: 6.2.0.0.1.1424975374
Clang: 6.0 build 600
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /Users/dpo/.rbenv/shims/ruby => /usr/bin/ruby
Java: 1.6.0_65
```

scpeters, 2015-10-09:
@dpo thanks for the report. I'm cc'ing @sherm1

sherm1, 2015-10-10:
Looks like the test case is trying to run the Visualizer which depends on there being a graphics window present. Is this test then being run from a headless machine somewhere? If so, the visualization should be commented out in test.cpp.

scpeters, 2016-10-02:
I removed the visualizer from the test in https://github.com/osrf/homebrew-simulation/pull/140 and it passes for me in 10.10

scpeters, 2016-10-03:
the test is working now without the visualizer

closed 2016-10-03 by scpeters