osrf/homebrew-simulation #47: Ogre fails to build on 10.11 / Xcode 7

LorenzMeier, 2015-09-26:
It fails to build because it can't find the framework for Xcode 6.x. Would it be possible that one of the maintainers builds a bottle for Ogre and uploads it? That would make the experience more stable.

Otherwise, could you maybe have a quick look at a CMake level patch to pick the 10.11 deployment target instead of 10.10? I'm not up to speed on CMake.

```
==> cmake -DCMAKE_OSX_ARCHITECTURES='x86_64' -DOGRE_BUILD_PLUGIN_CG=OFF -DCMAKE_C_FLAGS_RELEASE= -DCMAKE_CXX_FLAGS_RELEASE= -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/ogre/1.7.4 -DCMAKE_BUILD_T
==> make install
Last 15 lines from /Users/lorenzmeier/Library/Logs/Homebrew/ogre/02.make:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/string:434:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:23:10: fatal error: 'unistd.h' file not found
#include <unistd.h>
         ^
1 error generated.
1 error generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAlignedAllocator.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
1 error generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimable.cpp.o] Error 1
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimationState.cpp.o] Error 1
1 error generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OgreAnimation.cpp.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/homebrew):
  https://github.com/osrf/homebrew-simulation/issues
```

justinthomas, 2015-09-28:
I think this may be related, but I'm not sure. I'm running 10.10.5 with Xcode 7, and I receive the following error:

````
==> cmake -DCMAKE_OSX_ARCHITECTURES='x86_64' -DOGRE_BUILD_PLUGIN_CG=OFF -DCMAKE_C_FLAGS_RELEASE= -DCMAKE_CXX_FLAGS_RELEASE= -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/ogre/1.7.4 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev ..
==> make install
Last 15 lines from /Users/justin/Library/Logs/Homebrew/ogre/02.make:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/LaunchServices.framework/Headers/LSInfo.h:529:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/CoreServices.framework/Frameworks/LaunchServices.framework/Headers/LSInfoDeprecated.h:1264:63: error: expected function body after function declarator
LSGetHandlerOptionsForContentType(CFStringRef inContentType)  __OSX_AVAILABLE_BUT_DEPRECATED_MSG(__MAC_10_4, __MAC_10_11, __IPHONE_4_0, __IPHONE_9_0, "Creator codes are deprecated on OS X.");
                                                              ^
/usr/include/Availability.h:167:53: note: expanded from macro '__OSX_AVAILABLE_BUT_DEPRECATED_MSG'
                                                    __AVAILABILITY_INTERNAL##_osxIntro##_DEP##_osxDep##_MSG(_msg)
                                                    ^
<scratch space>:19:1: note: expanded from here
__AVAILABILITY_INTERNAL__MAC_10_4_DEP__MAC_10_11_MSG
^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [OgreMain/CMakeFiles/OgreMain.dir/src/OSX/OgreConfigDialog.mm.o] Error 1
make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
make: *** [all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/homebrew):
  https://github.com/osrf/homebrew-simulation/issues
````

scpeters, 2015-09-28:
You may also want to try the ogre 1.8 and 1.9 formulae in #24

scpeters, 2015-10-03:
Hm... I just upgraded an old laptop, and it is able to build the ogre formula and gazebo6 without any problems. It only has the command line tools, not the full Xcode suite, so maybe that is the difference?

burrimi, 2015-10-05:
I also have the same problem on OS X 10.10.5
maybe it's fixed in https://bitbucket.org/sinbad/ogre/pull-requests/572/fixes-to-build-ogre-on-macosx/diff

justinthomas, 2015-10-05:
Yes, I think it is an issue with Xcode. I'm currently running OS X 10.10.5,
but I downgraded Xcode to 6.4, and ogre compiles fine now.

LorenzMeier, 2015-10-05:
Can any of the maintainers upload a bottle of gazebo? The core issue is that people need to build it themselves.

scpeters, 2015-10-05:
I'll look into building bottles.

LorenzMeier, 2015-10-05:
It's absurdly easy to do and saves your adopter tons of time and headaches. It's not a great experience to have to wait for the build when almost in homebrew is bottled today - let alone when it doesn't build. It also saves the maintainers precious time because you can then decide when you want to release a new binary and you don't have to maintain rolling releases and tool upgrades any more as diligently.

The ability of Dronecode to start relying on Gazebo critically depends on this level of maintenance.

Instructions here:
https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/Bottles.md

scpeters, 2015-10-05:
I'll clarify, I was already aware of bottles and have experimented with them before, but they aren't yet part of our automated release process. I'm building an ogre bottle for 10.11 as we speak. The command-line tool does make it very easy, but it adds to our support workload, which is why it hasn't happened prior to now.

There may be a problem with bottling gazebo and the bullet physics engine. The official homebrew bottles all use default options. Gazebo versions 5+ depend on bullet by default but require the non-default `--with-shared` option to build shared libraries. I'm not sure if that will be a problem or not.

scpeters, 2015-10-05:
ogre bottles uploaded for 10.10, 10.11 in 22d01e59b25d4697796dc8c73c6c5976dc7829ac, please test

scpeters, 2015-10-06:
Link to our discussion about automating this:

* https://bitbucket.org/osrf/release-tools/issues/47/update-homebrew-formulae-from-releasepy

burrimi, 2015-10-06:
Thanks a lot @scpeters, seems to work for me!

LorenzMeier, 2015-10-07:
A bottle for sdformat would be great:

```
Last 15 lines from /Users/lorenzmeier/Library/Logs/Homebrew/sdformat/02.make:
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/sphere_shape.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 42%] Running xml schema compiler on sonar.sdf
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/sonar.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 43%] Running xml schema compiler on state.sdf
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/state.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 44%] Running xml schema compiler on surface.sdf
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/surface.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 45%] Running xml schema compiler on transceiver.sdf
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/transceiver.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 46%] Running xml schema compiler on visual.sdf
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/visual.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 47%] Running xml schema compiler on world.sdf
cd /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/sdf/1.5/world.sdf -o /tmp/sdformat20151007-26350-izs945/sdformat-2.3.2/build/sdf/1.5
[ 47%] Built target schema
make: *** [all] Error 2
```

LorenzMeier, 2015-10-07:
I presume you're aware of the OS X support for Travis? That could help building and deploying the bottles. Or is there a way for you to upstream that to the mainline homebrew project?
http://docs.travis-ci.com/user/workers/os-x-infrastructure/

LorenzMeier, 2015-10-07:
The Ogre bottle worked btw, many thanks! :+1:

scpeters, 2015-10-07:
That sdformat log doesn't show the error. Could you paste the whole thing in a gist?

We currently are using travis for this tap, but it's just linting the formulae (`brew audit` and `brew style`). We are looking into using the same CI script that mainline homebrew uses.

Glad the ogre bottle works :)

LorenzMeier, 2015-10-07:
It fails at the make install step, sec.

LorenzMeier, 2015-10-07:
```
==> Installing gazebo from osrf/homebrew-simulation
==> Installing dependencies for osrf/simulation/gazebo: sdformat
==> Installing osrf/simulation/gazebo dependency: sdformat
==> Downloading http://gazebosim.org/distributions/sdformat/releases/sdformat-2.3.2.tar.bz2
Already downloaded: /Library/Caches/Homebrew/sdformat-2.3.2.tar.bz2
==> cmake -DCMAKE_C_FLAGS_RELEASE= -DCMAKE_CXX_FLAGS_RELEASE= -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/sdformat/2.3.2 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MA
==> make install
Last 15 lines from /Users/lorenzmeier/Library/Logs/Homebrew/sdformat/02.make:
make[2]: *** [src/CMakeFiles/sdformat.dir/Assert.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 46%] Running xml schema compiler on transceiver.sdf
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5/transceiver.sdf -o /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5
[ 47%] Running xml schema compiler on visual.sdf
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5/visual.sdf -o /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5
[ 48%] Running xml schema compiler on world.sdf
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5/world.sdf -o /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5
[ 48%] Built target schema
1 error generated.
make[2]: *** [src/CMakeFiles/sdformat.dir/Console.cc.o] Error 1
1 error generated.
make[2]: *** [src/CMakeFiles/sdformat.dir/Converter.cc.o] Error 1
make[1]: *** [src/CMakeFiles/sdformat.dir/all] Error 2
make: *** [all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/homebrew):
  https://github.com/osrf/homebrew-simulation/issues
```

LorenzMeier, 2015-10-07:
I'm pretty sure its again XCode version and it just needs a bottle.

LorenzMeier, 2015-10-07:
Yep, again failing because of moved standard headers:

```
make
install

/usr/local/Cellar/cmake/3.3.2/bin/cmake -H/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2 -B/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/local/Cellar/cmake/3.3.2/bin/cmake -E cmake_progress_start /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/CMakeFiles /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/CMakeFiles/progress.marks
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f CMakeFiles/Makefile2 all
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f test/CMakeFiles/gtest.dir/build.make test/CMakeFiles/gtest.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f src/CMakeFiles/sdformat.dir/build.make src/CMakeFiles/sdformat.dir/depend
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f sdf/1.5/CMakeFiles/schema.dir/build.make sdf/1.5/CMakeFiles/schema.dir/depend
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build && /usr/local/Cellar/cmake/3.3.2/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2 /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/test /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/test/CMakeFiles/gtest.dir/DependInfo.cmake --color=
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build && /usr/local/Cellar/cmake/3.3.2/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2 /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5 /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5 /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5/CMakeFiles/schema.dir/DependInfo.cmake --color=
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build && /usr/local/Cellar/cmake/3.3.2/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2 /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/src /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/src /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/src/CMakeFiles/sdformat.dir/DependInfo.cmake --color=
Scanning dependencies of target schema
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f sdf/1.5/CMakeFiles/schema.dir/build.make sdf/1.5/CMakeFiles/schema.dir/build
[  1%] Running xml schema compiler on actor.sdf
[  2%] Running xml schema compiler on audio_source.sdf
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5/audio_source.sdf -o /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5 && ../../../tools/xmlschema.rb -s /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5 -i /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/sdf/1.5/actor.sdf -o /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/sdf/1.5
Scanning dependencies of target gtest
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f test/CMakeFiles/gtest.dir/build.make test/CMakeFiles/gtest.dir/build
[  3%] Building CXX object test/CMakeFiles/gtest.dir/gtest/src/gtest-all.cc.o
cd /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/test && /usr/local/Library/ENV/4.3/clang++   -DTIXML_USE_STL -fvisibility=hidden  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic   -O2 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test/gtest/include -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/src/win/tinyxml -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/include -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/build/include -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test/gtest -I/tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test -I/usr/local/include    -fPIC -o CMakeFiles/gtest.dir/gtest/src/gtest-all.cc.o -c /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test/gtest/src/gtest-all.cc
In file included from /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test/gtest/src/gtest-all.cc:39:
In file included from /tmp/sdformat20151007-62126-1niaf5t/sdformat-2.3.2/test/gtest/include/gtest/gtest.h:54:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/limits:109:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/__config:23:10: fatal error: 'unistd.h' file not found
#include <unistd.h>
```

scpeters, 2015-10-07:
Regarding the `unistd.h` problem, I found a [suggestion from GNURadio](https://gnuradio.org/redmine/projects/gnuradio/wiki/MacInstall#Fatal-error-unistdh-file-not-found):

If your install fails and you notice a line in the reported log file that says something like "Fatal error: 'unistd.h' file not found", it's likely due to the path to the active developer tools not correctly setup. To fix, simply execute the following command:

~~~
xcode-select --install
~~~

A confirmation dialog box will pop-up and the tools will be installed. From here, re-kickstart the gnuradio installation and you should be golden.

LorenzMeier, 2015-10-07:
Thanks, that did it. What is super weird is that I'm building Qt and command line projects successfully.

GerrieWell, 2015-10-15:
Is there some one tell me what is bottles and how to use these?I found this page while I was searching orge 'unistd.h' file not found .
I have tried reinstall manually the command-line-tools for EI capitan and upgraded Xcode to 7.0.1.But it Doesn't work.
BTW,I have tried  using gcc4.9 to build orge,It appear new problem:

```
                 from /usr/local/Cellar/gcc/4.9.2_1/include/c++/4.9.2/string:40,
                 from /Users/gerrie/source/offical/OgreSDK1.8/include/OGRE/OgrePrerequisites.h:33,
                 from /Users/gerrie/source/offical/OgreSDK1.8/include/OGRE/OgrePlugin.h:31,
                 from /Users/gerrie/source/offical/OgreSDK1.8/include/OGRE/SamplePlugin.h:31,
                 from /Users/gerrie/source/offical/OgreSDK1.8/Samples/BezierPatch/src/BezierPatch.cpp:1:
/usr/local/Cellar/gcc/4.9.2_1/include/c++/4.9.2/cwchar:44:19: fatal error: wchar.h: No such file or directory
 #include <wchar.h>
```

```
In file included from /Users/gerrie/source/offical/OgreSDK1.8/Samples/BezierPatch/src/BezierPatch.cpp:1:
In file included from /Users/gerrie/source/offical/OgreSDK1.8/include/OGRE/SamplePlugin.h:31:
In file included from /Users/gerrie/source/offical/OgreSDK1.8/include/OGRE/OgrePlugin.h:31:
In file included from /Users/gerrie/source/offical/OgreSDK1.8/include/OGRE/OgrePrerequisites.h:33:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/string:434:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/__config:23:10: fatal error: 'unistd.h' file not found
```

scpeters, 2015-10-15:
@GerrieWell bottles are a binary build of a homebrew formula and are [described here in the homebrew documentation](https://github.com/Homebrew/homebrew/blob/master/share/doc/homebrew/Bottles.md). I have built bottles for a few packages, including ogre, so you shouldn't have to compile them. All I can say about the compilation errors is that like the developer tools are not properly configured. Did you try the `xcode-select --install` command?

osrfbuild, 2015-10-29:
This problem just cropped up on our CI machine when I upgraded Xcode. The `xcode-select --install` command didn't help. I'm currently debugging...

scpeters, 2015-10-29:
Whoops, that comment by @osrfbuild was from me.

The following seems to indicate that it is a problem with Xcode 7:

* https://forums.developer.apple.com/thread/17334

scpeters, 2015-10-29:
The CI machine is running Yosemite (10.10), and I've uninstalled Xcode to see if that will work. I imagine that won't work for everyone, but there are some other suggestions in that forum thread that could be tried.

GerrieWell, 2015-10-29:
I have installed ogre1.8 by the bottles.Thank you.

scpeters, 2016-03-11:
I'm now seeing the same build errors as @justinthomas on my 10.10 machine that has Xcode installed. It seems to build properly if only the command-line tools are installed (though that required a recent fix #85). Strangely, if I `brew install ogre --build-from-source` with `--debug` and then repeat some of the commands in a debug bash terminal, it is able to finish successfully.

scpeters, 2016-10-06:
I ran into this problem again (still using 10.10) where it was trying to use the 10.11 SDK provided by Xcode. My solution is to export the following environment variables:

~~~
export MACOSX_DEPLOYMENT_TARGET=10.10
export SDKROOT=/
~~~

These are used by cmake:

* https://cmake.org/cmake/help/v3.6/variable/CMAKE_OSX_SYSROOT.html?highlight=sdkroot
* https://cmake.org/cmake/help/v3.6/variable/CMAKE_OSX_DEPLOYMENT_TARGET.html?highlight=MACOSX_DEPLOYMENT_TARGET

closed 2017-08-29 by scpeters