osrf/homebrew-simulation #54: gazebo5 (5.2.1) fails to install (due to failed patch)

jriesen, 2015-10-16:
$ brew install gazebo5
==> Installing gazebo5 from osrf/homebrew-simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-5.2.1.
==> Downloading from https://osrf-distributions.s3.amazonaws.com/gazebo/releases
######################################################################## 100.0%
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe59558
######################################################################## 100.0%
==> Downloading https://bitbucket.org/osrf/gazebo/commits/91f6f3c59f40af34855548
######################################################################## 100.0%
==> Downloading https://gist.githubusercontent.com/scpeters/ef860211b6536a3d7a20
######################################################################## 100.0%
==> Patching
patching file deps/opende/CMakeLists.txt
patching file gazebo/gui/TimePanel.cc
Hunk #1 FAILED at 178.
1 out of 1 hunk FAILED -- saving rejects to file gazebo/gui/TimePanel.cc.rej
patching file gazebo/gui/TimePanel.hh
Hunk #1 FAILED at 66.
1 out of 1 hunk FAILED -- saving rejects to file gazebo/gui/TimePanel.hh.rej
Error: Failure while executing: /usr/bin/patch -g 0 -f -p1 -i /private/tmp/gazebo5--patch20151015-70580-5qtyg8/raw

Looking at https://github.com/osrf/homebrew-simulation/blob/master/gazebo5.rb, it seems like the patch at https://bitbucket.org/osrf/gazebo/commits/91f6f3c59f40af34855548c37857955d08fd1368/raw/ is no longer valid for Gazebo 5.2.1 as it was for 5.1.0 -- this is broken by https://github.com/osrf/homebrew-simulation/commit/c00c6bc8dd591805c93c54fee6f06e6494252047 since, if I use the previous version of gazebo5.rb, everything installs fine (albeit version 5.1.0).

scpeters, 2015-10-16:
Thanks for the report! I've removed the offending patches in 316267bf6fae3d8542f6f881cd5b75a1f75dfee5. That's what happens when I merge without testing...

The good news is I'm working on CI for pull requests using the same scripts that upstream homebrew is using.

closed 2015-10-16 by scpeters

scpeters, 2015-10-16:
I also just added a check to travis that will attempt to apply all specified patches (9cf102dad9cd0ba29319f6b6bee8dccee24366e3). It doesn't try to build everything, but at least we should catch these issues with outdated patches much sooner.

Thanks again for the report!