osrf/homebrew-simulation #63: Gazebo 6 bottle segfaults on 10.10

LorenzMeier, 2015-11-23:
```bash
$ gzserver
Segmentation fault: 11
$ gzclient
Error. Gui Invalid arguments
Segmentation fault: 11
```

LorenzMeier, 2015-11-23:
```
lldb gzserver
(lldb) target create "gzserver"
Current executable set to 'gzserver' (x86_64).
(lldb) run
Process 11844 launched: '/usr/local/bin/gzserver' (x86_64)
Process 11844 stopped
* thread #1: tid = 0x7d59c3, 0x00000001029397b2 libboost_program_options-mt.dylib`boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, bool) const + 50, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0xffffffffffffffe8)
    frame #0: 0x00000001029397b2 libboost_program_options-mt.dylib`boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, bool) const + 50
libboost_program_options-mt.dylib`boost::program_options::value_semantic_codecvt_helper<char>::parse:
->  0x1029397b2 <+50>: movq   (%rbx), %rax
    0x1029397b5 <+53>: cmpq   %rax, 0x8(%rbx)
    0x1029397b9 <+57>: je     0x102939857               ; <+215>
    0x1029397bf <+63>: movl   $0x1, %r14d
```

LorenzMeier, 2015-11-23:
@mcharleb It would be great if we could have a coordinated effort with your team and OSRF to overcome the usability issues with Gazebo on all POSIX platforms in the next week or so and target a stable release.

scpeters, 2016-03-11:
The bottles have been rebuilt and I believe this has been resolved.

closed 2016-03-11 by scpeters