osrf/homebrew-simulation #72: gazebo 6 fails to launch

kenzierocks, 2016-01-19:
After doing `brew install gazebo6`, this occurs when running `gazebo`:

```
dyld: Symbol not found: __ZNK5boost9re_detail31cpp_regex_traits_implementationIcE17transform_primaryEPKcS4_
  Referenced from: /usr/local/opt/sdformat3/lib/libsdformat.3.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib
 in /usr/local/opt/sdformat3/lib/libsdformat.3.dylib
[1]    61230 trace trap  gazebo
```

scpeters, 2016-01-21:
Sorry, our support for bottles is still a bit flaky. The bottles can be broken when new versions of dependencies are released (in this case boost) and they need to be rebuilt. We are still working on infrastructure for detecting bottle failures and rebuilding them.

In the meantime, can you use `--build-from-source` when you install `gazebo6`? Sorry for the trouble.

kenzierocks, 2016-01-22:
Same error with `--build-from-source`. I am sure it builds from source, as it takes quite a while :P

kenzierocks, 2016-01-22:
Oops, had to `--build-from-source` `sdformat3`. All is good now.

scpeters, 2016-03-11:
The bottles have been rebuilt and I believe they are working properly now.

closed 2016-03-11 by scpeters