osrf/homebrew-simulation #76: player error

lianne, 2016-01-27:
I am not able to install player 3.0.2 on Yosemite. I get an error on building libplayerc++.dylib

My gist is [here](https://gist.github.com/lianne/48f3d6b10ed6399709fe)

I also tried the HEAD version: `brew install player --HEAD` but a patch error occurs.

Is it a configuration problem?
Thanks,

   Luigi

scpeters, 2016-01-27:
It works for me, but I have an old version of Xcode (7.0.1). I'll try it with a more recent version and see if I get the same failure.

scpeters, 2016-01-27:
I just successfully built it on my virtual machine with the latest Xcode:

~~~
$ brew config
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew
HEAD: 890db3738d139a6ee3c5a6a44d5d332e85e7b196
Last commit: 25 minutes ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: dual-core 64-bit ivybridge
OS X: 10.10.5-x86_64
Xcode: N/A
CLT: 7.2.0.0.1.1447826929
Clang: 7.0 build 700
X11: 2.7.8 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/bin/python
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: N/A
~~~

scpeters, 2016-01-27:
never mind, that machine doesn't have Xcode. I'll install it now...

scpeters, 2016-01-27:
I've tested every permutation I can think of, and I can't reproduce this error. I'll punt by cc'ing @gerkey

gerkey, 2016-01-27:
I haven't tried to reproduce it, but based on the fact that the linker errors are for the swig-generated bindings to `playerc`,  you might try disabling them. If you're outside brew, this `cmake` command should do it:

    cmake -DBUILD_RUBY_BINDINGS=0 -DBUILD_PYTHONC_BINDINGS=0 ..

You could try that manually, then try it as a modification to the brew formula.

(Of course, I'm assuming here that you don't *need* the Python or Ruby bindings. If you do need them, then we'll need to figure out what's causing them to fail to link.)

gerkey, 2016-01-27:
More drastically, you could also uninstall `swig`; that should disable the build of those bindings.

lianne, 2016-01-28:
yes!! without `swig` everything worked fine! thanks, @gerkey

closed 2016-01-28 by lianne

gerkey, 2016-01-28:
I'm now able to reproduce the problem by brew installing swig and then trying to brew install player.

Turns out that there's already an issue for this upstream: playerproject/player#3.

scpeters, 2016-01-29:
Is there a cmake argument we could pass to disable the swig bindings?

gerkey, 2016-01-29:
@scpeters I didn't see a specific argument to disable swig, but adding these cmake args would disable both the swig-generated bindings that are causing trouble:

    -DBUILD_RUBY_BINDINGS=0 -DBUILD_PYTHONC_BINDINGS=0

scpeters, 2016-01-30:
Ok, I added those cmake args to the player formula in 19922e1724592fc8ce9976b8e0bb0248ff52950b