osrf/homebrew-simulation #79: SHA1 conflict

jacksonkr, 2016-02-09:
This is related to a fork of this package:

https://github.com/Cescante/homebrew-simulation/
forked from
https://github.com/NikolausDemmel/homebrew-simulation/

```
$ brew install https://raw.githubusercontent.com/Cescante/homebrew-simulation/master/ogre1.9.rb --devel
######################################################################## 100.0%
==> Downloading https://bitbucket.org/sinbad/ogre/get/v1-9.tar.bz2
######################################################################## 100.0%
Error: SHA1 mismatch
Expected: 545f954b7869f3093f9d7d9ec63638a6aaa8eb4e
Actual: c7cde37a23d739fa5c691f08e5452522b7ac160a
Archive: /Library/Caches/Homebrew/ogre1.9-1.9.1-devel.tar.bz2
To retry an incomplete download, remove the file above.
```

I've tried removing the cache and redownload a number of times. What's the solution here?

scpeters, 2016-02-09:
It looks like it's building the head of a branch. If the branch has had new commits, then the SHA won't match.

jacksonkr, 2016-02-09:
If I `brew install ogre` then it installs 1.7.4 and I need 1.8. I went with the HEAD and it's giving me 1.9.1. How would I go about installing 1.8?

scpeters, 2016-02-09:
Try the ogre1.8 formula:

~~~
brew install https://raw.githubusercontent.com/NikolausDemmel/homebrew-simulation/ogre-fixes/ogre1.8.rb
~~~

jacksonkr, 2016-02-09:
So this was odd:

```
$ brew install https://raw.githubusercontent.com/NikolausDemmel/homebrew-simulation/ogre-fixes/ogre1.8.rb
######################################################################## 100.0%
==> Downloading http://downloads.sourceforge.net/project/ogre/ogre/1.8/1.8.1/ogr
==> Downloading from http://iweb.dl.sourceforge.net/project/ogre/ogre/1.8/1.8.1/
######################################################################## 100.0%
==> Downloading https://gist.githubusercontent.com/NikolausDemmel/dad2dec7e52bed
######################################################################## 100.0%
==> Patching
==> Applying fixogre1.8.patch
patching file CMake/Packages/FindFreetype.cmake
Hunk #1 FAILED at 31.
Hunk #2 succeeded at 43 with fuzz 1.
1 out of 2 hunks FAILED -- saving rejects to file CMake/Packages/FindFreetype.cmake.rej
patching file CMake/Utils/OgreConfigTargets.cmake
patching file CMakeLists.txt
Hunk #1 succeeded at 238 (offset 2 lines).
patching file Components/Paging/CMakeLists.txt
patching file Components/Property/CMakeLists.txt
patching file Components/RTShaderSystem/CMakeLists.txt
patching file Components/Terrain/CMakeLists.txt
patching file OgreMain/CMakeLists.txt
patching file PlugIns/BSPSceneManager/CMakeLists.txt
patching file PlugIns/CgProgramManager/CMakeLists.txt
patching file PlugIns/OctreeSceneManager/CMakeLists.txt
patching file PlugIns/OctreeZone/CMakeLists.txt
patching file PlugIns/PCZSceneManager/CMakeLists.txt
patching file PlugIns/ParticleFX/CMakeLists.txt
patching file RenderSystems/GL/CMakeLists.txt
patching file CMake/ConfigureBuild.cmake
patching file CMake/FeatureSummary.cmake
patching file CMake/Templates/OgreBuildSettings.h.in
patching file CMakeLists.txt
Hunk #6 succeeded at 250 (offset 2 lines).
Hunk #7 succeeded at 320 (offset 3 lines).
Hunk #8 succeeded at 345 (offset 3 lines).
Hunk #9 succeeded at 361 (offset 3 lines).
Hunk #10 succeeded at 378 (offset 3 lines).
Hunk #11 succeeded at 406 (offset 3 lines).
patching file Components/Volume/include/OgreVolumeMeshBuilder.h
patching file OgreMain/include/OgreMemorySTLAllocator.h
patching file RenderSystems/GL/include/OgreGLFBORenderTexture.h
patching file RenderSystems/GL/src/OSX/OgreOSXCarbonWindow.cpp
patching file RenderSystems/GL/src/OSX/OgreOSXCocoaWindow.mm
patching file RenderSystems/GL/src/atifs/include/ps_1_4.h
patching file RenderSystems/GL3Plus/include/OgreGL3PlusFBORenderTexture.h
patching file RenderSystems/GL3Plus/src/OSX/OgreOSXCocoaWindow.mm
Error: Failure while executing: /usr/bin/patch -g 0 -f -p1 -i /private/tmp/ogre1.8--patch20160209-66840-10nylq4/fixogre1.8.patch
```

scpeters, 2016-02-09:
Can we move this discussion to https://github.com/osrf/homebrew-simulation/pull/24 ? I don't know enough about these alternate formulae for ogre to answer your questions.

jacksonkr, 2016-02-09:
Do we close this and post the issue into the other thread or is there some fancy way of moving this over?

scpeters, 2016-02-09:
I don't know of a way to move the discussion. I think we need to close and then repost your question over there. I'll close this now if that's alright.

closed 2016-02-09 by scpeters