osrf/homebrew-simulation #84: OGRE formula fails to link with AGL framework

scpeters, 2016-03-09:
It's failing in the gazebo CI builds:

* http://build.osrfoundation.org/job/gazebo-ci-default-homebrew-amd64/48/

Here's the detailed build log:

* https://gist.github.com/anonymous/06f8790ab617f2b30d21

It can't find some symbols from the AGL framework.

scpeters, 2016-03-09:
I think the bottle still works on yosemite, but there's a problem with building from source.

scpeters, 2016-03-09:
That [first set of logs](https://gist.github.com/anonymous/06f8790ab617f2b30d21) is from a machine that only has command-line tools.

On my laptop that has Xcode 7.2 installed, I see a different set of errors.

scpeters, 2016-03-09:
@wjwwood any advice? You've patched the ogre formula a few times already.

wjwwood, 2016-03-09:
I dunno, looking at this page: https://developer.apple.com/library/mac/documentation/GraphicsImaging/Reference/AGL_OpenGL/index.html

It looks like some of those functions might be deprecated in 10.9? It's not clear since they're struct-through but it doesn't explicitly say deprecated anywhere.

Either way it doesn't make sense that it has the headers but not the library symbols. I'd guess it has something to do with the OGRE build system not passing a `-framework` argument correctly or something like that.

commit 5ec46a01 (2016-03-10): ogre: patch for finding AGL framework

Closes #84.

Closes #85.
closed 2016-03-10 by scpeters