osrf/homebrew-simulation #87: gazebo depends on specific protobuf version

moriarty, 2016-03-17:
I had an issue installing gazebo_ros_pkgs using the script from mikepurvis: 
https://github.com/mikepurvis/ros-install-osx/issues/33.

I solved it, but I don't think my solution is very nice. I think that gazebo should specify a protobuf version in the dependencies because the tar includes pb.h files which specify a version. 

I have ```protobuf --devel``` installed with brew for another project which causes the problem. Gazebo installs fine, and it's not until trying to use the headers somewhere else that it fails. It would be nice if gazebo could specify the version, so the error would at least show up sooner. 

See link to issue above for more detail.

scpeters, 2016-03-17:
I commented at the other issue and I'll repeat it here:

Maybe the issue is that we depend on `protobuf-c`, which depends on `protobuf`. If you installed the `protobuf-c` bottle, that may be build against the old version of `protobuf`. What if you build `protobuf-c` from source against your `--devel` version of `protobuf`?

moriarty, 2016-03-24:
Thanks,

Until I get a chance to test I will close the issues, if the fix suggested doesn't work I will reopen one of the issues.

closed 2016-03-24 by moriarty