osrf/homebrew-simulation #112: Problem with the compilation of Gazebo

lilouch, 2016-07-13:
Hi, I'm trying to install ROS on my OSX El Capitan 10.11.5  but without success.

I'm not able to compile perfectly Gazebo. I got this error : 
```
,/usr/local/lib -Wl,-rpath,/usr/local/Cellar/bullet/2.83.7/lib 
ld: library not found for -ltinyxml
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [gazebo/msgs/libgazebo_msgs.7.2.0.dylib] Error 1
make[1]: *** [gazebo/msgs/CMakeFiles/gazebo_msgs.dir/all] Error 2
make: *** [all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/osrf/homebrew-simulation/issues

ERROR: the following rosdeps failed to install
  homebrew: command [brew install gazebo7] failed
```

It seems that it is not able to find tinyxml. However I checked and It well installed.
The output of the following command:
`brew info tinyxml`

```
tinyxml: stable 2.6.2 (bottled)
XML parser
http://www.grinninglizard.com/tinyxml/
/usr/local/Cellar/tinyxml/2.6.2 (6 files, 180.6K) *
  Poured from bottle on 2016-07-12 at 15:23:38
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/tinyxml.rb
==> Dependencies
Build: cmake ✔
==> Options
--universal
	Build a universal binary
```

And the output of 
`brew list tinyxml`

```
/usr/local/Cellar/tinyxml/2.6.2/include/tinyxml.h
/usr/local/Cellar/tinyxml/2.6.2/lib/libtinyxml.dylib
/usr/local/Cellar/tinyxml/2.6.2/lib/pkgconfig/tinyxml.pc
```

Coud you please help me ?

Thank in advance,
Regards

scpeters, 2016-07-13:
We've had issues with linking to tinyxml before, but I thought we had resolved them. Can you confirm that tinyxml is linked: `brew link tinyxml`

And if that doesn't work, can you provide more verbose error logs:

~~~
brew install gazebo7 --verbose
brew gist-logs gazebo7
~~~

Thanks!

tarasrudyy, 2016-07-18:
Hi, I've just faced with similar issue but on gazebo_rendering library building

https://gist.github.com/tarasrudyy/154a7bf0e18a4b1b042c403be62f79e4

Thank you for help!

scpeters, 2016-07-18:
@tarasrudyy thanks for the report. That appears to be a different error, which has also affected our build farm:

* http://build.osrfoundation.org/job/gazebo-ci-default-homebrew-amd64/115/

I'm working on it now.

scpeters, 2016-07-18:
Looks like an incompatibility with the recently released version of tinyxml2; I've submitted a pull request to fix upstream gazebo:

* https://bitbucket.org/osrf/gazebo/pull-requests/2397/use-xml_success-enum-instead-of/diff

I'll patch the gazebo7 formula shortly.

scpeters, 2016-07-18:
I've patched the formula for the problem reported by @tarasrudyy in 0a113c46af115da4c87e3537c7db23e58b350eee

It should work if you try it again.

lilouch, 2016-07-19:
Here the output of brew link : 
```
Warning: Already linked: /usr/local/Cellar/tinyxml/2.6.2
To relink: brew unlink tinyxml && brew link tinyxml
```
So It's already linked.

I tried brew install gazebo7 --verbose and it has been installed successfully !
```
==> Summary
🍺  /usr/local/Cellar/gazebo7/7.3.1: 1,118 files, 110.6M, built in 12 minutes 57 seconds
```

The output of
`brew gist-logs gazebo7`

 https://gist.github.com/072a98989d6fee1561b93199e7b7f7b6

Maybe you solved the problem ! I will try now to install again ROS.
Thank.

stanchiang, 2016-09-13:
i had the same issue. mytinyxml seems fine though. 
[gazeboerror.txt](https://github.com/osrf/homebrew-simulation/files/468880/gazeboerror.txt)

scpeters, 2016-09-14:
@stanchiang that is a different error with compiling ignition-transport

can you post it to a different issue because this one was about gazebo and I think it's fixed now

closed 2016-09-14 by scpeters

ReemAlattas, 2017-12-03:
I have the same problem!!! I've been trying to update xCode but no luck.

```
Error: You are using macOS 10.10.
We (and Apple) do not provide support for this old version.
You may encounter build failures or other breakages.
Please create pull-requests instead of filing issues.

Error: Your Xcode (7.2) is outdated.
Please update to Xcode 7.2.1 (or delete it).
Xcode can be updated from the App Store.
```