osrf/homebrew-simulation #120: ignition-transport 1.3.0 compilation failure

bowang, 2016-07-28:
```
Last 15 lines from /Users/bowang/Library/Logs/Homebrew/ignition-transport/02.make:
/tmp/ignition-transport-20160727-16750-r7edtp/ignition-transport-1.3.0/include/ignition/transport/SubscriptionHandler.hh:148:53: error: expected expression
          auto msgPtr = google::protobuf::down_cast<const T*>(&_msg);
                                                    ^
7 errors generated.
make[2]: *** [src/CMakeFiles/ignition-transport1.dir/ign.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [src/CMakeFiles/ignition-transport1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  8%] Linking CXX static library libgtest.a
cd /tmp/ignition-transport-20160727-16750-r7edtp/ignition-transport-1.3.0/test && /usr/local/Cellar/cmake/3.6.1/bin/cmake -P CMakeFiles/gtest.dir/cmake_clean_target.cmake
cd /tmp/ignition-transport-20160727-16750-r7edtp/ignition-transport-1.3.0/test && /usr/local/Cellar/cmake/3.6.1/bin/cmake -E cmake_link_script CMakeFiles/gtest.dir/link.txt --verbose=1
/usr/bin/ar qc libgtest.a  CMakeFiles/gtest.dir/gtest/src/gtest-all.cc.o
/usr/bin/ranlib libgtest.a
[  8%] Built target gtest
make: *** [all] Error 2
```

OS: Mac OSX 10.11.6 El Capitan
Compiler: Apple LLVM version 7.3.0 (clang-703.0.29)

scpeters, 2016-07-29:
Thanks for the report. I just set up a jenkins build to replicate the failure, since I'm still using 10.10 on my personal machine ([the first build failed](http://build.osrfoundation.org/job/ignition_transport-ci-pr_any-homebrew_elcap-amd64/1/) for an unrelated issue (missing `hg` installation), cc @j-rivero ). We'll continue working to reproduce and diagnose the issue.

uframer, 2016-09-15:
I encountered the same issue. It looks like the "osrf/simulation/ignition-transport: stable 1.3.0" package depends on protobuf260 but now "protobuf: stable 3.0.2" is the default protobuf version.

I added "#include <google/protobuf/stubs/casts.h>" to ignition-transport's RepHandler.hh to include the declaration of "down_cast" then the build passed. Unfortunately I don't know how to fix this in Homebrew. Hope this info can help you guys come up an solution.

uframer, 2016-09-16:
Here is a quick solution:
```patch
diff --git a/ignition-transport.rb b/ignition-transport.rb
index ea63d5c..3474955 100644
--- a/ignition-transport.rb
+++ b/ignition-transport.rb
@@ -16,7 +16,7 @@ class IgnitionTransport < Formula
   depends_on "pkg-config" => :run

   depends_on "ignition-tools"
-  depends_on "protobuf"
+  depends_on "protobuf260"
   depends_on "protobuf-c" => :build
   depends_on "ossp-uuid"
   depends_on "zeromq"
```

I will make a pull request for it.

j-rivero, 2016-09-16:
I've [patched the debian unstable package](https://anonscm.debian.org/cgit/debian-science/packages/ignition-transport.git/tree/debian/patches/0002-fix-protobuf3.patch) to compile with the new protobuf3 series. Still need to prepare a source change for upstream code though.

You can use the same patch in the formula just in case that you want to keep using protobuf3.

j-rivero, 2016-09-16:
> Still need to prepare a source change for upstream code though.

https://bitbucket.org/ignitionrobotics/ign-transport/pull-requests/173/support-for-compilations-with-protobuf3/diff.

uframer, 2016-09-16:
@j-rivero 
I believe your patch to keep using protobuf3 is the bettor solution because gazebo7 itself may rely on protobuf3 and introducing dependency to probobuf260 could cause dependency conflicts.

scpeters, 2016-09-16:
I've added that patch to the ignition-transport formula in #133

scpeters, 2016-09-20:
closed by #133

closed 2016-09-20 by scpeters