osrf/homebrew-simulation #121: No bottle for Gazebo 7 on 10.11

LorenzMeier, 2016-07-28:
@wjwwood Would it be possible to get a bottle up for Gazebo 7 on Mac OS? I can help to maintain it if you like - I can stick one onto the PX4 AWS.

scpeters, 2016-07-28:
@j-rivero and I have been working on the bottling infrastructure for this tap on the http://build.osrfoundation.org jenkins instance (example [pull request](https://bitbucket.org/osrf/release-tools/pull-requests/253/fix-bottle-building/diff)). We are currently generating bottles for 10.10 automatically as part of our release pipeline that also generates debians, but we still need to add support for multiple distros. That will involve adding an additional jenkins parameter and tweaking some of the scripts. I'll move it up our todo list.

LorenzMeier, 2016-12-17:
@wjwwood This is becoming a critical issue for us now. Can we discuss via email (lorenz@px4.io) how we can assist you with the maintenance of Gazebo bottles? @julianoes building a bottle is easy, and we probably need to help OSRF a bit with the Mac maintenance so its not a 2 hour install (which it currently is for us).

wjwwood, 2016-12-22:
@LorenzMeier the gazebo formula is maintained by @scpeters, but I'm happy to help him as I have time. I'll talk to him offline and see where we can collaborate on this.

scpeters, 2016-12-22:
I have a second build machine running 10.11, and I am configuring it right now. This is a side project for me, so I can't promise when it will be working, but I think it could be working soon.

scpeters, 2016-12-22:
I just manually built an el_capitan bottle for ogre1.9 in #172 to verify that the build machine is properly configured. Please check it if you have a chance. I'll start working on adding that machine to our buildfarm.

scpeters, 2017-07-05:
we have bottles for 10.11 and 10.12 now

closed 2017-07-05 by scpeters