osrf/homebrew-simulation #135: simbody build failed

uframer, 2016-09-17:
```bash
$ brew install simbody
==> Installing simbody from osrf/simulation
==> Downloading https://github.com/simbody/simbody/archive/Simbody-3.5.3.tar.gz
Already downloaded: /Users/jiaoye/Library/Caches/Homebrew/simbody-3.5.3.tar.gz
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/simbody/3.5.3 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRA
==> make doxygen
==> make install
Last 15 lines from /Users/jiaoye/Library/Logs/Homebrew/simbody/03.make:
    ^~~~~~~~~~~~~
/tmp/simbody-20160917-18434-g4s3gc/simbody-Simbody-3.5.3/SimTKcommon/./include/SimTKcommon/internal/Timing.h:110:34: note: candidate function not viable: no known conversion from 'int' to 'clockid_t' for 1st argument
    SimTK_SimTKCOMMON_EXPORT int clock_gettime(clockid_t clock_id,
                                 ^
/tmp/simbody-20160917-18434-g4s3gc/simbody-Simbody-3.5.3/SimTKcommon/./include/SimTKcommon/internal/Timing.h:234:9: error: no matching function for call to 'clock_gettime'
        clock_gettime(CLOCK_MONOTONIC_RAW, &ts);
        ^~~~~~~~~~~~~
/tmp/simbody-20160917-18434-g4s3gc/simbody-Simbody-3.5.3/SimTKcommon/./include/SimTKcommon/internal/Timing.h:110:34: note: candidate function not viable: no known conversion from 'int' to 'clockid_t' for 1st argument
    SimTK_SimTKCOMMON_EXPORT int clock_gettime(clockid_t clock_id,
                                 ^
4 errors generated.
make[2]: *** [SimTKcommon/sharedTarget/CMakeFiles/SimTKcommon.dir/__/src/Timing.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [SimTKcommon/sharedTarget/CMakeFiles/SimTKcommon.dir/all] Error 2
make: *** [all] Error 2
```

I haven't looked into it closely but it looks like a type mismatch issue.

scpeters, 2016-09-20:
Can you paste your `brew config`?

bwithbroe, 2016-09-20:
```
brew config
HOMEBREW_VERSION: 0.9.9
ORIGIN: https://github.com/Homebrew/brew.git
HEAD: f6fef7cdce5302ae6f3fed357612d31d1d08cd5a
Last commit: 73 minutes ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: f58fdfd9c4d4060b5cb5cad45980b634e5f5c405
Core tap last commit: 4 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local/Homebrew
HOMEBREW_CELLAR: /usr/local/Cellar

HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: octa-core 64-bit haswell
Homebrew Ruby: 2.0.0-p648
Clang: 8.0 build 800
Git: 2.6.2 => /usr/local/bin/git
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.8.0_66, 1.6.0_65-b14-468
macOS: 10.11.6-x86_64
Xcode: 8.0
CLT: 7.3.1.0.1.1461711523
X11: 2.7.9 => /opt/X11
```

I have the same problem as OP - here's my `brew config` output

scpeters, 2016-09-20:
Ok, it looks like you're using macOS 10.11. I'm using 10.10 and am not able to reproduce this failure, so I'm guessing it's an issue with the OS version. Can you run the following commands to get a little bit more debugging information and upload it to a gist:

~~~
brew install simbody --verbose
brew gist-logs simbody
~~~

Then if you could paste the link to the gist here I'll take a look and maybe refer it to the simbody folks. Thanks!

bwithbroe, 2016-09-20:
Here's the gist:
https://gist.github.com/9247eaf889cdaf3e0a97f9df27682af6

It doesn't always fail with macOS 10.11 - two other students working with me installed successfully with that OS.

scpeters, 2016-09-20:
Here's the crux of the problem:

~~~
In file included from /tmp/simbody-20160920-28673-1o5ma3f/simbody-Simbody-3.5.3/SimTKcommon/src/Timing.cpp:33:
/tmp/simbody-20160920-28673-1o5ma3f/simbody-Simbody-3.5.3/
SimTKcommon/./include/SimTKcommon/internal/Timing.h:88:18:
error: typedef redefinition with different types ('long' vs 'enum clockid_t')
    typedef long clockid_t;
                 ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/
Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: 
note: previous definition is here
} clockid_t;
^
~~~

You appear to be using the macOS 10.12 SDK, which appears to define the `clockid_t` type. Maybe the other students were using the 10.11 SDK? In either case, I'll report this upstream to simbody, since I think this will cause compilation errors on 10.12.

Thanks for posting the gist so quickly!

bwithbroe, 2016-09-21:
Thanks for the quick responses/upstream report!

junhohan, 2016-09-22:
FYI, I got the same problem.
Here is my ```brew config``` result.

```
HOMEBREW_VERSION: 1.0.0
ORIGIN: https://github.com/Homebrew/brew
HEAD: a381b597d3ebf40887b50657df269d867a209eb7
Last commit: 23 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 7687c444bb39f937e0bced9d0ec2674412592b44
Core tap last commit: 33 minutes ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local/Homebrew
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: octa-core 64-bit haswell
Homebrew Ruby: 2.0.0-p648
Clang: 8.0 build 800
Git: 2.8.4 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Perl: /usr/bin/perl
Python: /usr/bin/python
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.8.0_45
macOS: 10.11.6-x86_64
Xcode: 8.0
CLT: 7.3.1.0.1.1461711523
X11: 2.7.9 => /opt/X11
```

scpeters, 2016-10-03:
This should be fixed by #140

closed 2016-10-03 by scpeters