osrf/homebrew-simulation #142: gazebo_rebdering for gazebo8 fails on homebrew/sierra

GlenHenshaw, 2016-10-03:
Build fails as 58% of gazebo_rendering. I'm on a Macbook Pro running macOS Sierra.

gist-logs link:

https://gist.github.com/2f63f0624eca75f5bd131e85c0715bac

Last failed command is:

`cd /tmp/gazebo8-20161003-53868-14vlrbv/osrf-gazebo-17137a79cbe8/build/gazebo/rendering && /usr/local/Cellar/cmake/3.6.2/bin/cmake -E cmake_link_script CMakeFiles/gazebo_rendering.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -std=c++11 -stdlib=libc++ -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11 -DBUILDING_DLL_GZ_RENDERING -DSKYX_LIB   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -dynamiclib -Wl,-headerpad_max_install_names   -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 8.0.0 -current_version 8.0.0 -o libgazebo_rendering.8.0.0.dylib -install_name /tmp/gazebo8-20161003-53868-14vlrbv/osrf-gazebo-17137a79cbe8/build/gazebo/rendering/libgazebo_rendering.8.dylib CMakeFiles/gazebo_rendering.dir/ApplyWrenchVisual.cc.o CMakeFiles/gazebo_rendering.dir/ArrowVisual.cc.o CMakeFiles/gazebo_rendering.dir/AxisVisual.cc.o CMakeFiles/gazebo_rendering.dir/Camera.cc.o CMakeFiles/gazebo_rendering.dir/CameraVisual.cc.o CMakeFiles/gazebo_rendering.dir/COMVisual.cc.o CMakeFiles/gazebo_rendering.dir/ContactVisual.cc.o CMakeFiles/gazebo_rendering.dir/Conversions.cc.o CMakeFiles/gazebo_rendering.dir/DepthCamera.cc.o CMakeFiles/gazebo_rendering.dir/Distortion.cc.o CMakeFiles/gazebo_rendering.dir/DynamicLines.cc.o CMakeFiles/gazebo_rendering.dir/DynamicRenderable.cc.o CMakeFiles/gazebo_rendering.dir/FPSViewController.cc.o CMakeFiles/gazebo_rendering.dir/GpuLaser.cc.o CMakeFiles/gazebo_rendering.dir/Grid.cc.o CMakeFiles/gazebo_rendering.dir/Heightmap.cc.o CMakeFiles/gazebo_rendering.dir/InertiaVisual.cc.o CMakeFiles/gazebo_rendering.dir/JointVisual.cc.o CMakeFiles/gazebo_rendering.dir/LaserVisual.cc.o CMakeFiles/gazebo_rendering.dir/LinkFrameVisual.cc.o CMakeFiles/gazebo_rendering.dir/SonarVisual.cc.o CMakeFiles/gazebo_rendering.dir/Light.cc.o CMakeFiles/gazebo_rendering.dir/LogicalCameraVisual.cc.o CMakeFiles/gazebo_rendering.dir/Material.cc.o CMakeFiles/gazebo_rendering.dir/MovableText.cc.o CMakeFiles/gazebo_rendering.dir/OrbitViewController.cc.o CMakeFiles/gazebo_rendering.dir/OriginVisual.cc.o CMakeFiles/gazebo_rendering.dir/OrthoViewController.cc.o CMakeFiles/gazebo_rendering.dir/Projector.cc.o CMakeFiles/gazebo_rendering.dir/RayQuery.cc.o CMakeFiles/gazebo_rendering.dir/RenderEngine.cc.o CMakeFiles/gazebo_rendering.dir/RenderEvents.cc.o CMakeFiles/gazebo_rendering.dir/RenderingIface.cc.o CMakeFiles/gazebo_rendering.dir/Road2d.cc.o CMakeFiles/gazebo_rendering.dir/RFIDVisual.cc.o CMakeFiles/gazebo_rendering.dir/RFIDTagVisual.cc.o CMakeFiles/gazebo_rendering.dir/RTShaderSystem.cc.o CMakeFiles/gazebo_rendering.dir/Scene.cc.o CMakeFiles/gazebo_rendering.dir/SelectionObj.cc.o CMakeFiles/gazebo_rendering.dir/TransmitterVisual.cc.o CMakeFiles/gazebo_rendering.dir/UserCamera.cc.o CMakeFiles/gazebo_rendering.dir/VideoVisual.cc.o CMakeFiles/gazebo_rendering.dir/ViewController.cc.o CMakeFiles/gazebo_rendering.dir/Visual.cc.o CMakeFiles/gazebo_rendering.dir/WideAngleCamera.cc.o CMakeFiles/gazebo_rendering.dir/WireBox.cc.o CMakeFiles/gazebo_rendering.dir/WindowManager.cc.o CMakeFiles/gazebo_rendering.dir/WrenchVisual.cc.o CMakeFiles/gazebo_rendering.dir/selection_buffer/MaterialSwitcher.cc.o CMakeFiles/gazebo_rendering.dir/selection_buffer/SelectionBuffer.cc.o CMakeFiles/gazebo_rendering.dir/selection_buffer/SelectionRenderListener.cc.o CMakeFiles/gazebo_rendering.dir/skyx/src/AtmosphereManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/BasicController.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/CloudsManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/ColorGradient.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/GPUManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/MeshManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/MoonManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/SkyX.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VCloudsManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/DataManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/Ellipsoid.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/FastFakeRandom.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/GeometryBlock.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/GeometryManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/Lightning.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/LightningManager.cpp.o CMakeFiles/gazebo_rendering.dir/skyx/src/VClouds/VClouds.cpp.o  -L/tmp/gazebo8-20161003-53868-14vlrbv/osrf-gazebo-17137a79cbe8/build/test  -L/usr/local/Cellar/ignition-transport/1.3.0_1/lib  -L/usr/local/Cellar/zeromq/4.1.5/lib  -L/usr/local/Cellar/ossp-uuid/1.6.2_2/lib  -L/usr/local/Cellar/ogre/1.7.4/lib  -L/usr/local/Cellar/ignition-math2/2.5.0/lib  -L/usr/local/Cellar/bullet/2.83.7/lib  -L/usr/local/opt/tinyxml/lib ../transport/libgazebo_transport.8.0.0.dylib -lOgreRTShaderSystem -lOgreMain -lpthread -lOgreMain -lpthread -lOgreTerrain -lOgrePaging -lOgreMain -lpthread -framework OpenGL -ltinyxml ../msgs/libgazebo_msgs.8.0.0.dylib ../common/libgazebo_common.8.0.0.dylib ../math/libgazebo_math.8.0.0.dylib /usr/lib/libdl.dylib /usr/local/lib/libfreeimage.dylib -lcurl /usr/local/lib/libtar.dylib /usr/local/Cellar/sdformat4/4.1.1_1/lib/libsdformat.dylib -framework OpenAL /usr/local/lib/libprotobuf.dylib /usr/local/Cellar/ignition-math2/2.5.0/lib/libignition-math2.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib /usr/local/lib/libtbb.dylib -lOgreTerrain -lOgrePaging -lpthread -framework OpenGL -ltinyxml -Wl,-rpath,/tmp/gazebo8-20161003-53868-14vlrbv/osrf-gazebo-17137a79cbe8/build/test -Wl,-rpath,/usr/local/Cellar/ignition-transport/1.3.0_1/lib -Wl,-rpath,/usr/local/Cellar/zeromq/4.1.5/lib -Wl,-rpath,/usr/local/Cellar/ossp-uuid/1.6.2_2/lib -Wl,-rpath,/usr/local/Cellar/ogre/1.7.4/lib -Wl,-rpath,/usr/local/Cellar/ignition-math2/2.5.0/lib -Wl,-rpath,/usr/local/Cellar/bullet/2.83.7/lib -Wl,-rpath,/usr/local/opt/tinyxml/lib 
cd /tmp/gazebo8-20161003-53868-14vlrbv/osrf-gazebo-17137a79cbe8/build/gazebo/rendering && /usr/local/Cellar/cmake/3.6.2/bin/cmake -E cmake_symlink_library libgazebo_rendering.8.0.0.dylib libgazebo_rendering.8.dylib libgazebo_rendering.dylib
[ 58%] Built target gazebo_rendering
make: *** [all] Error 2
`

scpeters, 2016-10-03:
Thanks for the report. This is related to #135, which should be fixed by #140. I'll expedite review of that pull request.

scpeters, 2016-10-03:
Can you try again? I think it is fixed now.

GlenHenshaw, 2016-10-04:
Works. Thanks!

closed 2016-10-04 by GlenHenshaw