osrf/homebrew-simulation #149: gazebo build error on Mac 10.11.6

studiocardo, 2016-10-07:
Hi

Tried to build gazebo with the following command on my MBP running 10.11.6.  I ran "brew update" twice prior to the build and still getting this message.  

Any advice on how to proceed is much appreciated!!!

$ curl -ssL http://get.gazebosim.org | sh
=======================================================
GAZEBO INSTALLATION SCRIPT
=======================================================

This script is installing the latest stable version of
Gazebo Simulator available from your package manager

package-id: org.macosforge.xquartz.pkg
version: 2.7.94
volume: /
location:
install-time: 1474091503
Already up-to-date.
==> Installing gazebo7 from osrf/simulation
==> Installing dependencies for osrf/simulation/gazebo7: protobuf-c, bullet, simbody
==> Installing osrf/simulation/gazebo7 dependency: protobuf-c
==> Downloading https://homebrew.bintray.com/bottles/protobuf-c-1.2.1_2.el_capitan.bottle.tar.gz
Already downloaded: /Users/howard/Library/Caches/Homebrew/protobuf-c-1.2.1_2.el_capitan.bottle.tar.gz
==> Pouring protobuf-c-1.2.1_2.el_capitan.bottle.tar.gz
🍺  /usr/local/Cellar/protobuf-c/1.2.1_2: 12 files, 287.2K
==> Installing osrf/simulation/gazebo7 dependency: bullet
==> Using the sandbox
==> Downloading https://github.com/bulletphysics/bullet3/archive/2.83.7.tar.gz
Already downloaded: /Users/howard/Library/Caches/Homebrew/bullet-2.83.7.tar.gz
==> cmake -DINSTALL_EXTRA_LIBS=ON -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/bullet/2.83.7 -DUSE_DOUBLE_PRECISION=ON -DBUILD_BULLET2_DEMOS=OFF
==> make
==> make install
🍺  /usr/local/Cellar/bullet/2.83.7: 456 files, 6.5M, built in 30 seconds
==> Installing osrf/simulation/gazebo7 dependency: simbody
==> Downloading https://github.com/simbody/simbody/archive/Simbody-3.5.4.tar.gz
Already downloaded: /Users/howard/Library/Caches/Homebrew/simbody-3.5.4.tar.gz
==> Downloading https://github.com/scpeters/simbody/commit/a37deef08af530b57601251900fadee0d0be6cfd.diff
Already downloaded: /Users/howard/Library/Caches/Homebrew/simbody--patch-0b99f7b6df46ddcc43094b29ba834896490fe5a76ceaa15f646b3b4579fed51e.diff
==> Patching
==> Applying a37deef08af530b57601251900fadee0d0be6cfd.diff
patching file cmake/pkgconfig/simbody.pc.in
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/simbody/3.5.4 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LA
==> make doxygen
==> make install
🍺  /usr/local/Cellar/simbody/3.5.4: 3,847 files, 118.7M, built in 4 minutes 31 seconds
==> Installing osrf/simulation/gazebo7
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-7.3.1.tar.bz2
Already downloaded: /Users/howard/Library/Caches/Homebrew/gazebo7-7.3.1.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/howard/Library/Caches/Homebrew/gazebo7--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
==> Downloading https://bitbucket.org/osrf/gazebo/commits/90b82ee6efe79d83dc9fa4f9706e050e658d1221/raw/
Already downloaded: /Users/howard/Library/Caches/Homebrew/gazebo7--patch-73f3724ce0123cd71968b8b03822d45a3b893ca40f746feed76870c9b29e7603
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> Applying raw
patching file gazebo/util/LogPlay.cc
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo7/7.3.1_5 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=
==> make install
Last 15 lines from /Users/howard/Library/Logs/Homebrew/gazebo7/02.make:
cd /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build/gazebo/physics && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DBT_USE_DOUBLE_PRECISION -DHAVE_OPENGL -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_EXPORTS -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/test/gtest/include -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1 -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build -isystem /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/deps/opende/include -isystem /usr/local/include -I/usr/local/Cellar/tinyxml2/4.0.1/include -isystem /usr/local/Cellar/ignition-math2/2.5.0/include/ignition/math2 -iframework /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Paging -isystem /usr/local/Cellar/sdformat4/4.1.1_1/include/sdformat-4.1 -isystem /usr/local/opt/tinyxml/include -I/usr/local/Cellar/bullet/2.83.7/include/bullet -I/usr/local/Cellar/bullet/2.83.7/include -isystem /usr/local/Cellar/simbody/3.5.4/include/simbody  -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11 -fPIC   -fPIC -I/usr/local/Cellar/bullet/2.83.7/include/bullet -I/usr/local/Cellar/bullet/2.83.7/include -I"/usr/local/Cellar/simbody/3.5.4/include/simbody" -o CMakeFiles/gazebo_physics.dir/State.cc.o -c /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/gazebo/physics/State.cc
[ 57%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/SurfaceParams.cc.o
cd /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build/gazebo/physics && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DBT_USE_DOUBLE_PRECISION -DHAVE_OPENGL -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_EXPORTS -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/test/gtest/include -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1 -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build -isystem /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/deps/opende/include -isystem /usr/local/include -I/usr/local/Cellar/tinyxml2/4.0.1/include -isystem /usr/local/Cellar/ignition-math2/2.5.0/include/ignition/math2 -iframework /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Paging -isystem /usr/local/Cellar/sdformat4/4.1.1_1/include/sdformat-4.1 -isystem /usr/local/opt/tinyxml/include -I/usr/local/Cellar/bullet/2.83.7/include/bullet -I/usr/local/Cellar/bullet/2.83.7/include -isystem /usr/local/Cellar/simbody/3.5.4/include/simbody  -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11 -fPIC   -fPIC -I/usr/local/Cellar/bullet/2.83.7/include/bullet -I/usr/local/Cellar/bullet/2.83.7/include -I"/usr/local/Cellar/simbody/3.5.4/include/simbody" -o CMakeFiles/gazebo_physics.dir/SurfaceParams.cc.o -c /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/gazebo/physics/SurfaceParams.cc
[ 57%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/UserCmdManager.cc.o
cd /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build/gazebo/physics && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DBT_USE_DOUBLE_PRECISION -DHAVE_OPENGL -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_EXPORTS -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/test/gtest/include -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1 -I/tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build -isystem /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/deps/opende/include -isystem /usr/local/include -I/usr/local/Cellar/tinyxml2/4.0.1 [... cut ...] deps/libccd -Wl,-rpath,/usr/local/Cellar/simbody/3.5.4/lib
cd /tmp/gazebo7-20161007-15136-1qjgxus/gazebo-7.3.1/build/gazebo/physics && /usr/local/Cellar/cmake/3.6.2/bin/cmake -E cmake_symlink_library libgazebo_physics.7.3.1.dylib libgazebo_physics.7.dylib libgazebo_physics.dylib
[ 57%] Built target gazebo_physics
make: *** [all] Error 2

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/osrf/homebrew-simulation/issues

scpeters, 2016-10-07:
Thanks for the report; we just noticed this as well. I believe it is related to a new version of boost that was just released. I think it may be fixed by the patches in #148

tarasrudyy, 2016-10-10:
Hi,

I've tried patched version, same error still.

Regards

scpeters, 2016-10-10:
@tarasrudyy can you build with the `--verbose` flag and then post the gist logs?

~~~
brew install --verbose gazebo7
brew gist-logs gazebo7
~~~

tarasrudyy, 2016-10-10:
@scpeters sure

https://gist.github.com/fdbcee56e7ba08559b71ef5be23feeca

scpeters, 2016-10-10:
It's having a problem finding qt headers:

~~~
/tmp/gazebo7-20161010-1475-8bfz7z/gazebo-7.3.1/gazebo/rendering/RenderEngine.cc:30:11: fatal error: 'QtCore/qglobal.h' file not found
# include <QtCore/qglobal.h>
          ^
~~~

What do you see in the following folder:

~~~
/usr/local/Cellar/qt/4.8.7_2/lib/QtCore.framework/Headers
~~~

I'm guessing that is where it wants to find `QtCore/qglobal.h`

tarasrudyy, 2016-10-10:
It's empty. I've just tried to `brew uninstall qt` and `brew install qt`

> qt: This formula either does not compile or function as expected on macOS
versions newer than El Capitan due to an upstream incompatibility.
Error: An unsatisfied requirement failed this build.

scpeters, 2016-10-10:
Oh, I didn't realize that qt4 is not supported on macOS 10.12. That is a very big problem since gazebo currently needs qt4. Sorry that you're the one who had to discover it.

We are currently working on supporting qt5 in gazebo. It currently compiles but does run properly on homebrew (works on Ubuntu though). I'll try to keep you updated on this status. In the meantime, I'm going to recommend that people wait to upgrade to macOS 10.12 if they want to use gazebo.

scpeters, 2016-10-10:
In the meantime, there is a tap that may work for you (from https://github.com/Homebrew/homebrew-core/pull/5216#issuecomment-251478385 )

~~~
brew install cartr/qt4/qt
~~~

tarasrudyy, 2016-10-10:
Thank you Steven! I'm looking forward for Gazebo 8. If no luck with macOS then good old Ubuntu :)

tarasrudyy, 2016-10-10:
`brew install cartr/qt4/qt` works fine for me

closed 2016-12-09 by j-rivero