osrf/homebrew-simulation #158: error: Built target gazebo_physics_ode on OS X EL Capitan

xiaobin-yang, 2016-11-21:
patching file gazebo/gui/TimePanel.cc
Hunk #1 succeeded at 118 (offset -60 lines).
patching file gazebo/gui/TimePanel.hh
Hunk #1 succeeded at 62 (offset -4 lines).
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo1/1.9.7 -DCMAKE_BUILD_TYPE=Rel
==> make install
Last 15 lines from /Users/xyang/Library/Logs/Homebrew/gazebo1/02.make:
[ 45%] Building CXX object gazebo/physics/ode/CMakeFiles/gazebo_physics_ode.dir/ODEHeightmapShape.cc.o
cd /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_ode_EXPORTS -I/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/test/gtest/include -I/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7 -I/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build -isystem /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/deps/opende/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat/2.3.2_1/include/sdformat-2.3  -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -fPIC   -fPIC -o CMakeFiles/gazebo_physics_ode.dir/ODEHeightmapShape.cc.o -c /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/gazebo/physics/ode/ODEHeightmapShape.cc
[ 45%] Building CXX object gazebo/physics/ode/CMakeFiles/gazebo_physics_ode.dir/ODEScrewJoint.cc.o
cd /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_ode_EXPORTS -I/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/test/gtest/include -I/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7 -I/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build -isystem /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/deps/opende/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat/2.3.2_1/include/sdformat-2.3  -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -fPIC   -fPIC -o CMakeFiles/gazebo_physics_ode.dir/ODEScrewJoint.cc.o -c /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/gazebo/physics/ode/ODEScrewJoint.cc
[ 45%] Linking CXX shared library libgazebo_transport.dylib
cd /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/transport && /usr/local/Cellar/cmake/3.7.0/bin/cmake -E cmake_link_script CMakeFiles/gazebo_transport.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++    -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -dynamiclib -Wl,-headerpad_max_install_names   -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 1.0.0 -current_version 1.9.7 -o libgazebo_transport.1.9.7.dylib -install_name /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/transport/libgazebo_transport.1.dylib CMakeFiles/gazebo_transport.dir/CallbackHelper.cc.o CMakeFiles/gazebo_transport.dir/Connection.cc.o CMakeFiles/gazebo_transport.dir/ConnectionManager.cc.o CMakeFiles/gazebo_transport.dir/IOManager.cc.o CMakeFiles/gazebo_transport.dir/Node.cc.o CMakeFiles/gazebo_transport.dir/Publication.cc.o CMakeFiles/gazebo_transport.dir/PublicationTransport.cc.o CMakeFiles/gazebo_transport.dir/Publisher.cc.o CMakeFiles/gazebo_transport.dir/Subscriber.cc.o CMakeFiles/gazebo_transport.dir/SubscriptionTransport.cc.o CMakeFiles/gazebo_transport.dir/TopicManager.cc.o CMakeFiles/gazebo_transport.dir/TransportIface.cc.o  -L/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/test  -L/usr/local/Cellar/ogre/1.7.4/lib  -L/usr/local/Cellar/sdformat/2.3.2_1/lib -Wl,-rpath,/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/test -Wl,-rpath,/usr/local/Cellar/ogre/1.7.4/lib -Wl,-rpath,/usr/local/Cellar/sdformat/2.3.2_1/lib ../msgs/libgazebo_msgs.1.9.7.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib /usr/local/lib/libtbb.dylib ../common/libgazebo_common.1.9.7.dylib /usr/lib/libdl.dylib /usr/local/lib/libltdl.dylib -lfreeimage -ltinyxml -lcurl -ltar /usr/local/lib/libtbb.dylib -lpthread -lsdformat ../math/libgazebo_math.1.9.7.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib /usr/local/lib/libprotobuf.dylib 
cd /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/transport && /usr/local/Cellar/cmake/3.7.0/bin/cmake -E cmake_symlink_library libgazebo_transport.1.9.7.dylib libgazebo_transport.1.dylib libgazebo_transport.dylib
[ 45%] Built target gazebo_transport
[ 45%] Linking CXX shared library libgazebo_physics_ode.dylib
cd /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Cellar/cmake/3.7.0/bin/cmake -E cmake_link_script CMakeFiles/gazebo_physics_ode.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++    -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -dynamiclib -Wl,-headerpad_max_install_names   -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 1.0.0 -current_version 1.9.7 -o libgazebo_physics_ode.1.9.7.dylib -install_name /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/physics/ode/libgazebo_physics_ode.1.dylib CMakeFiles/gazebo_physics_ode.dir/ODEPhysics.cc.o CMakeFiles/gazebo_physics_ode.dir/ODECollision.cc.o CMakeFiles/gazebo_physics_ode.dir/ODELink.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODESliderJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEHingeJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEHinge2Joint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEUniversalJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEBallJoint.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEMeshShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODERayShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEMultiRayShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEHeightmapShape.cc.o CMakeFiles/gazebo_physics_ode.dir/ODEScrewJoint.cc.o  -L/tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/test  [... cut ...] cal/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib -lfreeimage -ltinyxml -lcurl -ltar /usr/local/lib/libtbb.dylib -lpthread -lsdformat 
cd /tmp/gazebo1-20161121-92606-1c36xv7/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Cellar/cmake/3.7.0/bin/cmake -E cmake_symlink_library libgazebo_physics_ode.1.9.7.dylib libgazebo_physics_ode.1.dylib libgazebo_physics_ode.dylib
[ 45%] Built target gazebo_physics_ode
make: *** [all] Error 2

j-rivero, 2016-12-09:
There is no error detail in that log, probably is located some lines above. Please attach the compilation error exactly where it happen.

Thanks.

scpeters, 2016-12-10:
You can use the `--verbose` option to `brew install` to see the whole compilation log. You can also use `brew gist-logs gazebo` to upload all the logs, then just post a link here.

christoferjennings, 2017-01-23:
I'm trying to get the FRCSim running with macOS Sierra 10.12.2
I have what seems to be the same problem. Here's my console output.
Thanks!

[gazebo-install.txt](https://github.com/osrf/homebrew-simulation/files/722628/gazebo-install.txt)

j-rivero, 2017-01-23:
Ok Thanks. Looks like this is the problem:

```
[ 71%] Building CXX object gazebo/gui/CMakeFiles/gazebo_gui.dir/Actions.cc.o
cd "/tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/build/gazebo/gui" && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DBUILDING_DLL -DBUILDING_DLL_GZ_GUI -DHAVE_OPENGL -DLIBBULLET_VERSION=2.85 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQWT_DLL -Dgazebo_gui_EXPORTS -I"/tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/test/gtest/include" -I"/tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1" -I"/tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/build" -I"/tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/deps" -isystem /usr/local/include -isystem /usr/local/Cellar/tinyxml2/4.0.1/include -isystem /usr/local/include/ignition/msgs0 -isystem /usr/local/Cellar/ignition-math2/2.7.0/include/ignition/math2 -I/usr/local/Cellar/ignition-transport3/3.0.1/include/ignition/transport3 -I/usr/local/Cellar/zeromq/4.2.1/include -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -iframework /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_2/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_2/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_2/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_2/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_2/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_2/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat4/4.2.0_3/include/sdformat-4.2 -isystem /usr/local/opt/tinyxml/include -iframework /usr/local/opt/qt5/lib -isystem /usr/local/opt/qt5/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt5/./mkspecs/macx-clang -I/usr/local/lib/qwt.framework/Headers -isystem /usr/local/opt/qt5/lib/QtWidgets.framework/Headers -isystem /usr/local/opt/qt5/lib/QtGui.framework/Headers -isystem /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers  -std=c++11 -stdlib=libc++ -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -fPIC -F/usr/local/lib  -std=c++11 -fPIC -std=gnu++11 -o CMakeFiles/gazebo_gui.dir/Actions.cc.o -c "/tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/gazebo/gui/Actions.cc"
In file included from /tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/gazebo/gui/Actions.cc:18:
In file included from /tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/gazebo/gui/Actions.hh:22:
In file included from /tmp/gazebo8-20170122-49455-19cwgg6/gazebo-8.0.0~beta1/gazebo/gui/qt.h:36:
In file included from /usr/local/opt/qt5/include/QtGui/QWindow:1:
In file included from /usr/local/opt/qt5/lib/QtGui.framework/Headers/qwindow.h:50:
In file included from /usr/local/opt/qt5/include/QtGui/qsurface.h:44:
/usr/local/opt/qt5/include/QtGui/qsurfaceformat.h:129:24: error: token is not a valid binary operator in a preprocessor subexpression
#if QT_DEPRECATED_SINCE(5, 2)
    ~~~~~~~~~~~~~~~~~~~^
```

By any chance do you have qt4 and qt5 installed in the same system? Could you attach the versions of: Xcode, Qt5  that you are using and the output of qmake -query? Thanks.

christoferjennings, 2017-01-24:
I just so happen to be in the middle of updating Xcode. It was updated just before I tried to install gazebo, though. So it was the latest as ofJanuary 22, 2017.

```
» brew info qt5
qt5: stable 5.7.1 (bottled), HEAD [keg-only]
Cross-platform application and UI framework
https://www.qt.io/
/usr/local/Cellar/qt5/5.7.1_1 (8,273 files, 242.5M)
  Poured from bottle on 2017-01-22 at 13:42:46
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt5.rb
```

```
» qmake -query
QT_INSTALL_PREFIX:/usr/local/Cellar/qt/4.8.7_2
QT_INSTALL_DATA:/usr/local/Cellar/qt/4.8.7_2
QT_INSTALL_DOCS:/usr/local/Cellar/qt/4.8.7_2/doc
QT_INSTALL_HEADERS:/usr/local/Cellar/qt/4.8.7_2/include
QT_INSTALL_LIBS:/usr/local/Cellar/qt/4.8.7_2/lib
QT_INSTALL_BINS:/usr/local/Cellar/qt/4.8.7_2/bin
QT_INSTALL_PLUGINS:/usr/local/Cellar/qt/4.8.7_2/plugins
QT_INSTALL_IMPORTS:/usr/local/Cellar/qt/4.8.7_2/imports
QT_INSTALL_TRANSLATIONS:/usr/local/Cellar/qt/4.8.7_2/translations
QT_INSTALL_CONFIGURATION:/Library/Preferences/Qt
QT_INSTALL_EXAMPLES:/usr/local/Cellar/qt/4.8.7_2/examples
QT_INSTALL_DEMOS:/usr/local/Cellar/qt/4.8.7_2/demos
QMAKE_MKSPECS:/usr/local/Cellar/qt/4.8.7_2/mkspecs
QMAKE_VERSION:2.01a
QT_VERSION:4.8.7

```

andromeda-gal63, 2017-08-13:
Is it safe to replace qt4 with qt5 with the frcsim-gazebo-models-4, or will there be larger code impact? Is there a nifty online command to do this for FRCSIM on Mac OSX? (I am not a ruby, perl, curl or brew guru). Thanks!

scpeters, 2017-08-14:
Is there a formula for `frcsim` that you are using?

scpeters, 2017-09-15:
I'm closing this since I haven't heard any responses. Please re-open if the issue persists.

closed 2017-09-15 by scpeters