osrf/homebrew-simulation #160: Cannot build for MacOS Sierra (10.12.1)

chenny408, 2016-11-25:
The makefile produces an error when during installation.  Is there a way to install gazebo for this mac version?

brew install gazebo7 

==> Installing gazebo7 from osrf/simulation
qt: This formula either does not compile or function as expected on macOS
versions newer than El Capitan due to an upstream incompatibility.
Error: An unsatisfied requirement failed this build.

scpeters, 2016-11-26:
We are currently migrating gazebo8 to use qt5, so that version will be work for MacOS Sierra, but older versions of gazebo (7 and under) won't work on MacOS Sierra. Gazebo8 is in development in the `qt5` branch. See [gazebo issue 1804](https://bitbucket.org/osrf/gazebo/issues/1804/migrate-to-qt5) as well for more info.

chenny408, 2016-11-26:
ok got it, when's the planned release date for Gazebo8?

scpeters, 2016-11-27:
release scheduled for Jan 25 2017: http://gazebosim.org/#status

starthal, 2017-02-09:
Refer to PR #198 for Gazebo 7 installation on Sierra.

scpeters, 2017-07-05:
gazebo7 and gazebo8 now have bottles on sierra

closed 2017-07-05 by scpeters