osrf/homebrew-simulation #162: gazebo2 build error on El Capitan

bhaskara, 2016-11-30:
I get a compilation error on doing 'brew install gazebo2'.  I believe the relevant section is:

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f gazebo/physics/CMakeFiles/gazebo_physics.dir/build.make gazebo/physics/CMakeFiles/gazebo_physics.dir/depend
cd /tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6/build && /usr/local/Cellar/cmake/3.7.0/bin/cmake -E cmake_depends "Unix Makefiles" /tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6 /tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6/gazebo/physics /tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6/build /tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6/build/gazebo/physics /tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6/build/gazebo/physics/CMakeFiles/gazebo_physics.dir/DependInfo.cmake --color=
/tmp/gazebo2-20161130-36028-1o41ale/gazebo-2.2.6/gazebo/rendering/Camera.cc:976:30: error: no member named 'format' in namespace 'boost'
    pathToFile /= str(boost::format("%s-%04d.jpg")
```

I'm guessing `Camera.cc` might be relying on `boost/filesystem.hpp` including `format.hpp`, and this is no longer true?

scpeters, 2016-11-30:
Something changed in boost 1.62, and we fixed it upstream ([gazebo PR 2463](https://bitbucket.org/osrf/gazebo/pull-requests/2463/add-boost-formathhpp-header/diff)) but haven't made new releases of gazebo2 in a while. I've added a patch in https://github.com/scpeters/homebrew-simulation/commit/a856e905311e241a9d5f37b6d55c80a276aff961; use the following to try it:

~~~
brew install https://github.com/scpeters/homebrew-simulation/blob/a856e905311e241a9d5f37b6d55c80a276aff961/gazebo2.rb
~~~

I'll make a pull request once it passes the travis CI build.

bhaskara, 2016-12-01:
Thanks for the quick response!  The patch above didn't work for me though:

```
Updating Homebrew...
==> Auto-updated Homebrew!
Updated Homebrew from 0f529da to eb20f59.
Updated 3 taps (homebrew/core, homebrew/science, osrf/simulation).
==> Updated Formulae
ansifilter                 mongodb                    osrf/simulation/gazebo6
homebrew/science/infernal  offlineimap                osrf/simulation/gazebo7 ✔
lean-cli                   osrf/simulation/gazebo5

######################################################################## 100.0%
Error: /Users/tempsetup/Library/Caches/Homebrew/Formula/gazebo2.rb:5: syntax error, unexpected '<'
<!DOCTYPE html>
 ^
/Users/tempsetup/Library/Caches/Homebrew/Formula/gazebo2.rb:6: syntax error, unexpected '<'
<html lang="en" class="">
 ^
/Users/tempsetup/Library/Caches/Homebrew/Formula/gazebo2.rb:6: syntax error, unexpected keyword_class, expecting end-of-input
<html lang="en" class="">
                     ^
Please report this bug:
  https://git.io/brew-troubleshooting
/usr/local/Homebrew/Library/Homebrew/formulary.rb:25:in `module_eval'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:25:in `load_formula'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:42:in `load_formula_from_path'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:97:in `load_file'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:153:in `load_file'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:88:in `klass'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:84:in `get_formula'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:241:in `factory'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:43:in `block in formulae'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:41:in `map'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:41:in `formulae'
/usr/local/Homebrew/Library/Homebrew/cmd/install.rb:107:in `install'
/usr/local/Homebrew/Library/Homebrew/brew.rb:94:in `<main>'
Temps-MacBook-Pro:robotics-osx tempsetup$ 
Temps-MacBook-Pro:robotics-osx tempsetup$ 

```

scpeters, 2016-12-01:
sorry, that was the wrong link:

~~~
brew install https://raw.githubusercontent.com/scpeters/homebrew-simulation/a856e905311e241a9d5f37b6d55c80a276aff961/gazebo2.rb
~~~

I've edited the original comment as well so they should both work.

bhaskara, 2016-12-01:
Hmm, now I get:
```

Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core, homebrew/science).
==> Updated Formulae
camlp4          cmake ✔         coq             flow            hunspell        ledit           nifi            ocamlbuild      offlineimap     ssreflect       twarc           zero-install
camlp5          collectd        coreutils       fstar           lablgtk         mkvtoolnix      ocaml           ocamlsdl        sourcekitten    texmath         youtube-dl      zstd

######################################################################## 100.0%
==> Installing dependencies for gazebo2: cmake
==> Installing gazebo2 dependency: cmake
==> Downloading https://homebrew.bintray.com/bottles/cmake-3.7.1.el_capitan.bottle.tar.gz
######################################################################## 100.0%
==> Pouring cmake-3.7.1.el_capitan.bottle.tar.gz
==> Caveats
Emacs Lisp files have been installed to:
  /usr/local/share/emacs/site-lisp/cmake
==> Summary
🍺  /usr/local/Cellar/cmake/3.7.1: 2,143 files, 29M
==> Installing gazebo2
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-2.2.6.tar.bz2
Already downloaded: /Users/tempsetup/Library/Caches/Homebrew/gazebo2-2.2.6.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/tempsetup/Library/Caches/Homebrew/gazebo2--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
==> Downloading https://bitbucket.org/osrf/gazebo/commits/91f6f3c59f40af34855548c37857955d08fd1368/raw/
Already downloaded: /Users/tempsetup/Library/Caches/Homebrew/gazebo2--patch-1a8b232be58f36bf5fa0129169f4d4d40d72624b460735457c781ba3e02c7900
==> Downloading https://bitbucket.org/osrf/gazebo/commits/9c5ce8a121904cf3373502320510ee74bc84b01d/raw/
######################################################################## 100.0%
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> Applying raw
patching file gazebo/gui/TimePanel.cc
Hunk #1 succeeded at 118 (offset -60 lines).
patching file gazebo/gui/TimePanel.hh
Hunk #1 succeeded at 62 (offset -4 lines).
==> Applying raw
patching file gazebo/rendering/Camera.cc
Hunk #1 FAILED at 20.
1 out of 1 hunk FAILED -- saving rejects to file gazebo/rendering/Camera.cc.rej
Error: Failure while executing: /usr/bin/patch -g 0 -f -p1 -i /private/tmp/gazebo2--patch-20161130-41797-1ajkwre/raw
```

Unfortunately I'm not able to find that .rej file, so it's hard to tell what's going wrong.

scpeters, 2016-12-01:
Ok, that patch didn't apply cleanly. I'll take another crack at it. Thanks for testing.

scpeters, 2016-12-02:
Maybe this will work now?

~~~
brew install https://raw.githubusercontent.com/scpeters/homebrew-simulation/a856e905311e241a9d5f37b6d55c80a276aff961/gazebo2.rb
~~~

bhaskara, 2016-12-02:
No, it still doesn't like that patch...

scpeters, 2016-12-02:
~~~
brew install --verbose --debug https://raw.githubusercontent.com/scpeters/homebrew-simulation/13c12df235dcbce658871106090f04aa1fcb47c6/gazebo2.rb
~~~

bhaskara, 2016-12-02:
OK, that worked.  Thanks!

scpeters, 2016-12-02:
Per #163, `brew install gazebo2` should now work

commit b00b161f (2016-12-02): gazebo2: patch for compatibility with boost 1.62 (#163)

* gazebo2: patch for compatibility with boost 1.62

* use better patch

Closes #162
closed 2016-12-02 by scpeters