osrf/homebrew-simulation #165: gazebo (from default) fails to compile on ElCapitan(10.11)/Xcode 8.1. os/availability.h not found.

j-rivero, 2016-12-09:
A gazebo (manual) compilation from current default branch fails on my ElCapitan machine with Xcode 8.1 since it can not find `os/availability.h`.

The file is present at `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/os/availability.h` but it is not in '/usr/include/os'.

```
[ 63%] Building CXX object gazebo/sensors/CMakeFiles/gazebo_sensors.dir/CameraSensor.cc.o
cd /Users/jrivero/code/gazebo/build/gazebo/sensors && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DBUILDING_DLL -DHAVE_OPENGL -DLIBBULLET_VERSION=2.83 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_sensors_EXPORTS -I/Users/jrivero/code/gazebo/test/gtest/include -I/Users/jrivero/code/gazebo -I/Users/jrivero/code/gazebo/build -isystem /usr/local/include -isystem /usr/local/Cellar/tinyxml2/4.0.1/include -I/usr/local/Cellar/libusb/1.0.20/include/libusb-1.0 -isystem /usr/local/include/ignition/msgs0 -isystem /usr/local/Cellar/ignition-math2/2.6.0/include/ignition/math2 -I/usr/local/Cellar/ignition-transport/1.4.0/include/ignition/transport1 -I/usr/local/Cellar/zeromq/4.1.6/include -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/Paging -isystem /usr/local/Cellar/sdformat4/4.2.0/include/sdformat-4.2 -isystem /usr/local/opt/tinyxml/include  -std=c++11 -stdlib=libc++ -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11  -g -O2 -mssse3 -msse3 -msse -msse2  -fPIC   -fPIC -F/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks  -std=c++11 -o CMakeFiles/gazebo_sensors.dir/CameraSensor.cc.o -c /Users/jrivero/code/gazebo/gazebo/sensors/CameraSensor.cc
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f gazebo/gui/CMakeFiles/gazebo_gui.dir/build.make gazebo/gui/CMakeFiles/gazebo_gui.dir/build
In file included from /Users/jrivero/code/gazebo/gazebo/sensors/CameraSensor.cc:33:
In file included from /Users/jrivero/code/gazebo/gazebo/rendering/Camera.hh:46:
In file included from /Users/jrivero/code/gazebo/gazebo/rendering/ogre_gazebo.h:26:
In file included from /usr/local/include/OGRE/Ogre.h:31:
In file included from /usr/local/Cellar/ogre/1.7.4/include/OGRE/OgrePrerequisites.h:315:
In file included from /usr/local/Cellar/ogre/1.7.4/include/OGRE/OgreStdHeaders.h:106:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/CoreFoundation.framework/Headers/CoreFoundation.h:43:
In file included from /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/CoreFoundation.framework/Headers/CFBase.h:18:
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/CoreFoundation.framework/Headers/CFAvailability.h:21:10: fatal error: 
      'os/availability.h' file not found
#include <os/availability.h>
         ^
1 error generated.
```

scpeters, 2016-12-10:
Yeah, this gets really confusing. I think the `SDKROOT` and `MACOSX_DEPLOYMENT_TARGET` environment variables may be helpful with this problem, but I'm not sure what they need to be set to.

j-rivero, 2016-12-12:
I've fixed it doing the following:

 * Uninstall Xcode 8 and Install Xcode 7
 * export SDKROOT=macosx10.11
 * remove all the dependencies
 * install gazebo8 --dependencies-only.

My hypothesis is that there was a mixture of libraries compiled using Xcode8 (10.12) and Xcode7 (10.11).

closed 2016-12-12 by j-rivero