osrf/homebrew-simulation #167: Unable to install player using brew

Scalpel78, 2016-12-10:
Hi, I'm having trouble installing player using brew.

Found a link you've put up on the player-support site that pointed to a brew player.rb file here.

On Ubuntu 16.04 I've done:
```
sudo apt-get install linuxbrew-wrapper
brew
brew tap osrf/simulation
brew install player --HEAD
```

but then it crashes with this:

```
==> Installing player from osrf/simulation
Error: undefined method `rebuild' for #<BottleSpecification:0x000000013af100>
Please report this bug:
    https://github.com/Linuxbrew/linuxbrew/blob/master/share/doc/homebrew/Troubleshooting.md#troubleshooting
/home/frode/.linuxbrew/Library/Taps/homebrew/homebrew-core/Formula/curl.rb:9:in `block in <class:Curl>'
/home/frode/.linuxbrew/Library/Homebrew/software_spec.rb:91:in `instance_eval'
/home/frode/.linuxbrew/Library/Homebrew/software_spec.rb:91:in `bottle'
/home/frode/.linuxbrew/Library/Homebrew/formula.rb:1729:in `bottle'
/home/frode/.linuxbrew/Library/Taps/homebrew/homebrew-core/Formula/curl.rb:7:in `<class:Curl>'
/home/frode/.linuxbrew/Library/Taps/homebrew/homebrew-core/Formula/curl.rb:1:in `load_formula'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:21:in `module_eval'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:21:in `load_formula'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:34:in `load_formula_from_path'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:83:in `load_file'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:74:in `klass'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:70:in `get_formula'
/home/frode/.linuxbrew/Library/Homebrew/formulary.rb:207:in `factory'
/home/frode/.linuxbrew/Library/Homebrew/dependency.rb:32:in `to_formula'
/home/frode/.linuxbrew/Library/Homebrew/dependency.rb:93:in `block in expand'
/home/frode/.linuxbrew/Library/Homebrew/dependencies.rb:9:in `each'
/home/frode/.linuxbrew/Library/Homebrew/dependencies.rb:9:in `each'
/home/frode/.linuxbrew/Library/Homebrew/dependency.rb:80:in `expand'
/home/frode/.linuxbrew/Library/Homebrew/dependency.rb:93:in `block in expand'
/home/frode/.linuxbrew/Library/Homebrew/dependencies.rb:9:in `each'
/home/frode/.linuxbrew/Library/Homebrew/dependencies.rb:9:in `each'
/home/frode/.linuxbrew/Library/Homebrew/dependency.rb:80:in `expand'
/home/frode/.linuxbrew/Library/Homebrew/formula.rb:1223:in `recursive_dependencies'
/home/frode/.linuxbrew/Library/Homebrew/requirement.rb:196:in `expand'
/home/frode/.linuxbrew/Library/Homebrew/formula.rb:1229:in `recursive_requirements'
/home/frode/.linuxbrew/Library/Homebrew/formula_installer.rb:313:in `expand_requirements'
/home/frode/.linuxbrew/Library/Homebrew/formula_installer.rb:263:in `compute_dependencies'
/home/frode/.linuxbrew/Library/Homebrew/formula_installer.rb:130:in `verify_deps_exist'
/home/frode/.linuxbrew/Library/Homebrew/formula_installer.rb:123:in `prelude'
/home/frode/.linuxbrew/Library/Homebrew/cmd/install.rb:231:in `install_formula'
/home/frode/.linuxbrew/Library/Homebrew/cmd/install.rb:93:in `block in install'
/home/frode/.linuxbrew/Library/Homebrew/cmd/install.rb:93:in `each'
/home/frode/.linuxbrew/Library/Homebrew/cmd/install.rb:93:in `install'
/home/frode/.linuxbrew/Library/brew.rb:84:in `<main>'

```
I haven't used brew before, so it might be something basic I'm doing wrong?

scpeters, 2016-12-10:
If you're using Ubuntu, you could just install `robot-player-dev`

* http://packages.ubuntu.com/trusty/robot-player

Scalpel78, 2016-12-10:
It seems like that package is only for Trusty, which is Ubuntu 14.04. If I try on Ubuntu 16.04 I get these:

```
frode@xenialxerus:~$ sudo apt-get install robot-player-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package robot-player-dev
```

and

```
frode@xenialxerus:~$ sudo apt-get install robot-player
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package robot-player is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'robot-player' has no installation candidate
```

Is there a way to force an older package onto Ubuntu 16.04?

scpeters, 2016-12-10:
@j-rivero do you know why the Player package isn't available on Ubuntu Xenial? I can't remember if there was a ticket about it.

j-rivero, 2016-12-12:
> @j-rivero do you know why the Player package isn't available on Ubuntu Xenial? I can't remember if there was a ticket about it.

[Debain removed it](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=801898) some time ago, It needed a good bunch of upstream (missing) patches and fixes to fulfill the debian QA policy.

scpeters, 2016-12-14:
@Scalpel78 it sounds like it is not being actively maintained and is missing some patches.

Scalpel78, 2016-12-14:
Yes, I noticed that as well, so I've decided to ditch Player, and instead go for STDR or Gazebo.

closed 2016-12-14 by Scalpel78