osrf/homebrew-simulation #168: can't install gazebo with hoembrew install gazebo

spk921, 2016-12-12:
I can install gazebo7 but I was not able to build with ROS do you know how I could do it? 

Also brew install gazebo is not working with following error messages. 

==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo1/1.9.7 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRA
==> make install
Last 15 lines from /Users/sangpil_kim/Library/Logs/Homebrew/gazebo1/02.make:
[ 44%] Building CXX object gazebo/physics/ode/CMakeFiles/gazebo_physics_ode.dir/ODEHeightmapShape.cc.o
cd /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_ode_EXPORTS -I/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/test/gtest/include -I/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7 -I/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build -isystem /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/deps/opende/include -iframework /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat/2.3.2_1/include/sdformat-2.3  -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11 -fPIC   -fPIC -o CMakeFiles/gazebo_physics_ode.dir/ODEHeightmapShape.cc.o -c /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/gazebo/physics/ode/ODEHeightmapShape.cc
[ 44%] Building CXX object gazebo/physics/ode/CMakeFiles/gazebo_physics_ode.dir/ODEScrewJoint.cc.o
cd /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_ode_EXPORTS -I/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/test/gtest/include -I/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7 -I/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build -isystem /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/deps/opende/include -iframework /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre/1.7.4/include -isystem /usr/local/Cellar/ogre/1.7.4/include/OGRE -isystem /usr/local/include -isystem /usr/local/Cellar/sdformat/2.3.2_1/include/sdformat-2.3  -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11 -fPIC   -fPIC -o CMakeFiles/gazebo_physics_ode.dir/ODEScrewJoint.cc.o -c /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/gazebo/physics/ode/ODEScrewJoint.cc
[ 44%] Linking CXX shared library libgazebo_transport.dylib
cd /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/transport && /usr/local/Cellar/cmake/3.7.1/bin/cmake -E cmake_link_script CMakeFiles/gazebo_transport.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++    -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11 -dynamiclib -Wl,-headerpad_max_install_names   -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 1.0.0 -current_version 1.9.7 -o libgazebo_transport.1.9.7.dylib -install_name /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/transport/libgazebo_transport.1.dylib CMakeFiles/gazebo_transport.dir/CallbackHelper.cc.o CMakeFiles/gazebo_transport.dir/Connection.cc.o CMakeFiles/gazebo_transport.dir/ConnectionManager.cc.o CMakeFiles/gazebo_transport.dir/IOManager.cc.o CMakeFiles/gazebo_transport.dir/Node.cc.o CMakeFiles/gazebo_transport.dir/Publication.cc.o CMakeFiles/gazebo_transport.dir/PublicationTransport.cc.o CMakeFiles/gazebo_transport.dir/Publisher.cc.o CMakeFiles/gazebo_transport.dir/Subscriber.cc.o CMakeFiles/gazebo_transport.dir/SubscriptionTransport.cc.o CMakeFiles/gazebo_transport.dir/TopicManager.cc.o CMakeFiles/gazebo_transport.dir/TransportIface.cc.o  -L/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/test  -L/usr/local/Cellar/ogre/1.7.4/lib  -L/usr/local/Cellar/sdformat/2.3.2_1/lib -Wl,-rpath,/tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/test -Wl,-rpath,/usr/local/Cellar/ogre/1.7.4/lib -Wl,-rpath,/usr/local/Cellar/sdformat/2.3.2_1/lib ../msgs/libgazebo_msgs.1.9.7.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib /usr/local/lib/libtbb.dylib ../common/libgazebo_common.1.9.7.dylib /usr/lib/libdl.dylib /usr/local/lib/libltdl.dylib -lfreeimage -ltinyxml -lcurl -ltar /usr/local/lib/libtbb.dylib -lpthread -lsdformat ../math/libgazebo_math.1.9.7.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_iostreams-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib /usr/local/lib/libprotobuf.dylib 
cd /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/transport && /usr/local/Cellar/cmake/3.7.1/bin/cmake -E cmake_symlink_library libgazebo_transport.1.9.7.dylib libgazebo_transport.1.dylib libgazebo_transport.dylib
[ 44%] Built target gazebo_transport
[ 44%] Linking CXX shared library libgazebo_physics_ode.dylib
cd /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Cellar/cmake/3.7.1/bin/cmake -E cmake_link_script CMakeFiles/gazebo_physics_ode.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++    -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3   -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic -msse -msse2 -msse3 -mssse3 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.11 -dynamiclib -Wl,-headerpad_max_install_names   -Wl,-undefined -Wl,dynamic_lookup -compatibility_version 1.0.0 -current_version 1.9.7 -o libgazebo_physics_ode.1.9.7.dylib -install_name /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/physics/ [... cut ...] ltinyxml -lcurl -ltar /usr/local/lib/libtbb.dylib -lpthread -lsdformat 
cd /tmp/gazebo1-20161212-46501-o8ssux/gazebo-1.9.7/build/gazebo/physics/ode && /usr/local/Cellar/cmake/3.7.1/bin/cmake -E cmake_symlink_library libgazebo_physics_ode.1.9.7.dylib libgazebo_physics_ode.1.dylib libgazebo_physics_ode.dylib
[ 44%] Built target gazebo_physics_ode
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osrf/homebrew-simulation/issues

scpeters, 2016-12-22:
can you build with `--verbose` flags and then post a gist-log?

~~~
brew install --verbose gazebo
brew gist-logs gazebo
~~~

scpeters, 2017-07-05:
closing due to lack of a response. please re-open if you have further trouble

closed 2017-07-05 by scpeters

scpeters, 2017-07-25:
The `gazebo1` formula is very old; perhaps we should change the `gazebo` alias to point to a more recent formula?