osrf/homebrew-simulation #175: OGRE fails to build

GlenHenshaw, 2016-12-22:
Error is:

   ld: framework not found FREETYPE
   clang: error: linker command failed with exit code 1 (use -v to see invocation)
   make[2]: *** [lib/libOgreMain.1.7.4.dylib] Error 1
   make[1]: *** [OgreMain/CMakeFiles/OgreMain.dir/all] Error 2
   make: *** [all] Error 2

Logfiles attached.

GlenHenshaw, 2016-12-22:
[00.options.out.txt](https://github.com/osrf/homebrew-simulation/files/669611/00.options.out.txt)
[01.cmake.cc.txt](https://github.com/osrf/homebrew-simulation/files/669609/01.cmake.cc.txt)
[01.cmake.txt](https://github.com/osrf/homebrew-simulation/files/669612/01.cmake.txt)
[02.make.cc.txt](https://github.com/osrf/homebrew-simulation/files/669610/02.make.cc.txt)
[02.make.txt](https://github.com/osrf/homebrew-simulation/files/669613/02.make.txt)

scpeters, 2016-12-22:
Thanks for the bug report. I don't have a machine running Sierra yet, so I'm not able to debug it for now.

GlenHenshaw, 2016-12-22:
I managed to fix it by manually adding  "-F/Library/Frameworks" and "-lfreetype" flags to the cmake_args line of the formula. I'm very new to both homebrew and cmake, so I can't say I understand for certain what's going on, but my guess is that the formula incorrectly finds (or maybe fails to find) freetype, as it appears to be looking for the freetype *framework* and not the library installed as a dependency by homebrew - and further, maybe, it specifies the framework as ALL CAPS (ie FREETYPE).

Adding the -F flag lets it find the library where XQuartz has installed the freetype framework, so it gets past the include errors, but it still fails to link. Manually adding the -lfreetype flag lets it link.

scpeters, 2016-12-22:
Can you post a patch of the changes you made (or better yet submit a pull request)?

GlenHenshaw, 2016-12-22:
Only if somebody helps e figure out how to do it...

Links:

  1. https://github.com/osrf/homebrew-simulation/issues/175#issuecomment-268899931
  2. https://github.com/

scpeters, 2016-12-23:
I'll try to help. I assume you modified the formula in order to make it compile? Can you post the modified formula?

scpeters, 2016-12-23:
If you modified the formula in the tap, you can also see a patch with the following:

~~~
cd `brew --repo`/Library/Taps/osrf/homebrew-simulation
git diff
~~~

GlenHenshaw, 2016-12-23:
diff --git a/ogre.rb b/ogre.rb
index 116bba7..ae94a99 100644
--- a/ogre.rb
+++ b/ogre.rb
@@ -61,7 +61,7 @@ class Ogre < Formula
     ENV.m64
 
     cmake_args = [
-      "-DCMAKE_OSX_ARCHITECTURES='x86_64'",
+      "-DCMAKE_OSX_ARCHITECTURES='x86_64' -lfreetype -F/Library/Frameworks/",
     ]
     cmake_args << "-DOGRE_BUILD_PLUGIN_CG=OFF" if build.without? "cg"
     cmake_args.concat(std_cmake_args)

scpeters, 2017-01-05:
Ok, I applied your patch in #184, thanks for the contribution!

closed 2017-01-05 by scpeters

GlenHenshaw, 2017-01-05:
Unfortunately, with the latest update, the error is back, and the patch doesn't fix it. Don't know what to do at this point.

commit c787e178 (2017-01-05): Ogre sierra (#184)

* ogre: fix sierra build

* bump revision

* ogre 1.7.4_2 el_capitan bottle

* ogre 1.7.4_2 yosemite bottle

Closes #175