osrf/homebrew-simulation #180: Homebrew update to qt5 breaks gazebo formula

GlenHenshaw, 2017-01-03:
When I updated homebrew this morning, it deleted the "qt" formula and pulled a new formula for qt5 from the repository. This breaks both the gazebo7 and gazebo8 formulae:

`aes048076:build glenhenshaw$ brew install gazebo7
==> Installing gazebo7 from osrf/simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-7.4.0.tar.bz2
Already downloaded: /Users/glenhenshaw/Library/Caches/Homebrew/gazebo7-7.4.0.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/glenhenshaw/Library/Caches/Homebrew/gazebo7--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo7/7.4.0_3 -DCMAKE_BUILD_TYPE=Release -DCMA
Last 15 lines from /Users/glenhenshaw/Library/Logs/Homebrew/gazebo7/01.cmake:
-- Checking for module 'OculusVR'
--   No package 'OculusVR' found
-- 	Oculus Rift support will be disabled.
-- Boost version: 1.63.0
-- Looking for SDFormat - found
CMake Error at /usr/local/Cellar/cmake/3.7.1/share/cmake/Modules/FindQt4.cmake:1318 (message):
  Found unsuitable Qt version "5.7.1" from /usr/local/opt/qt5/bin/qmake, this
  code requires Qt 4.x
Call Stack (most recent call first):
  cmake/SearchForStuff.cmake:547 (find_package)
  CMakeLists.txt:128 (include)

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo7-20170103-61925-19ngszi/gazebo-7.4.0/build/CMakeFiles/CMakeOutput.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osrf/homebrew-simulation/issues
`

scpeters, 2017-01-03:
Thanks for the report; I'm aware of the situation. I've added a qt4 formula in #179 for the old versions of gazebo but still need to change the dependency declaration, and we are currently adding support for qt5 to gazebo8

ppetroski, 2017-01-04:
Can this formula be updated to allow installation on MAC Sierra? There is a similar effort that installs on Sierra, cartr/qt4/qt, but this formula here is referenced as the dependency for gazebo... Thanks

scpeters, 2017-01-04:
I'm not planning to support gazebo 7 and earlier on 10.12; I've only added the `qt4-no-webkit` formula to restore prior functionality. gazebo8 will work on Sierra with qt5

ppetroski, 2017-01-04:
Sounds good, thanks. Regarding gazebo8, it currently has a brew dependency on qt4-no-webkit, is that not necessary and could/should be removed?

GlenHenshaw, 2017-01-04:
The gazebo8 formula throws the exact same error - it lists qt5 as a dependency but the cmake config halts complaining about needing qt4.  

Also, the qt4-no-WebKit formula refuses to install under Sierra, saying it only compiles under El Capitan and earlier.

____________________________
Glen Henshaw • glen@salus.farm  
KC3AYG

scpeters, 2017-01-04:
I will change the gazebo8 dependency to `qt5` after the following is merged:

* https://bitbucket.org/osrf/gazebo/pull-requests/2572/qt5/diff

ppetroski, 2017-01-04:
That works for me. Thanks

closed 2017-01-05 by scpeters