osrf/homebrew-simulation #186: Strange errors after latest brew update on ogre and gazebo7

GlenHenshaw, 2017-01-05:
For instance:

aes048076:/ glenhenshaw$ brew upgrade ogre
Error: /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ogre.rb:66: syntax error, unexpected <<, expecting ']'
<<<<<<< Updated upstream
  ^
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ogre.rb:67: syntax error, unexpected ',', expecting keyword_end
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ogre.rb:68: syntax error, unexpected ',', expecting keyword_end
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ogre.rb:69: syntax error, unexpected ',', expecting keyword_end
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ogre.rb:71: syntax error, unexpected ',', expecting keyword_end
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ogre.rb:73: syntax error, unexpected ']', expecting keyword_end
Please report this bug:
  https://git.io/brew-troubleshooting
/usr/local/Homebrew/Library/Homebrew/formulary.rb:25:in `module_eval'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:25:in `load_formula'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:42:in `load_formula_from_path'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:97:in `load_file'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:88:in `klass'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:84:in `get_formula'
/usr/local/Homebrew/Library/Homebrew/formulary.rb:255:in `factory'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:67:in `block in resolved_formulae'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:52:in `map'
/usr/local/Homebrew/Library/Homebrew/extend/ARGV.rb:52:in `resolved_formulae'
/usr/local/Homebrew/Library/Homebrew/cmd/upgrade.rb:43:in `upgrade'
/usr/local/Homebrew/Library/Homebrew/brew.rb:94:in `<main>'

scpeters, 2017-01-05:
it sounds like you have a conflict with your locally modified file. can you:

~~~
cd `brew --repo`/Library/Taps/osrf/homebrew-simulation
git stash
git reset --hard
~~~

then try again?

GlenHenshaw, 2017-01-05:
Yes, I think that was the problem. What I actually did was

    $ brew untap osrf/simulations
    $ brew tap osrf/simulations

and then the build proceeded. Unfortunately, it didn't succeed and reproduced issue #175.

starthal, 2017-02-09:
Are you sure it was the same error as #175? Because I'm seeing a similar issue, but it's failing to find `-lIOKit`. 

[gist-logs](https://gist.github.com/a722c309fc67479430e5060226aa92dc)

scpeters, 2017-09-15:
I'm closing since this hasn't had recent activity. Please re-open if the issue persists.

closed 2017-09-15 by scpeters