osrf/homebrew-simulation #187: gazebo8 configuration error

GlenHenshaw, 2017-01-05:
$ brew install gazebo8
==> Installing gazebo8 from osrf/simulation
==> Downloading https://bitbucket.org/osrf/gazebo/get/d60939b4f3d4.tar.gz
Already downloaded: /Users/glen/Library/Caches/Homebrew/gazebo8-8.0.0-20170104-d60939b4f3d4.tar.gz
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/glen/Library/Caches/Homebrew/gazebo8--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo8/8.0.0-20170104-d60939b4f3d4 -DCMAKE_BUIL
Last 15 lines from /Users/glen/Library/Logs/Homebrew/gazebo8/01.cmake:
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang++
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Gazebo version 8.0.0
CMake Error at CMakeLists.txt:27 (message):
  The parameter ENABLE_TESTS_COMPILATION is deprecated.  The new behaviour is
  to call 'make tests' explicitly to compile the testsuite.  Calling 'make'
  or 'make all' won't compile the tests

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo8-20170105-37053-134twzi/osrf-gazebo-d60939b4f3d4/build/CMakeFiles/CMakeOutput.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osrf/homebrew-simulation/issues

These open issues may also help:
gazebo8: build bottles https://github.com/osrf/homebrew-simulation/pull/185

scpeters, 2017-01-05:
can you try again now that #185 has been merged?

GlenHenshaw, 2017-01-05:
Well, now I get:

    $ brew install gazebo8
    ==> Installing gazebo8 from osrf/simulation
    ==> Downloading https://bitbucket.org/osrf/gazebo/get/d60939b4f3d4.tar.gz
    Already downloaded: /Users/glen/Library/Caches/Homebrew/gazebo8-8.0.0-20170104-    d60939b4f3d4.tar.gz
    ==> Downloading     https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
    Already downloaded: /Users/glenhenshaw/Library/Caches/Homebrew/gazebo8--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
    ==> Patching
    ==> Applying brew_python_fix.patch
    patching file deps/opende/CMakeLists.txt
    ==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo8/8.0.0-20170104-d60939b4f3d4 -DCMAKE_BUIL
    ==> make install
    Last 15 lines from /Users/glenhenshaw/Library/Logs/Homebrew/gazebo8/02.make:
                       ^
    /usr/local/opt/qt5/include/QtWidgets/qsizepolicy.h:53:7: note: forward declaration of 'Q_WIDGETS_EXPORT'
    class Q_WIDGETS_EXPORT QSizePolicy
          ^
    fatal error: too many errors emitted, stopping now [-ferror-limit=]
    20 errors generated.

GlenHenshaw, 2017-01-05:
This might be because I still had qt4 installed from a different tap. I've norm `brew link --force qt5'-ed and am rebuilding...

GlenHenshaw, 2017-01-05:
Success!

scpeters, 2017-01-05:
yeah, there's a problem if both qt5 and qt4 are installed at the same time; they seem to conflict with each other when building gazebo

closed 2017-01-05 by scpeters