osrf/homebrew-simulation #188: Ogre 1.9 fails to build on MacOS Sierra

GlenHenshaw, 2017-01-05:
...with a mysterious error message.

    $ brew install ogre1.9

eventually results in:

    [ 95%] Linking CXX shared library ../../lib/macosx/libOgreRTShaderSystem.dylib
    cd /tmp/ogre1.9-20170105-26159-9dk3ww/sinbad-ogre-108ab0bcc696/build/Components/RTShaderSystem && /usr/local/Cellar/cmake/3.7.1/bin/cmake -E cmake_link_script CMakeFiles/OgreRTShaderSystem.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -msse -Wall -Winit-self -Wno-overloaded-virtual -Wcast-qual -Wwrite-strings -Wextra -Wno-unused-parameter -Wshadow -Wno-missing-field-initializers -Wno-long-long -stdlib=libc++ -DNDEBUG -arch x86_64 -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.7 -dynamiclib -Wl,-headerpad_max_install_names -compatibility_version 1.9.0 -current_version 1.9.0 -o ../../lib/macosx/libOgreRTShaderSystem.1.9.0.dylib -install_name /tmp/ogre1.9-20170105-26159-9dk3ww/sinbad-ogre-108ab0bcc696/build/lib/macosx/libOgreRTShaderSystem.1.9.0.dylib CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderCGProgramProcessor.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderCGProgramWriter.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExDualQuaternionSkinning.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExHardwareSkinning.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExHardwareSkinningTechnique.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExIntegratedPSSM3.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExLayeredBlending.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExLinearSkinning.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExNormalMapLighting.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExPerPixelLighting.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExTextureAtlasSampler.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderExTriplanarTexturing.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFFPColour.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFFPFog.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFFPLighting.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFFPRenderStateBuilder.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFFPTexturing.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFFPTransform.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFunction.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderFunctionAtom.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderGLSLESProgramProcessor.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderGLSLESProgramWriter.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderGLSLProgramProcessor.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderGLSLProgramWriter.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderGenerator.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderHLSLProgramProcessor.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderHLSLProgramWriter.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderMaterialSerializerListener.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderParameter.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderProgram.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderProgramManager.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderProgramProcessor.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderProgramSet.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderProgramWriter.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderProgramWriterManager.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderRenderState.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderScriptTranslator.cpp.o CMakeFiles/OgreRTShaderSystem.dir/src/OgreShaderSubRenderState.cpp.o  -L/usr/local/lib -Wl,-rpath,/usr/local/lib ../../lib/macosx/libOgreMain.1.9.0.dylib /usr/local/lib/libboost_thread-mt.a /usr/local/lib/libboost_date_time-mt.a /usr/local/lib/libboost_system-mt.a /usr/local/lib/libboost_atomic-mt.a /usr/local/lib/libboost_chrono-mt.a /usr/local/lib/libfreeimage.dylib /usr/local/lib/libzzip.dylib /usr/lib/libz.dylib 
cd /tmp/ogre1.9-20170105-26159-9dk3ww/sinbad-ogre-108ab0bcc696/build/Components/RTShaderSystem && /usr/local/Cellar/cmake/3.7.1/bin/cmake -E cmake_symlink_library ../../lib/macosx/libOgreRTShaderSystem.1.9.0.dylib ../../lib/macosx/libOgreRTShaderSystem.1.9.0.dylib ../../lib/macosx/libOgreRTShaderSystem.dylib
    [ 95%] Built target OgreRTShaderSystem
    make: *** [all] Error 2

The log file is attached, but it doesn't actually list what the error is.

Note that this formula worked just fine on one of my two MacOS Sierra machines, but fails on the other one.

[00.options.out.txt](https://github.com/osrf/homebrew-simulation/files/688113/00.options.out.txt)
[01.cmake.cc.txt](https://github.com/osrf/homebrew-simulation/files/688112/01.cmake.cc.txt)
[01.cmake.txt](https://github.com/osrf/homebrew-simulation/files/688114/01.cmake.txt)
[02.make.cc.txt](https://github.com/osrf/homebrew-simulation/files/688116/02.make.cc.txt)
[02.make.txt](https://github.com/osrf/homebrew-simulation/files/688115/02.make.txt)

scpeters, 2017-01-05:
this has the same error as before about freetype. search make.txt for `***`

GlenHenshaw, 2017-01-05:
Ah. Yes it does.

OK, so the issue here seems to be that the build fails if you have a Freetype.framework already installed in /Library/Frameworks. I don't know why that should be a problem, and I don't know enough about cmake to fix it. Manually moving the framework out of the way lets the build proceed.

scpeters, 2017-09-15:
We have sierra bottles for ogre1.9 now so I'll close this. Please re-open if the issue persists.

closed 2017-09-15 by scpeters