osrf/homebrew-simulation #196: Latest gazebo 8 formula crashes on startup

GlenHenshaw, 2017-01-13:
Log file is attached. It doesn't even get to the splash screen.

[gzserver-8.0.0_2017-01-13-091941_wonko.crash.txt](https://github.com/osrf/homebrew-simulation/files/704491/gzserver-8.0.0_2017-01-13-091941_wonko.crash.txt)

chapulina, 2017-01-26:
Our Jenkins jobs are also failing:

http://build.osrfoundation.org/view/main/view/BuildCopFail/job/gazebo-ci-default-homebrew-amd64/195/console

http://build.osrfoundation.org/view/main/view/BuildCopFail/job/gazebo-ci-gazebo8-homebrew-amd64/11/console

The message is:

~~~
==> Installing osrf/simulation/gazebo8 dependency: ignition-math2
Error: Cannot install osrf/simulation/ignition-math2 because conflicting formulae are installed.

  ignition-math3: because Symbols collision between the two libraries

Please `brew unlink ignition-math3` before continuing.
~~~

Not sure who's pulling ign-math2.

j-rivero, 2017-01-26:
We've merged the latest PR #212 5 hours ago which solves many of the problems (both in runtime and in dependency resolution) found during the gazebo8 beta series. The gazebo8 8.0.0 is in place. Could please double check if the problems still persist?

scpeters, 2017-01-27:
The jenkins jobs should be fixed now.

closed 2017-08-29 by scpeters