osrf/homebrew-simulation #214: How do you manage bottles etc?

ahundt, 2017-02-02:
I assume you guys have some backend stuff that manages your bottles and such. Any chance that's released/recorded somewhere? I'd love to be able to replicate that functionality for my own tap. :-)

scpeters, 2017-02-02:
We have some scripts in our release-tools repository. You're free to look at them, and we may be able to answer some brief questions, but you're welcome to read the code:

* https://bitbucket.org/osrf/release-tools/src/default/jenkins-scripts/lib/homebrew_bottle_creation.bash
* https://bitbucket.org/osrf/release-tools/src/default/jenkins-scripts/dsl/core.dsl
* https://bitbucket.org/osrf/release-tools/src/default/jenkins-scripts/lib/homebrew_bottle_pullrequest.bash

scpeters, 2017-09-15:
Discussion seems to have died down. Please re-open if anyone has any more questions.

closed 2017-09-15 by scpeters