osrf/homebrew-simulation #215: Build failure on Sierra (10.12.2) due to libqwt-dev dependency

starthal, 2017-02-03:
```
brew install gazebo8
==> Installing gazebo8 from osrf/simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-8.0.0.tar.bz2
Already downloaded: /Users/salbert/Library/Caches/Homebrew/gazebo8-8.0.0.tar.bz2
==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
Already downloaded: /Users/salbert/Library/Caches/Homebrew/gazebo8--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
==> Patching
==> Applying brew_python_fix.patch
patching file deps/opende/CMakeLists.txt
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/opt/brew/Cellar/gazebo8/8.0.0_2 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev
Last 15 lines from /Users/salbert/Library/Logs/Homebrew/gazebo8/01.cmake:
-- 	Graphviz not found, Model editor's schematic view will be disabled.
-- 	OpenSource Virtual Reality (OSVR) support will be disabled.
-- END BUILD WARNINGS

-- BUILD ERRORS: These must be resolved before compiling.
-- 	Missing: libqwt-dev. Required for plotting.
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:303 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo8-20170203-95067-1g1mcoo/gazebo-8.0.0/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/gazebo8-20170203-95067-1g1mcoo/gazebo-8.0.0/build/CMakeFiles/CMakeError.log".
```

qwt-6.1.3_3 is installed according to Homebrew.

I would post the CMake logs, but I don't know how to keep Homebrew from deleting them.

scpeters, 2017-02-03:
can you try:

~~~
brew install gazebo8 --verbose
brew gist-logs gazebo8
~~~

that should upload the logs automatically. thanks!

scpeters, 2017-02-03:
Also, can you include the output of `brew info qwt`?

starthal, 2017-02-06:
[Logs](https://gist.github.com/anonymous/5a1ec901c4899e5797083dc01b737c76)

`brew info qwt`:

```
qwt: stable 6.1.3
Qt Widgets for Technical Applications
http://qwt.sourceforge.net/
/opt/brew/Cellar/qwt/6.1.3_3 (1,669 files, 18.7M) *
  Built from source on 2017-02-02 at 16:40:43
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qwt.rb
==> Dependencies
Required: qt5 ✔
==> Options
--with-qwtmathml
	Build the qwtmathml library
--without-plugin
	Skip building the Qt Designer plugin
```

starthal, 2017-02-06:
I did try moving MacPorts out of the way as well -- no effect.

scpeters, 2017-02-06:
It looks like gazebo doesn't handle an alternative install prefix of `qwt`:

* https://bitbucket.org/osrf/gazebo/src/56be256c4b/cmake/SearchForStuff.cmake#SearchForStuff.cmake-753

It searches in either `/usr` or `/usr/local`

scpeters, 2017-02-06:
Can you try applying this patch to the gazebo8 formula:

~~~
diff --git a/gazebo8.rb b/gazebo8.rb
index d17eb24..97a859b 100644
--- a/gazebo8.rb
+++ b/gazebo8.rb
@@ -63,6 +63,8 @@ class Gazebo8 < Formula
     ENV.m64
 
     cmake_args = std_cmake_args
+    cmake_args << "-DQWT_WIN_INCLUDE_DIR=${HOMEBREW_PREFIX}/lib/qwt.framework/Headers"
+    cmake_args << "-DQWT_WIN_LIBRARY_DIR=${HOMEBREW_PREFIX}/lib/qwt.framework"
 
     mkdir "build" do
       system "cmake", "..", *cmake_args
~~~

scpeters, 2017-02-06:
I'll make a pull request with this if it works for you.

starthal, 2017-02-06:
It didn't work with `${HOMEBREW_PREFIX}` --  that must not be expanding properly. But when I replaced that with my hardcoded prefix it worked.

starthal, 2017-02-06:
I think it needs to be expanded within Ruby with `#{HOMEBREW_PREFIX}` instead.

starthal, 2017-02-06:
I can confirm that it builds and runs with that change. I do need to run it with `DYLD_FALLBACK_LIBRARY_PATH=$(brew --prefix)/lib` but I suspect that's a different issue.

scpeters, 2017-02-06:
ok, thanks for testing. I'll make a PR using the `#{}` syntax

scpeters, 2017-02-07:
I merged that fix in #216. Please re-open if that fix is not sufficient (excluding the `DYLD_FALLBACK_LIBRARY_PATH` issue)

commit b8fb52b3 (2017-02-06): gazebo8: help find qwt in brew prefix

Closes #215.
PR #216 (2017-02-07): gazebo8: help find qwt in brew prefix
Closes #215.
commit e00db483 (2017-02-07): gazebo8: help find qwt in brew prefix (#216)

Closes #215.
closed 2017-02-07 by scpeters