osrf/homebrew-simulation #217: gazebo8: libBulletDynamics not in @rpath (fails to run for alternative brew prefix)

starthal, 2017-02-07:
Sierra 10.12.2
My Homebrew prefix is `/opt/brew`.

```
$ gazebo-8.0.0 
dyld: Library not loaded: @rpath/libBulletDynamics.2.85.dylib
  Referenced from: /opt/brew/opt/bullet/lib/libBulletSoftBody.2.85.dylib
  Reason: image not found
Abort trap: 6
```
```
$ otool -L /opt/brew/bin/gazebo-8.0.0 
/opt/brew/bin/gazebo-8.0.0:
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_sensors.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_gui.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_physics.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_ode.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_gimpact.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_opende_ou.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_ccd.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_opcode.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/opt/bullet/lib/libBulletSoftBody.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	/opt/brew/opt/bullet/lib/libBulletDynamics.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	/opt/brew/opt/bullet/lib/libBulletCollision.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	/opt/brew/opt/bullet/lib/libLinearMath.2.85.dylib (compatibility version 2.85.0, current version 2.85.0)
	/opt/brew/opt/simbody/lib/libSimTKsimbody.3.5.dylib (compatibility version 3.5.0, current version 0.0.0)
	/opt/brew/opt/simbody/lib/libSimTKmath.3.5.dylib (compatibility version 3.5.0, current version 0.0.0)
	/opt/brew/opt/simbody/lib/libSimTKcommon.3.5.dylib (compatibility version 3.5.0, current version 0.0.0)
	/System/Library/Frameworks/Accelerate.framework/Versions/A/Accelerate (compatibility version 1.0.0, current version 4.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.0.0)
	/opt/brew/opt/ogre1.9/lib/libOgreRTShaderSystem.1.9.0.dylib (compatibility version 1.9.0, current version 1.9.0)
	/opt/brew/opt/ogre1.9/lib/libOgreMain.1.9.0.dylib (compatibility version 1.9.0, current version 1.9.0)
	/opt/brew/opt/ogre1.9/lib/libOgreTerrain.1.9.0.dylib (compatibility version 1.9.0, current version 1.9.0)
	/opt/brew/opt/ogre1.9/lib/libOgrePaging.1.9.0.dylib (compatibility version 1.9.0, current version 1.9.0)
	/opt/brew/opt/ogre1.9/lib/libOgreOverlay.1.9.0.dylib (compatibility version 1.9.0, current version 1.9.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_client.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_util.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/opt/tinyxml2/lib/libtinyxml2.4.dylib (compatibility version 4.0.0, current version 4.0.1)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_rendering.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_transport.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_msgs.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_common.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/Cellar/gazebo8/8.0.0_2/lib/libgazebo_math.8.dylib (compatibility version 8.0.0, current version 8.0.0)
	/opt/brew/opt/freeimage/lib/libfreeimage.dylib (compatibility version 3.0.0, current version 3.17.0)
	/usr/lib/libcurl.4.dylib (compatibility version 7.0.0, current version 9.0.0)
	/opt/brew/opt/libtar/lib/libtar.0.dylib (compatibility version 1.0.0, current version 1.0.0)
	/opt/brew/opt/tbb/lib/libtbb.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/sdformat5/lib/libsdformat.5.dylib (compatibility version 5.0.0, current version 5.0.0)
	/System/Library/Frameworks/OpenAL.framework/Versions/A/OpenAL (compatibility version 1.0.0, current version 1.0.0)
	/opt/brew/opt/boost/lib/libboost_thread-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_signals-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_system-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_filesystem-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_program_options-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_regex-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_iostreams-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_date_time-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_chrono-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/boost/lib/libboost_atomic-mt.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/ignition-msgs/lib/libignition-msgs0.0.dylib (compatibility version 0.0.0, current version 0.7.0)
	/opt/brew/opt/ignition-math3/lib/libignition-math3.3.dylib (compatibility version 3.0.0, current version 3.0.0)
	/opt/brew/opt/tinyxml/lib/libtinyxml.dylib (compatibility version 0.0.0, current version 0.0.0)
	/opt/brew/opt/qt5/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.8.0, current version 5.8.0)
	/opt/brew/opt/qt5/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.8.0, current version 5.8.0)
	/opt/brew/opt/qt5/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.8.0, current version 5.8.0)
	/opt/brew/opt/qwt/lib/qwt.framework/Versions/6/qwt (compatibility version 6.1.0, current version 6.1.3)
	/opt/brew/opt/ignition-transport3/lib/libignition-transport3.3.dylib (compatibility version 3.0.0, current version 3.0.1)
	/opt/brew/opt/protobuf/lib/libprotobuf.12.dylib (compatibility version 13.0.0, current version 13.0.0)
	/opt/brew/opt/zeromq/lib/libzmq.5.dylib (compatibility version 7.0.0, current version 7.1.0)
	/opt/brew/opt/ossp-uuid/lib/libuuid.16.dylib (compatibility version 17.0.0, current version 17.22.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.4.0)
```
```
$ otool -l /opt/brew/bin/gazebo-8.0.0
...
Load command 22
          cmd LC_LOAD_DYLIB
      cmdsize 80
         name /opt/brew/opt/bullet/lib/libBulletDynamics.2.85.dylib (offset 24)
   time stamp 2 Wed Dec 31 19:00:02 1969
      current version 2.85.0
compatibility version 2.85.0
...
```
It seems the library path is directly specified in the executable, and yet it's not found.

The library is certainly present at that path. And if I set `LD_LIBRARY_FALLBACK_PATH=/opt/brew/opt/bullet/lib`, or just `/opt/brew/lib`, Gazebo runs fine. 

Presumably this is not a problem with the default install prefix because `/usr/local/lib` is searched by default.

starthal, 2017-02-07:
```
$ brew info bullet
bullet: stable 2.86 (bottled), HEAD
Physics SDK
http://bulletphysics.org/wordpress/
/opt/brew/Cellar/bullet/2.86 (461 files, 6.6M) *
  Built from source on 2017-02-01 at 18:48:51 with: --with-shared --with-double-precision
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/bullet.rb
==> Dependencies
Build: cmake ✔
==> Options
--with-demo
	Build demo applications
--with-double-precision
	Use double precision
--with-framework
	Build frameworks
--with-shared
	Build shared libraries
--HEAD
	Install HEAD version
```

scpeters, 2017-02-07:
It looks like you built gazebo against bullet 2.85 then upgraded it to 2.86. Maybe that's the problem?

starthal, 2017-02-08:
OK, so I just uninstalled both `gazebo8` and `bullet`. Then manually ran:

```
$ brew install bullet --with-shared --with-double-precision
==> Using the sandbox
==> Downloading https://github.com/bulletphysics/bullet3/archive/2.86.tar.gz
Already downloaded: /Users/salbert/Library/Caches/Homebrew/bullet-2.86.tar.gz
==> cmake -DINSTALL_EXTRA_LIBS=ON -DBUILD_UNIT_TESTS=OFF -DBUILD_SHARED_LIBS=ON -DCMAKE_INSTALL_PREFIX=/opt/brew/Cellar/bullet/2.86 -DUSE_DOUBLE_PRECISION=ON -DBUILD_BULLET2_DEMOS=OFF
==> make
==> make install
```
That appeared to install Bullet 2.86. But...

```
$ ls /opt/brew/Cellar/bullet/2.86/lib/*.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet2FileLoader.2.85.dylib   /opt/brew/Cellar/bullet/2.86/lib/libBulletInverseDynamics.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet2FileLoader.dylib        /opt/brew/Cellar/bullet/2.86/lib/libBulletInverseDynamics.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Collision.2.85.dylib    /opt/brew/Cellar/bullet/2.86/lib/libBulletInverseDynamicsUtils.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Collision.dylib         /opt/brew/Cellar/bullet/2.86/lib/libBulletInverseDynamicsUtils.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Common.2.85.dylib       /opt/brew/Cellar/bullet/2.86/lib/libBulletSoftBody.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Common.dylib            /opt/brew/Cellar/bullet/2.86/lib/libBulletSoftBody.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Dynamics.2.85.dylib     /opt/brew/Cellar/bullet/2.86/lib/libBulletWorldImporter.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Dynamics.dylib          /opt/brew/Cellar/bullet/2.86/lib/libBulletWorldImporter.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Geometry.2.85.dylib     /opt/brew/Cellar/bullet/2.86/lib/libBulletXmlWorldImporter.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3Geometry.dylib          /opt/brew/Cellar/bullet/2.86/lib/libBulletXmlWorldImporter.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3OpenCL_clew.2.85.dylib  /opt/brew/Cellar/bullet/2.86/lib/libConvexDecomposition.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBullet3OpenCL_clew.dylib       /opt/brew/Cellar/bullet/2.86/lib/libConvexDecomposition.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBulletCollision.2.85.dylib     /opt/brew/Cellar/bullet/2.86/lib/libGIMPACTUtils.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBulletCollision.dylib          /opt/brew/Cellar/bullet/2.86/lib/libGIMPACTUtils.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBulletDynamics.2.85.dylib      /opt/brew/Cellar/bullet/2.86/lib/libHACD.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBulletDynamics.dylib           /opt/brew/Cellar/bullet/2.86/lib/libHACD.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBulletFileLoader.2.85.dylib    /opt/brew/Cellar/bullet/2.86/lib/libLinearMath.2.85.dylib
/opt/brew/Cellar/bullet/2.86/lib/libBulletFileLoader.dylib         /opt/brew/Cellar/bullet/2.86/lib/libLinearMath.dylib 
```

starthal, 2017-02-08:
I found out that Bullet's `CMakeLists.txt` still has the version set to 2.85: https://github.com/bulletphysics/bullet3/issues/953

Then I found this same RPATH issue reported in Bullet: https://github.com/bulletphysics/bullet3/issues/923

According to that reporter, the real issue is with CMake itself: https://gitlab.kitware.com/cmake/cmake/issues/16589

Either way, this doesn't appear to be a problem with any of the OSRF formulae.

scpeters, 2017-02-08:
Thanks for tracking down those other issues. I'll leave this open until it is resolved upstream

scpeters, 2017-05-01:
It looks like the cmake issue has been fixed upstream for cmake 3.9 but hasn't been released yet.

serser, 2017-05-13:
so, it is an open issue? No way to run gazebo on Sierra? Neither 7 nor 8?

scpeters, 2017-05-15:
I think gazebo8 will run on sierra, but I haven't finished building bottles for it yet

claudiofantacci, 2017-06-21:
Hi everyone, thanks for linking to issue [bulletphysics/bullet3#923](https://github.com/bulletphysics/bullet3/issues/923).

On my side, I am able to run Gazebo8 (and only Gazebo8) on Sierra quite smoothly.
The workaround I proposed is not required anymore as there was a problem in the CMakeLists Version number.

Beside that, it seems that something is still wrong in the Bullet installation.
If I issue
```shell
~ $ otool -L /usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib
/usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib:
	/usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	@rpath/libBulletDynamics.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	@rpath/libBulletCollision.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	@rpath/libLinearMath.2.86.dylib (compatibility version 2.86.0, current version 2.86.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 307.4.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1238.0.0)

~ $ otool -l /usr/local/opt/bullet/lib/libBulletSoftBody.2.86.dylib | grep -i rpath
         name @rpath/libBulletDynamics.2.86.dylib (offset 24)
         name @rpath/libBulletCollision.2.86.dylib (offset 24)
         name @rpath/libLinearMath.2.86.dylib (offset 24)
```
I can still find `@rpath` prefix which should not be there. There should be absolute paths only.
This _may be_ connected to the CMake issue that is fixed in version 3.9. Not 100% sure.

CaptStephan, 2017-08-29:
I have been trying to get Gazebo8 running on my Mac with OS X Sierra 10.12.6 with no luck.  I have tried several of the above suggestions as well as others with similar issues to no avail.  It seems to install correctly, then when I run Gazebo, I get the following:  

MacBook-Pro:~ CaptStephan$ gazebo
dyld: Library not loaded: /usr/local/opt/protobuf/lib/libprotobuf.13.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found
Abort trap: 6

I have found a similar file, /usr/local/opt/protobuf/lib/libprotobuf.14.dylib was installed with my current version of Protobuf.  Do I need to install an older version of Protobuf?

scpeters, 2017-08-29:
They recently upgraded protobuf, and we just upgraded the gazebo8 bottles. Can you `brew upgrade gazebo8` and see if that fixes it?

CaptStephan, 2017-08-29:
Perfect.  Thanks.

closed 2017-09-15 by scpeters

sailingmaster92, 2017-11-20:
I guess i am facing similar issues, but I am not sure. I tried installing gazebo. Running it prompts the following problem:
`$ gazebo
dyld: Library not loaded: @rpath/libBulletDynamics.2.87.dylib
  Referenced from: /usr/local/opt/bullet/lib/libBulletSoftBody.2.87.dylib
  Reason: image not found
Abort trap: 6
`

updating it was not possible:
`$ brew upgrade gazebo
Error: osrf/simulation/gazebo8 8.1.1_7 already installed`

fredowski, 2018-09-04:
I think this is addressed here: Homebrew/homebrew-core#31765

I think the preferred way would be not to use rpath but bullet does. Therefore bullet should set
the rpath.