osrf/homebrew-simulation #233: Brew doesn't like optional build flags (double-precision bullet)

scpeters, 2017-05-01:
`brew audit` now complains if optional build flags are specified for default dependencies (https://github.com/Homebrew/brew/pull/2482). Currently the gazebo formulae depend on bullet with `with-shared` and `with-double-precision` options. As of https://github.com/Homebrew/homebrew-core/pull/12636, the `with-shared` option is no longer needed, so could be in compliance with `brew audit` if we default to using a single precision version of bullet. The ubuntu and debian platforms use single precision also, so I think this is a safe change to make.

I will submit a pull request to make this change, but I thought I would create an issue as well to increase visibility.

sloretz, 2017-05-11:
Test failures described by [Gazebo #848](https://bitbucket.org/osrf/gazebo/issues/848) are happening again in [gazebo-ci-default-homebrew-amd64](https://build.osrfoundation.org/view/main/view/BuildCopTests/job/gazebo-ci-default-homebrew-amd64/) and [gazebo-ci-gazebo8-homebrew-amd64](https://build.osrfoundation.org/view/main/view/BuildCopTests/job/gazebo-ci-gazebo8-homebrew-amd64/). Could this have been caused by the removal of `with-double-precision` in pull request #235 ?

scpeters, 2017-05-11:
Yes, I believe this is the cause. It's unexpected, though, because on xenial we use a single-precision bullet library without these test failures. When someone has a chance it would be nice to reproduce these test failures and study them a little.

scpeters, 2017-08-29:
gazebo now uses the default single precision build options for bullet.

closed 2017-08-29 by scpeters