osrf/homebrew-simulation #254: sdformat5 --withurdfdom fails

ahundt, 2017-06-06:
I'm trying to find a way to convert urdfs to sdf files and I figured including the extended features might be relevant, but when I ran `brew reinstall sdformat5 --with-urdfdom` the install failed as follows:

https://gist.github.com/3bdd2e01c5ce6baf49181b9b0357559c

For context, what I'm actually trying to do is convert urdfs to sdf as described at:
https://bitbucket.org/osrf/gazebo/issues/2289/how-to-convert-urdf-to-sdf-and-load-it

So pybullet can become more compatible with the sdf format:
https://github.com/bulletphysics/bullet3/issues/1156

scpeters, 2017-06-06:
urdfdom 1.0 support is in sdformat's default branch, which will become sdformat6

* https://bitbucket.org/osrf/sdformat/pull-requests/324/remove-boost-8/diff

i haven't made a formula for sdformat6 yet, since we haven't finished removing boost, but you can try building it with `--HEAD` for now to see if that works

ahundt, 2017-06-07:
cool thanks I'll give that a try! Does `-p` make use of it or will it be a separate option?

scpeters, 2017-06-07:
I don't know what the `-p` option is. Using `--HEAD` will clone the repository and build the default branch.

scpeters, 2018-10-08:
sdformat6 now has a formula

closed 2018-10-08 by scpeters