osrf/homebrew-simulation #256: Sierra 10.12.5 gazebo doesn't start

emilianogagliardi, 2017-06-15:
I installed gazebo in this way:
`brew install gazebo8`

when I try to run it there is this problem:
```
dyld: Symbol not found: __ZN18btSliderConstraint8setParamEifi
  Referenced from: /usr/local/Cellar/gazebo8/8.1.1/lib/libgazebo_physics.8.dylib
  Expected in: /usr/local/opt/bullet/lib/libBulletDynamics.2.86.dylib
 in /usr/local/Cellar/gazebo8/8.1.1/lib/libgazebo_physics.8.dylib
Abort trap: 6
```
Do you know how to fix it?
thank you in advance

scpeters, 2017-06-16:
Can you post `brew config` and `brew doctor`?

scpeters, 2017-06-16:
Also `brew info bullet`

emilianogagliardi, 2017-06-16:
`brew config` output:
```
HOMEBREW_VERSION: 1.2.3-9-g3fe32053
ORIGIN: https://github.com/Homebrew/brew
HEAD: 3fe32053a7f0d7f817a90e308ab0c62436a5c208
Last commit: 2 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 6de4acf0d944534405bacd780a795ce91cf717d9
Core tap last commit: 11 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local/Homebrew
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit skylake
Homebrew Ruby: 2.0.0-p648
Clang: 8.1 build 802
Git: 2.11.0 => /Library/Developer/CommandLineTools/usr/bin/git
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.13/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.8.0_112, 1.6.0_65-b14-468
macOS: 10.12.5-x86_64
Xcode: N/A
CLT: 8.3.2.0.1.1492020469
X11: 2.7.11 => /opt/X11
```

`brew doctor` output:
```
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry and just ignore them. Thanks!

Warning: Your default Python does not recognize the Homebrew site-packages
directory as a special site-packages directory, which means that .pth
files will not be followed. This means you will not be able to import
some modules after installing them with Homebrew, like wxpython. To fix
this for the current user, you can run:
  mkdir -p /Users/emilianogagliardi/Library/Python/2.7/lib/python/site-packages
  echo 'import site; site.addsitedir("/usr/local/lib/python2.7/site-packages")' >> /Users/emilianogagliardi/Library/Python/2.7/lib/python/site-packages/homebrew.pth

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/libopencv_calib3d.3.2.0.dylib
  /usr/local/lib/libopencv_core.3.2.0.dylib
  /usr/local/lib/libopencv_features2d.3.2.0.dylib
  /usr/local/lib/libopencv_flann.3.2.0.dylib
  /usr/local/lib/libopencv_highgui.3.2.0.dylib
  /usr/local/lib/libopencv_imgcodecs.3.2.0.dylib
  /usr/local/lib/libopencv_imgproc.3.2.0.dylib
  /usr/local/lib/libopencv_ml.3.2.0.dylib
  /usr/local/lib/libopencv_objdetect.3.2.0.dylib
  /usr/local/lib/libopencv_photo.3.2.0.dylib
  /usr/local/lib/libopencv_shape.3.2.0.dylib
  /usr/local/lib/libopencv_stitching.3.2.0.dylib
  /usr/local/lib/libopencv_superres.3.2.0.dylib
  /usr/local/lib/libopencv_video.3.2.0.dylib
  /usr/local/lib/libopencv_videoio.3.2.0.dylib
  /usr/local/lib/libopencv_videostab.3.2.0.dylib

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/opencv/cv.h
  /usr/local/include/opencv/cvaux.h
[... 94 lines cut ...]

```

`brew info bullet` output
```
bullet: stable 2.86.1 (bottled), HEAD
Physics SDK
http://bulletphysics.org/wordpress/
/usr/local/Cellar/bullet/2.86.1 (461 files, 6.6MB)
  Built from source on 2017-03-21 at 21:26:22 with: --with-shared --with-double-precision
/usr/local/Cellar/bullet/2.86.1_1 (479 files, 11.6MB) *
  Built from source on 2017-06-16 at 01:03:17 with: --with-double-precision
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/bullet.rb
==> Dependencies
Build: cmake ✔
==> Options
--with-demo
	Build demo applications
--with-double-precision
	Use double precision
--with-framework
	Build frameworks
--HEAD
	Install HEAD version
```

scpeters, 2017-06-16:
It looks like `gazebo8` is not linked. I think you may have another version of gazebo installed? Only one can be linked at a time, since they both try to install files with the same name to `/usr/local`.

You can also try `brew test gazebo8` to see if it is working.

I also see that you have bullet built from source. We recently switched the bullet install options, so if you could reinstall bullet that will probably help too.

emilianogagliardi, 2017-06-16:
I think that I uninstalled gazebo in the wrong way some time ago, without using brew uninstall. Should I uninstall brew and restart from scratch?

scpeters, 2017-06-16:
try the following:

~~~
brew link --overwrite gazebo8
~~~

ilidar, 2017-06-19:
@scpeters have the same problem, `brew link --overwrite gazebo8` didn't solve it ;-(

francesco-romano, 2017-06-19:
I `brew uninstall`ed `gazebo8` and `bullet` and I managed to solve this issue

ilidar, 2017-06-19:
@francesco-romano yeah, just did same thing, thanks

scpeters, 2017-06-19:
@emilianogagliardi did `brew link --overwrite gazebo8` work?

yapbenzet, 2017-08-14:
brew link --overwrite --dry-run python3 
solve this issue.

scpeters, 2017-08-14:
I'm closing now due to inactivity; please re-open if the issue persists.

closed 2017-08-14 by scpeters

LucienBrule, 2018-01-08:
This can be fixed with:
```
brew uninstall gazebo8
brew uninstall bullet
brew install gazebo8
```
(ie: removing and reinstalling bullet)
In my case I was building gazebo6 and switched to build 8. Something must have happened with bullet.