osrf/homebrew-simulation #263: Installation error on Mac OSX 10.12.6 Sierra

MartonKN, 2017-08-02:
Hi there,

I got this error after trying to install Gazebo6 using Homebrew. 

I followed the steps on the website http://gazebosim.org/tutorials?tut=install_on_mac&ver=6.0&cat=install and have installed XQuartz before.

Marton-MacBook-Pro:~ Marton$ brew install gazebo6 --verbose
==> Installing gazebo6 from osrf/simulation
==> Installing dependencies for osrf/simulation/gazebo6: sdformat3, bullet, simbody
==> Installing osrf/simulation/gazebo6 dependency: sdformat3
==> Using the sandbox
/usr/bin/sandbox-exec -f /tmp/homebrew20170802-4542-petajz.sb nice /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby -W0 -I /usr/local/Homebrew/Library/Homebrew -- /usr/local/Homebrew/Library/Homebrew/build.rb /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/sdformat3.rb --verbose
==> Downloading http://gazebosim.org/distributions/sdformat/releases/sdformat-3.7.0.tar.bz2
Already downloaded: /Users/Marton/Library/Caches/Homebrew/sdformat3-3.7.0.tar.bz2
==> Verifying sdformat3-3.7.0.tar.bz2 checksum
tar xjf /Users/Marton/Library/Caches/Homebrew/sdformat3-3.7.0.tar.bz2
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/sdformat3/3.7.0_5 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DCMAKE_VERBOSE_MAKEFILE=ON -Wno-dev ..
-- The C compiler identification is AppleClang 8.1.0.8020042
-- The CXX compiler identification is AppleClang 8.1.0.8020042
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang
-- Check for working C compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang++
-- Check for working CXX compiler: /usr/local/Homebrew/Library/Homebrew/shims/super/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- SDFormat version 3.7.0
-- 

====== Finding 3rd Party Packages ======
-- Using system tinyxml
-- Operating system is Apple MacOS X
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Building for arch: x86_64
-- Boost version: 1.64.0
-- Boost version: 1.64.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
-- Checking for module 'tinyxml'
--   Found tinyxml, version 2.6.2
-- Could NOT find PY_psutil (missing: PY_PSUTIL) 
-- 	Python psutil package not found. Memory leak tests will be skipped
-- Found ruby executable: /usr/bin/ruby
-- Looking for ignition-math2-config.cmake - found
-- ----------------------------------------

-- Enable host CFlags
-- Enable upstream CFlags
-- Performing Test GCC_SUPPORTS_VISIBILITY
-- Performing Test GCC_SUPPORTS_VISIBILITY - Success
-- Performing Test R-Waddress
-- Performing Test R-Waddress - Success
-- Performing Test R-Warray-bounds
-- Performing Test R-Warray-bounds - Success
-- Performing Test R-Wcomment
-- Performing Test R-Wcomment - Success
-- Performing Test R-Wformat
-- Performing Test R-Wformat - Success
-- Performing Test R-Wnonnull
-- Performing Test R-Wnonnull - Success
-- Performing Test R-Wparentheses
-- Performing Test R-Wparentheses - Success
-- Performing Test R-Wreorder
-- Performing Test R-Wreorder - Success
-- Performing Test R-Wreturn-type
-- Performing Test R-Wreturn-type - Success
-- Performing Test R-Wsequence-point
-- Performing Test R-Wsequence-point - Success
-- Performing Test R-Wsign-compare
-- Performing Test R-Wsign-compare - Success
-- Performing Test R-Wstrict-aliasing
-- Performing Test R-Wstrict-aliasing - Success
-- Performing Test R-Wstrict-overflow=1
-- Performing Test R-Wstrict-overflow=1 - Success
-- Performing Test R-Wswitch
-- Performing Test R-Wswitch - Success
-- Performing Test R-Wtrigraphs
-- Performing Test R-Wtrigraphs - Success
-- Performing Test R-Wuninitialized
-- Performing Test R-Wuninitialized - Success
-- Performing Test R-Wunused-function
-- Performing Test R-Wunused-function - Success
-- Performing Test R-Wunused-label
-- Performing Test R-Wunused-label - Success
-- Performing Test R-Wunused-value
-- Performing Test R-Wunused-value - Success
-- Performing Test R-Wunused-variable
-- Performing Test R-Wunused-variable - Success
-- Performing Test R-Wvolatile-register-var
-- Performing Test R-Wvolatile-register-var - Success
-- Performing Test R-Wextra
-- Performing Test R-Wextra - Success
-- Performing Test R-Wno-long-long
-- Performing Test R-Wno-long-long - Success
-- Performing Test R-Wno-unused-value
-- Performing Test R-Wno-unused-value - Success
-- Performing Test R-Wfloat-equal
-- Performing Test R-Wfloat-equal - Success
-- Performing Test R-Wshadow
-- Performing Test R-Wshadow - Success
-- Performing Test R-Winit-self
-- Performing Test R-Winit-self - Success
-- Performing Test R-Wswitch-default
-- Performing Test R-Wswitch-default - Success
-- Performing Test R-Wmissing-include-dirs
-- Performing Test R-Wmissing-include-dirs - Success
-- Performing Test R-pedantic
-- Performing Test R-pedantic - Success
-- Performing Test R-Wno-pragmas
-- Performing Test R-Wno-pragmas - Success
CMake Error at CMakeLists.txt:191 (else):
  A duplicate ELSE command was found inside an IF block.

-- Configuring incomplete, errors occurred!
See also "/tmp/sdformat3-20170802-4544-1pua9xt/sdformat-3.7.0/build/CMakeFiles/CMakeOutput.log".

==> Formula
Tap: osrf/simulation
Path: /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/sdformat3.rb
==> Configuration
HOMEBREW_VERSION: 1.3.0
ORIGIN: https://github.com/Homebrew/brew
HEAD: 05e9086355607e9103030b1f0e3f186e674ce063
Last commit: 2 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 762a5fb68b7e20fef2097b5eaedce5ce16a62fca
Core tap last commit: 2 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_REPOSITORY: /usr/local/Homebrew
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit haswell
Homebrew Ruby: 2.0.0-p648
Clang: 8.1 build 802
Git: 2.11.0 => /Applications/Xcode.app/Contents/Developer/usr/bin/git
Perl: /usr/bin/perl
Python: /usr/local/opt/python/libexec/bin/python => /usr/local/Cellar/python/2.7.13_1/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby => /System/Library/Frameworks/Ruby.framework/Versions/2.0/usr/bin/ruby
Java: 1.6.0_65-b14-468
macOS: 10.12.6-x86_64
Xcode: 8.3.3
CLT: N/A
X11: 2.7.11 => /opt/X11
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk
MAKEFLAGS: -j4
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/libxml2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/apache2:/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/System/Library/Frameworks
MACOSX_DEPLOYMENT_TARGET: 10.12
PKG_CONFIG_PATH: /usr/local/opt/ignition-math2/lib/pkgconfig:/usr/local/opt/tinyxml/lib/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/10.12
HOMEBREW_SDKROOT: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/super:/usr/local/opt/cmake/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/doxygen/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: osrf/simulation/sdformat3 3.7.0 did not build
Logs:
     /Users/Marton/Library/Logs/Homebrew/sdformat3/00.options.out
     /Users/Marton/Library/Logs/Homebrew/sdformat3/01.cmake
     /Users/Marton/Library/Logs/Homebrew/sdformat3/01.cmake.cc
If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osrf/homebrew-simulation/issues

scpeters, 2017-08-02:
There has been a syntax error in sdformat's cmake code which was ignored until cmake 3.9 was recently released. If you add the following patch to the sdformat3 formula, it should fix at least that part of the build:

* https://github.com/osrfbuild/homebrew-simulation/blob/b4cb7b6af8fafbbcd52a2e7962e89678ffc717cb/sdformat4.rb#L29-L33

Please consider submitting a pull request with this fix if it works for you.

MartonKN, 2017-08-02:
>    b4cb7b6af8fafbbcd52a2e7962e89678ffc717cb/sdformat4.rb#L29-L33

Hi Steven, thank you for your response. I have already been able to install
Gazebo using the command

'curl -ssL http://get.gazebosim.org | sh'

Best, Marton

closed 2017-08-29 by scpeters

aeden, 2017-10-10:
Thanks @scpeters for the patch, it worked for me. Are you looking for a PR here? I'll be happy to create one.

scpeters, 2017-10-10:
@aeden yes please, a pull request would be welcome for this formula

scpeters, 2018-01-25:
the patch has been merged in #365 

thanks @lucienbrule!