osrf/homebrew-simulation #267: Ogre and Ogre1.9 build errors with boost 1.65

scpeters, 2017-08-25:
The ogre and ogre1.9 builds are now failing since boost 1.65 was released.

* https://build.osrfoundation.org/view/os_homebrew/job/generic-release-homebrew_bottle_builder-sierra/46/
* https://build.osrfoundation.org/view/os_homebrew/job/generic-release-homebrew_bottle_builder-elcapitan/62/

Here's a summary of some compiler errors:

* https://gist.github.com/scpeters/32c9e7e56f479832d6b00316a1a24b11

scpeters, 2017-08-25:
The preprocessor is acting weird. Consider some error messages from `RenderSystems/GL/src/OSX/OgreOSXCarbonContext.cpp` related to `boost/thread/pthread/condition_variable.hpp` for example. The original error messages say this:

~~~
/usr/local/include/boost/thread/pthread/condition_variable.hpp:82:43: error: expected unqualified-id
            check_for_interruption.check();
                                          ^
/usr/local/include/boost/thread/pthread/condition_variable.hpp:116:43: error: expected unqualified-id
            check_for_interruption.check();
                                          ^
/usr/local/include/boost/thread/pthread/condition_variable.hpp:193:47: error: expected unqualified-id
                check_for_interruption.check();
                                              ^
/usr/local/include/boost/thread/pthread/condition_variable.hpp:423:45: error: expected unqualified-id
              check_for_interruption.check();
^
~~~

Then I tried saving the pre-processor output to a temporary .cpp file and compiling that with the same command, and it has different errors:

~~~
/usr/local/include/boost/thread/pthread/condition_variable.hpp:82:36: error: expected unqualified-id
            check_for_interruption.;
                                   ^
/usr/local/include/boost/thread/pthread/condition_variable.hpp:116:36: error: expected unqualified-id
            check_for_interruption.;
                                   ^
/usr/local/include/boost/thread/pthread/condition_variable.hpp:193:40: error: expected unqualified-id
                check_for_interruption.;
                                       ^
/usr/local/include/boost/thread/pthread/condition_variable.hpp:423:38: error: expected unqualified-id
              check_for_interruption.;
                                     ^
~~~

scpeters, 2017-08-25:
the `check()` function is also missing from the pre-processed output. the [line from thread_data.hpp](https://github.com/boostorg/thread/blob/boost-1.65.0/include/boost/thread/pthread/thread_data.hpp#L212) says `void check()` in the original code, but the pre-processed code just says `void`

does ogre have a macro for `check()`?

scpeters, 2017-08-25:
here's the compiler flags for the original command:

~~~
cd
/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/build/RenderSystems/GL
&&
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++

-DBOOST_ALL_NO_LIB
-DBOOST_THREAD_VERSION=4
-DOGRE_GLPLUGIN_EXPORTS
-DRenderSystem_GL_EXPORTS
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/include
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/OgreMain/include
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/build/include
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/OgreMain/include/OSX
-I/usr/local/include
-I/usr/local/include/freetype2
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/StateCacheManager
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/OSX
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/include/OSX
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/GLSL/include
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/atifs/include
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/nvparse
-I/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/OgreMain/include/Threading

-msse
-Wall
-Winit-self
-Wno-overloaded-virtual
-Wcast-qual
-Wwrite-strings
-Wextra
-Wno-unused-parameter
-Wshadow
-Wno-missing-field-initializers
-Wno-long-long
-stdlib=libc++
-DNDEBUG
-arch
x86_64
-isysroot
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk
-mmacosx-version-min=10.7
-fPIC

-pthread
-DOGRE_GCC_VISIBILITY
-fvisibility=hidden
-o
CMakeFiles/RenderSystem_GL.dir/src/OSX/OgreOSXCarbonContext.cpp.o
-c
/tmp/ogre1.9-20170825-54043-oel1tz/sinbad-ogre-108ab0bcc696/RenderSystems/GL/src/OSX/OgreOSXCarbonContext.cpp
~~~

iche033, 2017-08-25:
this related issue suggests conflict between boost and osx/xcode function definitions: 
https://stackoverflow.com/questions/31665095/xcode-error-compiling-c-expected-member-name-or-after-declaration-specifie

can we do a test building ogre 1.9 with this patch?

~~~
diff -r 108ab0bcc696 RenderSystems/GL/src/OSX/OgreOSXCarbonContext.cpp
--- a/RenderSystems/GL/src/OSX/OgreOSXCarbonContext.cpp	Fri Jul 15 02:11:59 2016 +0200
+++ b/RenderSystems/GL/src/OSX/OgreOSXCarbonContext.cpp	Fri Aug 25 10:46:30 2017 -0700
@@ -26,8 +26,8 @@
 -----------------------------------------------------------------------------
 */
 
+#include "OgreLogManager.h"
 #include "OgreOSXCarbonContext.h"
-#include "OgreLogManager.h"
 
 namespace Ogre
 { 
~~~

that's just swaps the headers to make sure that all boost headers (included by OgreLogManager.h), is included first before any other osx headers

iche033, 2017-08-25:
or maybe we could just do

~~~
#undef check
~~~

before the boost headers inside `OgreMain/include/Threading/OgreThreadHeadersBoost.h`

scpeters, 2017-08-25:
ok, I'll try that

scpeters, 2017-08-25:
re-ordering the headers worked on the first file I tried. I'm trying the undef trick, but it's rebuilding everything

scpeters, 2017-08-29:
fixed by #266 and #269

closed 2017-08-29 by scpeters