osrf/homebrew-simulation #268: Default installation of Gazebo on OS X fails

akalache, 2017-08-25:
Gazebo relies on protobuf13 (libprotobuf.13.dylib) while the latest installed through dependencies in protobuf14 (libprotobuf.14.dylib). As a result the following is displayed

```
osrf/simulation/gazebo8:
  * gazebo8 has broken dynamic library links:
      /usr/local/opt/protobuf/lib/libprotobuf.13.dylib
    Rebuild this from source with:
      brew reinstall --build-from-source gazebo8
    If that's successful, file an issue here:
      https://github.com/osrf/homebrew-simulation/issues
Error: 1 problem in 1 formula

```

scpeters, 2017-08-25:
Thanks for the report. I believe linking with boost is broken as well. We are in the process of building new bottles (#267 #266 #269). Thanks for your patience.

scpeters, 2017-08-29:
fixed by #266

closed 2017-08-29 by scpeters