osrf/homebrew-simulation #270: gazebo8 conflicting protoc version

julianoes, 2017-08-27:
The Gazebo8 bottle seems to be compiled with an older protoc version, therefore compiling PX4/Firmware fails (see https://github.com/PX4/Firmware/issues/7872).
```
brew install gazebo8
==> Installing gazebo8 from osrf/simulation
==> Installing dependencies for osrf/simulation/gazebo8: ignition-math3, ignition-tools, ignition-msgs, ignition-transport3, sdformat5
==> Installing osrf/simulation/gazebo8 dependency: ignition-math3
==> Downloading http://gazebosim.org/distributions/ign-math/releases/ignition-math3-3.2.0.sierra.bottle.tar.gz
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/ignition-math3-3.2.0.sierra.bottle.tar.gz
==> Pouring ignition-math3-3.2.0.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/ignition-math3/3.2.0: 50 files, 575.6KB
==> Installing osrf/simulation/gazebo8 dependency: ignition-tools
==> Downloading http://gazebosim.org/distributions/ign-tools/releases/ignition-tools-0.0.0-20160209-381c0a968fa.sierra.bottle.tar.gz
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/ignition-tools-0.0.0-20160209-381c0a968fa.sierra.bottle.tar.gz
==> Pouring ignition-tools-0.0.0-20160209-381c0a968fa.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/ignition-tools/0.0.0-20160209-381c0a968fa: 9 files, 22.2KB
==> Installing osrf/simulation/gazebo8 dependency: ignition-msgs
==> Downloading http://gazebosim.org/distributions/ign-msgs/releases/ignition-msgs-0.7.0_2.sierra.bottle.tar.gz
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/ignition-msgs-0.7.0_2.sierra.bottle.tar.gz
==> Pouring ignition-msgs-0.7.0_2.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/ignition-msgs/0.7.0_2: 326 files, 4.7MB
==> Installing osrf/simulation/gazebo8 dependency: ignition-transport3
==> Downloading http://gazebosim.org/distributions/ign-transport/releases/ignition-transport3-3.0.1_4.sierra.bottle.tar.gz
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/ignition-transport3-3.0.1_4.sierra.bottle.tar.gz
==> Pouring ignition-transport3-3.0.1_4.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/ignition-transport3/3.0.1_4: 39 files, 688.7KB
==> Installing osrf/simulation/gazebo8 dependency: sdformat5
==> Downloading http://gazebosim.org/distributions/sdformat/releases/sdformat5-5.2.0.sierra.bottle.tar.gz
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/sdformat5-5.2.0.sierra.bottle.tar.gz
==> Pouring sdformat5-5.2.0.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/sdformat5/5.2.0: 370 files, 1.8MB
==> Installing osrf/simulation/gazebo8
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo8-8.1.1_1.sierra.bottle.tar.gz
Already downloaded: /Users/julianoes/Library/Caches/Homebrew/gazebo8-8.1.1_1.sierra.bottle.tar.gz
==> Pouring gazebo8-8.1.1_1.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/gazebo8/8.1.1_1: 1,188 files, 121.6MB
```

```
head -n20 /usr/local/Cellar/gazebo8/8.1.1_1/include/gazebo-8/gazebo/msgs/quaternion.pb.h
// Generated by the protocol buffer compiler.  DO NOT EDIT!
// source: quaternion.proto

#ifndef PROTOBUF_quaternion_2eproto__INCLUDED
#define PROTOBUF_quaternion_2eproto__INCLUDED

#include <string>

#include <google/protobuf/stubs/common.h>

#if GOOGLE_PROTOBUF_VERSION < 3003000
#error This file was generated by a newer version of protoc which is
#error incompatible with your Protocol Buffer headers.  Please update
#error your headers.
#endif
#if 3003000 < GOOGLE_PROTOBUF_MIN_PROTOC_VERSION
#error This file was generated by an older version of protoc which is
#error incompatible with your Protocol Buffer headers.  Please
#error regenerate this file with a newer version of protoc.
#endif
```
However, the defines are:
```
GOOGLE_PROTOBUF_VERSION 3004000
GOOGLE_PROTOBUF_MIN_PROTOC_VERSION 3004000
```
with:
```
protoc --version
libprotoc 3.4.0
```

scpeters, 2017-08-29:
I think this was fixed by #266

closed 2017-08-29 by scpeters

julianoes, 2017-08-29:
This seems indeed fixed after a `brew upgrade`. Thank you!