osrf/homebrew-simulation #272: sdformat4 links to outdated libboost

julianoes, 2017-09-09:
During the PX4 and Gazebo7 compilation on 10.12.6, libsdformat4.dylib seems to link to a previous boost version which got updated:

```
dyld: Symbol not found: __ZNK5boost16re_detail_10650031cpp_regex_traits_implementationIcE17transform_primaryEPKcS4_
  Referenced from: /usr/local/opt/sdformat4/lib/libsdformat.4.dylib
  Expected in: /usr/local/opt/boost/lib/libboost_regex-mt.dylib
 in /usr/local/opt/sdformat4/lib/libsdformat.4.dylib
```

I have a couple of questions:
1. I'm assuming you need to rebuild sdformat4, right? (Or have I not understood the concept?)
2. Is it correct that this happens everytime a dependency such as boost changes?
3. And what could we do about it?
4. What about automation to detect dependency changes and rebuild automatically?
5. What about including a specific boost version in the osrf repo?
6. How is this handled normally in the brew world?

Thanks a lot in advance.

ateixeiramunoz, 2017-09-09:
Was exactly my problem. Rebuilding sdformat5 in my case was the solution 👍   thanks a lot 4 ur comment!! My last night was crazy looking for this solution!!

scpeters, 2017-09-11:
Thanks for the report; each time they release a new version of boost it breaks our bottles. It looks like they just released boost 1.65.1 ( https://github.com/Homebrew/homebrew-core/pull/17807 ), so I've started the process of building new bottles (#273 and #274).

julianoes, 2017-09-11:
@scpeters awesome, thanks a lot.

scpeters, 2017-09-11:
should be fixed by #273 and #274

closed 2017-09-11 by scpeters

julianoes, 2017-09-12:
@scpeters I can confirm it works again, thanks.

ateixeiramunoz, 2017-09-13:
will try tonight. Thnks!