osrf/homebrew-simulation #303: Request: formula for ogre1.10

scpeters, 2017-11-06:
It would be nice to have a working formula for ogre 1.10. A formula has been submitted to homebrew-core, but we should verify that it works with gazebo and rviz:

* https://github.com/Homebrew/homebrew-core/pull/20310/files

cc: @iche033 @wjwwood

scpeters, 2017-11-06:
I have the default branch of gazebo built against this ogre 1.10 formula on a macOS CI machine that I took offline for this testing. It builds but at run-time fails to find OpenGL rendering system.

scpeters, 2017-11-06:
After fixing the plugin path issue, it looks like it is expecting a `lib` prefix on the library names, but those aren't present in the 1.10 formula. They don't appear to be in the 1.9 formula either actually; there's some extra symlinks that fix it:

* https://github.com/osrf/homebrew-simulation/blob/master/ogre1.9.rb#L103-L111

scpeters, 2017-11-06:
Updating gazebo to not expect the `lib` prefix:

* https://bitbucket.org/osrf/gazebo/pull-requests/2803/dont-use-lib-prefix-for-ogre-plugins-as-of/diff

s172262, 2017-11-07:
There are  plugins in " /usr/local/opt/ogre/lib/OGRE/".
```
$ ls /usr/local/opt/ogre/lib/OGRE/
Plugin_BSPSceneManager.1.10.9.dylib	Plugin_PCZSceneManager.dylib
Plugin_BSPSceneManager.dylib		Plugin_ParticleFX.1.10.9.dylib
Plugin_OctreeSceneManager.1.10.9.dylib	Plugin_ParticleFX.dylib
Plugin_OctreeSceneManager.dylib		RenderSystem_GL.1.10.9.dylib
Plugin_OctreeZone.1.10.9.dylib		RenderSystem_GL.dylib
Plugin_OctreeZone.dylib			cmake
Plugin_PCZSceneManager.1.10.9.dylib
```

iche033, 2017-11-07:
A quick look at the ogre 1.10 source code seems to suggest that the cocoa window's contents scaling factor issue we were facing in ogre 1.9 was fixed so it's promising. It would be nice to start migrating to 1.10 as ogre 1.9 has reached EOL.

scpeters, 2017-11-08:
Ubuntu is still using ogre 1.9, but at least we could start moving forward on homebrew.

I was able to run gazebo against ogre 1.10, but I had a problem running the rendering tests. It complained about an uncaught `ItemIdentityException` (see ogre.log output below). It seems the font definitions are being parsed twice during tests?

~~~
16:15:48: Parsing scripts for resource group Fonts
16:15:48: Parsing script Gazebo.fontdef
16:15:48: Parsing script Gazebo.fontdef
16:15:48: ItemIdentityException: Resource with the name Console already exists. in ResourceManager::add at /tmp/ogre-20171106-85528-14kjopy/ogre-1.10.9/OgreMain/src/OgreResourceManager.cpp (line 156)
~~~

iche033, 2017-11-08:
oh ya I ran into that problem too and fixed it by doing something similar to [this PR](https://bitbucket.org/osrf/gazebo/pull-requests/2801/renderengine-setupresources-fix-resource/diff)

scpeters, 2017-11-09:
I think that PR helps (I just approved it), but a problem with tests is that ServerFixture adds a bunch of local folders to the `GAZEBO_RESOURCE_PATH` so you can run tests without installing. If you have installed, though, it will cause `ItemIdentityException` from having both the install paths and source paths.

I did a quick hack to workaround that issue and found another test seg-fault in `Ogre::ScriptCompiler::parseScript` called from `Ogre;:ResourceGroupManager::initialiseAllResourceGroups`. I would probably have to recompile ogre to investigate that further, so I'm going to pause for now.

scpeters, 2020-09-22:
I'm going to close this since I don't think anyone is working on it. Please re-open if someone resumes work on it.

closed 2020-09-22 by scpeters