osrf/homebrew-simulation #309: Gazebo8 bottles not built yet against latest protobuf

LorenzMeier, 2017-11-14:
On trying to compile a Gazebo plugin I noticed that the OS X bottles for High Sierra are not built against the latest protobuf version in homebrew.

The error message is:

```
In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Actor.hh:24:
In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Model.hh:30:
In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Entity.hh:25:
In file included from /usr/local/Cellar/ignition-transport3/3.0.1_7/include/ignition/transport3/ignition/transport/Node.hh:32:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
/usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
```

Install output:

~~~
:Firmware user$ brew install gazebo8 --verbose
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core, caskroom/cask).
==> Updated Formulae
apibuilder-cli   bitrise          faac             influxdb         mighttpd2        pumba            zenity
bchunk           couchdb          harfbuzz         mercurial        paket            xtensor          zimg

==> Installing gazebo8 from osrf/simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo8-8.1.1_5.high_sierra.bottle.tar.gz
Already downloaded: /Users/user/Library/Caches/Homebrew/gazebo8-8.1.1_5.high_sierra.bottle.tar.gz
==> Verifying gazebo8-8.1.1_5.high_sierra.bottle.tar.gz checksum
==> Pouring gazebo8-8.1.1_5.high_sierra.bottle.tar.gz
tar xzf /Users/user/Library/Caches/Homebrew/gazebo8-8.1.1_5.high_sierra.bottle.tar.gz
==> Finishing up
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gazebo gazebo
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gazebo-8.1.1 gazebo-8.1.1
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gz gz
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gz-8.1.1 gz-8.1.1
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gzclient gzclient
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gzclient-8.1.1 gzclient-8.1.1
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gzprop gzprop
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gzserver gzserver
ln -s ../Cellar/gazebo8/8.1.1_5/bin/gzserver-8.1.1 gzserver-8.1.1
ln -s ../Cellar/gazebo8/8.1.1_5/include/gazebo-8 gazebo-8
ln -s ../Cellar/gazebo8/8.1.1_5/share/gazebo gazebo
ln -s ../Cellar/gazebo8/8.1.1_5/share/gazebo-8 gazebo-8
ln -s ../../../Cellar/gazebo8/8.1.1_5/share/man/man1/gz.1.gz gz.1.gz
ln -s ../../Cellar/gazebo8/8.1.1_5/lib/cmake/gazebo gazebo
ln -s ../Cellar/gazebo8/8.1.1_5/lib/gazebo-8 gazebo-8
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo.8.1.1.dylib libgazebo.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo.8.dylib libgazebo.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo.dylib libgazebo.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_ccd.8.1.1.dylib libgazebo_ccd.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_ccd.8.dylib libgazebo_ccd.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_ccd.dylib libgazebo_ccd.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_client.8.1.1.dylib libgazebo_client.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_client.8.dylib libgazebo_client.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_client.dylib libgazebo_client.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_common.8.1.1.dylib libgazebo_common.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_common.8.dylib libgazebo_common.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_common.dylib libgazebo_common.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_gimpact.8.1.1.dylib libgazebo_gimpact.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_gimpact.8.dylib libgazebo_gimpact.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_gimpact.dylib libgazebo_gimpact.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_gui.8.1.1.dylib libgazebo_gui.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_gui.8.dylib libgazebo_gui.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_gui.dylib libgazebo_gui.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_math.8.1.1.dylib libgazebo_math.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_math.8.dylib libgazebo_math.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_math.dylib libgazebo_math.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_msgs.8.1.1.dylib libgazebo_msgs.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_msgs.8.dylib libgazebo_msgs.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_msgs.dylib libgazebo_msgs.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_ode.8.1.1.dylib libgazebo_ode.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_ode.8.dylib libgazebo_ode.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_ode.dylib libgazebo_ode.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_opcode.8.1.1.dylib libgazebo_opcode.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_opcode.8.dylib libgazebo_opcode.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_opcode.dylib libgazebo_opcode.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_opende_ou.8.1.1.dylib libgazebo_opende_ou.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_opende_ou.8.dylib libgazebo_opende_ou.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_opende_ou.dylib libgazebo_opende_ou.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_physics.8.1.1.dylib libgazebo_physics.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_physics.8.dylib libgazebo_physics.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_physics.dylib libgazebo_physics.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_rendering.8.1.1.dylib libgazebo_rendering.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_rendering.8.dylib libgazebo_rendering.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_rendering.dylib libgazebo_rendering.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_sensors.8.1.1.dylib libgazebo_sensors.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_sensors.8.dylib libgazebo_sensors.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_sensors.dylib libgazebo_sensors.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_test_fixture.a libgazebo_test_fixture.a
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_transport.8.1.1.dylib libgazebo_transport.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_transport.8.dylib libgazebo_transport.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_transport.dylib libgazebo_transport.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_util.8.1.1.dylib libgazebo_util.8.1.1.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_util.8.dylib libgazebo_util.8.dylib
ln -s ../Cellar/gazebo8/8.1.1_5/lib/libgazebo_util.dylib libgazebo_util.dylib
ln -s ../../Cellar/gazebo8/8.1.1_5/lib/pkgconfig/gazebo.pc gazebo.pc
ln -s ../../Cellar/gazebo8/8.1.1_5/lib/pkgconfig/gazebo_ode.pc gazebo_ode.pc
ln -s ../../Cellar/gazebo8/8.1.1_5/lib/pkgconfig/gazebo_transport.pc gazebo_transport.pc
/usr/bin/sandbox-exec -f /tmp/homebrew20171114-72820-wf4wyp.sb nice /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby -W0 -I /usr/local/Homebrew/Library/Homebrew -- /usr/local/Homebrew/Library/Homebrew/postinstall.rb /usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/gazebo8.rb --verbose
==> Summary
🍺  /usr/local/Cellar/gazebo8/8.1.1_5: 1,188 files, 122MB
:Firmware user$ brew gist-logs gazebo8
Error: No logs.
~~~

scpeters, 2017-11-14:
Thanks for your report. I opened #307  and #308 last night to build bottles but the internet went out at our office. I have been attempting to rebuild them this morning with slower internet but some of the jobs are failing via timeout. I'm working on it, but it will be easier if we can get internet access fixed here at our office.

scpeters, 2017-11-14:
I'm manually uploading a bottle right now at 12 kB/s

LorenzMeier, 2017-11-14:
Cool, thanks!

scpeters, 2017-11-14:
The elcapitan bottles aren't building, but you can download the high sierra gazebo8 bottles from the following link and manually install them if you need them now:

* https://build.osrfoundation.org/view/os_homebrew/job/generic-release-homebrew_bottle_builder-highsierra/47/

LorenzMeier, 2017-11-14:
Thanks! In terms of "needing": Our users need them, so the real relevance would be to have your official mirror updated. I guess that requires internet though. @tfoote must be a terrific day at the office 8).

scpeters, 2017-11-14:
For users, the temporary workaround is to backdate the `homebrew-core` repository to just before protobuf 3.5 was merged ( https://github.com/Homebrew/homebrew-core/commit/ac6eee86e506f0ffd04f7ff7dea6aa9f396ee6b7 ) and set the `HOMEBREW_NO_AUTO_UPDATE` environment variable in the shell they use for brew commands.

~~~
cd `brew --repo`/Library/Taps/homebrew/homebrew-core
git checkout ac6eee86e506f0ffd04f7ff7dea6aa9f396ee6b7
export HOMEBREW_NO_AUTO_UPDATE=1
brew install osrf/simulation/gazebo8
~~~

but we'll update the formula as soon as we can :)

scpeters, 2017-11-15:
Bottles are built!

closed 2017-11-15 by scpeters