osrf/homebrew-simulation #311: Cannot Install Gazebo8 on Mac OSX

chuckb, 2017-11-15:
I attempted to install gazebo8 on Mac OSX (10.12.5) via the instructions:
```
curl -ssL http://get.gazebosim.org | sh
```
I received a protobuf versioning error (version 3.5 on my machine, but one of the dependencies, libignition, needs version 3.2)

I then attempted to install the HEAD version via:
```
brew install gazebo8 --HEAD
```
That installation failed with a similar error.

Gist referenced here:
https://gist.github.com/590b945d6f10d8910835bd1fb4218dc9

scpeters, 2017-11-15:
You need to upgrade `ignition-msgs0`; it has a newer bottle that uses protobuf 3.5 as of #308:

~~~
brew up
brew upgrade `brew deps gazebo8`
brew install gazebo8
~~~

chuckb, 2017-11-15:
Per your instructions, here is the result:
```
charles-benedicts-macbook:~ chuck_benedict$ brew up
Updated 2 taps (caskroom/cask, homebrew/core).
==> New Formulae
kibana@5.6
==> Updated Formulae
arangodb                                   commandbox                                 kibana
charles-benedicts-macbook:~ chuck_benedict$ brew upgrade `brew deps gazebo8`
Error: boost 1.65.1 already installed
Error: bullet 2.87 already installed
Error: cmake 3.9.6 already installed
Error: osrf/simulation/cppzmq 0.0.0-brew already installed
Error: doxygen 1.8.13_1 already installed
Error: freeimage 3.17.0_1 already installed
Error: freetype 2.8.1 already installed
Error: gettext 0.19.8.1 already installed
Error: glib 2.54.2 already installed
Error: gts 0.7.6 already installed
Error: osrf/simulation/ignition-math3 3.2.0 already installed
Error: osrf/simulation/ignition-msgs0 0.7.0_5 already installed
Error: osrf/simulation/ignition-tools 0.0.0-20160209-381c0a968fa already installed
Error: osrf/simulation/ignition-transport3 3.0.1_8 already installed
Error: jasper 2.0.14 already installed
Error: jpeg 9b already installed
Error: libffi 3.2.1 already installed
Error: libpng 1.6.34 already installed
Error: libtar 1.2.20 already installed
Error: libtiff 4.0.8_5 already installed
Error: libzzip 0.13.67 already installed
Error: netpbm 10.73.17 already installed
Error: osrf/simulation/ogre1.9 1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4 already installed
Error: ossp-uuid 1.6.2_2 already installed
Error: pcre 8.41 already installed
Error: pkg-config 0.29.2 already installed
Error: protobuf 3.5.0 already installed
Error: protobuf-c 1.3.0_2 already installed
Error: qt 5.9.2 already installed
Error: qwt 6.1.3_4 already installed
Error: osrf/simulation/sdformat5 5.3.0 already installed
Error: osrf/simulation/simbody 3.5.4_2 already installed
Error: tbb 2018_U1 already installed
Error: tinyxml 2.6.2 already installed
Error: tinyxml2 5.0.1 already installed
Error: zeromq 4.2.2 already installed
charles-benedicts-macbook:~ chuck_benedict$ brew install gazebo8
==> Installing gazebo8 from osrf/simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo8-8.1.1_7.sierra.bottle.tar.gz
Already downloaded: /Users/chuck_benedict/Library/Caches/Homebrew/gazebo8-8.1.1_7.sierra.bottle.tar.gz
==> Pouring gazebo8-8.1.1_7.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/gazebo8/8.1.1_7: 1,190 files, 122.5MB
charles-benedicts-macbook:~ chuck_benedict$ gazebo
dyld: Library not loaded: /usr/local/opt/protobuf/lib/libprotobuf.12.dylib
  Referenced from: /usr/local/opt/ignition-msgs0/lib/libignition-msgs0.0.dylib
  Reason: image not found
Abort trap: 6
```

scpeters, 2017-11-15:
what does `brew doctor` say? how about `brew test ignition-msgs0`?

scpeters, 2017-11-15:
Also, `which gazebo` and `brew test gazebo8`

chuckb, 2017-11-15:
```
charles-benedicts-macbook:~ chuck_benedict$ brew test ignition-msgs0
Testing osrf/simulation/ignition-msgs0
==> pkg-config ignition-msgs0
==> /usr/bin/clang test.cpp -std=c++11 -I/usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0 -I/usr/local/Cellar/ign
Last 15 lines from /Users/chuck_benedict/Library/Logs/Homebrew/ignition-msgs0/test.02.clang:
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0/ignition/msgs/pose.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate this file with a newer version of protoc.
 ^
In file included from test.cpp:1:
In file included from /usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0/ignition/msgs.hh:3:
In file included from /usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:3:
In file included from /usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0/ignition/msgs/any.pb.h:34:
In file included from /usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0/ignition/msgs/pose.pb.h:31:
/usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0/ignition/msgs/vector3d.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
Error: osrf/simulation/ignition-msgs0: failed
Failed executing: /usr/bin/clang test.cpp -std=c++11 -I/usr/local/Cellar/ignition-msgs/0.7.0_1/include/ignition/msgs0 -I/usr/local/Cellar/ignition-math3/3.2.0/include/ignition/math3 -L/usr/local/Cellar/ignition-msgs0/0.7.0_5/lib -lignition-msgs0 -lc++ -o test
/usr/local/Homebrew/Library/Homebrew/formula.rb:1796:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `system'
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/ignition-msgs0.rb:49:in `block in <class:IgnitionMsgs0>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1640:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:554:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1639:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:834:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1638:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1632:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:91:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:106:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
```

chuckb, 2017-11-15:
```
charles-benedicts-macbook:~ chuck_benedict$ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry and just ignore them. Thanks!

Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and what additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew provided
script of the same name. We found the following "config" scripts:
  /opt/local/bin/ncurses5-config
  /opt/local/bin/ncursesw5-config
  /sw/bin/apt-config
  /sw/bin/croco-0.6-config
  /sw/bin/freetype-config
  /sw/bin/guile-1.8-config
  /sw/bin/libpng-config
  /sw/bin/libpng16-config
  /sw/bin/libwmf-config
  /sw/bin/ncurses5-config
  /sw/bin/ncursesw5-config
  /sw/bin/pcre-config
  /sw/bin/pkg-config
  /sw/bin/pykg-config
  /sw/bin/python2.7-config
  /sw/bin/xml2-config
  /sw/bin/xslt-config
  /usr/local/CrossPack-AVR/bin/libusb-config

Warning: Some frameworks can be picked up by CMake's build system and likely
cause the build to fail. To compile CMake, you may wish to move these
out of the way:
  /Library/Frameworks/libcurl.framework

Warning: You have MacPorts or Fink installed:
  /opt/local/bin/port, /sw/bin/fink

This can cause trouble. You don't have to uninstall them, but you may want to
temporarily move them out of the way, e.g.

  sudo mv /opt/local ~/macports

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/libruby.1.8.7.dylib

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/c-client.h
  /usr/local/include/dummy.h
  /usr/local/include/env.h
  /usr/local/include/env_unix.h
  /usr/local/include/fdstring.h
  /usr/local/include/flockcyg.h
  /usr/local/include/flocksim.h
  /usr/local/include/flstring.h
  /usr/local/include/fs.h
  /usr/local/include/ftl.h
  /usr/local/include/gdb/jit-reader.h
  /usr/local/include/imap4r1.h
[... 291 lines cut ...]

Warning: Your Xcode (8.3.2) is outdated.
Please update to Xcode 9.1 (or delete it).
Xcode can be updated from the App Store.

Warning: Some installed formula are missing dependencies.
You should `brew install` the missing dependencies:
  brew install adns gmp gnutls libtasn1 libunistring nettle npth openssl@1.1 p11-kit

Run `brew missing` for more details.

Warning: Some keg-only formula are linked into the Cellar.
Linking a keg-only formula, such as gettext, into the cellar with
`brew link <formula>` will cause other formulae to detect them during
the `./configure` step. This may cause problems when compiling those
other formulae.

Binaries provided by keg-only formulae may override system binaries
with other strange results.

You may wish to `brew unlink` these brews:
  dirmngr
  gpg-agent
  icu4c
```

chuckb, 2017-11-15:
```
charles-benedicts-macbook:~ chuck_benedict$ which gazebo
/usr/local/bin/gazebo
```

chuckb, 2017-11-15:
```
charles-benedicts-macbook:~ chuck_benedict$ brew test gazebo8
Testing osrf/simulation/gazebo8
==> /usr/local/Cellar/gazebo8/8.1.1_7/bin/gz sdf
Last 15 lines from /Users/chuck_benedict/Library/Logs/Homebrew/gazebo8/test.01.gz:
2017-11-15 11:28:13 -0800

/usr/local/Cellar/gazebo8/8.1.1_7/bin/gz
sdf

dyld: Library not loaded: /usr/local/opt/protobuf/lib/libprotobuf.12.dylib
  Referenced from: /usr/local/opt/ignition-msgs0/lib/libignition-msgs0.0.dylib
  Reason: image not found
Error: osrf/simulation/gazebo8: failed
Failed executing: /usr/local/Cellar/gazebo8/8.1.1_7/bin/gz sdf
/usr/local/Homebrew/Library/Homebrew/formula.rb:1796:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `system'
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/gazebo8.rb:76:in `block in <class:Gazebo8>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1640:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:554:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1639:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:834:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1638:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1632:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:91:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:106:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
charles-benedicts-macbook:~ chuck_benedict$ 
```

scpeters, 2017-11-16:
It looks like your `ignition-msgs` package is out of date, since I see version `0.7.0_1` in the error message for `brew test ignition-msgs0`. I renamed that formula in #290 so that may be preventing it from upgrading properly. If you can remove that package and reinstall it, that should fix the problem.

scpeters, 2017-11-16:
I forgot to mention that the latest version of ignition-msgs0 is `0.7.0_5`

chuckb, 2017-11-16:
Thanks for your help.  That worked.  Also, FYI, I had to uninstall bullet and re-install it.

```
charles-benedicts-macbook:~ chuck_benedict$ curl -ssL http://get.gazebosim.org | sh
=======================================================
GAZEBO INSTALLATION SCRIPT
=======================================================

This script is installing the latest stable version of
Gazebo Simulator available from your package manager

package-id: org.macosforge.xquartz.pkg
version: 2.7.112
volume: /
location: 
install-time: 1510711936
Already up-to-date.
==> Installing gazebo8 from osrf/simulation
==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo8-8.1.1
Already downloaded: /Users/chuck_benedict/Library/Caches/Homebrew/gazebo8-8.1.1_7.sierra.bottle.tar.gz
==> Pouring gazebo8-8.1.1_7.sierra.bottle.tar.gz
🍺  /usr/local/Cellar/gazebo8/8.1.1_7: 1,190 files, 122.5MB
Testing osrf/simulation/gazebo8
==> /usr/local/Cellar/gazebo8/8.1.1_7/bin/gz sdf
Last 15 lines from /Users/chuck_benedict/Library/Logs/Homebrew/gazebo8/test.01.gz:
2017-11-15 16:21:48 -0800

/usr/local/Cellar/gazebo8/8.1.1_7/bin/gz
sdf

dyld: Symbol not found: __ZN18btSliderConstraint8setParamEifi
  Referenced from: /usr/local/Cellar/gazebo8/8.1.1_7/lib/libgazebo_physics.8.dylib
  Expected in: /usr/local/opt/bullet/lib/libBulletDynamics.2.87.dylib
 in /usr/local/Cellar/gazebo8/8.1.1_7/lib/libgazebo_physics.8.dylib
Error: osrf/simulation/gazebo8: failed
Failed executing: /usr/local/Cellar/gazebo8/8.1.1_7/bin/gz sdf
/usr/local/Homebrew/Library/Homebrew/formula.rb:1796:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `system'
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/gazebo8.rb:76:in `block in <class:Gazebo8>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1640:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:554:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1639:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:834:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1638:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1632:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:91:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:106:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
```

scpeters, 2017-11-16:
Glad you got it working!

closed 2017-11-16 by scpeters