osrf/homebrew-simulation #323: Cannot install Gazebo8 Mac OSX.

alex-born, 2017-12-03:
I tried to install gazebo8 on OSX (10.12.6) via the install script.

**Error from install script:**
```
/usr/local/Cellar/gazebo8/8.1.1_7/bin/gz
sdf

[libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: ignition/msgs/altimeter.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1394] CHECK failed: generated_database_->Add(encoded_file_descriptor, size):
libc++abi.dylib: terminating with uncaught exception of type google::protobuf::FatalException: CHECK failed: generated_database_->Add(encoded_file_descriptor, size):
Error: osrf/simulation/gazebo8: failed
Failed executing: /usr/local/Cellar/gazebo8/8.1.1_7/bin/gz sdf
/usr/local/Homebrew/Library/Homebrew/formula.rb:1796:in `block in system'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `open'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `system'
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/gazebo8.rb:76:in `block in <class:Gazebo8>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1640:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:554:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1639:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:834:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1638:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1632:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:91:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:106:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
```
**I tried brew "upgrade `brew deps gazebo8`"**
```
Error: boost 1.65.1 already installed
Error: bullet 2.87 already installed
Error: cmake 3.10.0 already installed
Error: osrf/simulation/cppzmq 0.0.0-brew already installed
Error: doxygen 1.8.13_1 already installed
Error: freeimage 3.17.0_1 already installed
Error: freetype 2.8.1 already installed
Error: gettext 0.19.8.1 already installed
Error: glib 2.54.2 already installed
Error: gts 0.7.6 already installed
Error: osrf/simulation/ignition-math3 3.3.0 already installed
Error: osrf/simulation/ignition-msgs0 0.7.0_5 already installed
Error: osrf/simulation/ignition-tools 0.0.0-20160209-381c0a968fa already installed
Error: osrf/simulation/ignition-transport3 3.0.1_8 already installed
Error: jasper 2.0.14 already installed
Error: jpeg 9b already installed
Error: libffi 3.2.1 already installed
Error: libpng 1.6.34 already installed
Error: libtar 1.2.20 already installed
Error: libtiff 4.0.8_5 already installed
Error: libzzip 0.13.67 already installed
Error: netpbm 10.73.17 already installed
Error: osrf/simulation/ogre1.9 1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4 already installed
Error: ossp-uuid 1.6.2_2 already installed
Error: pcre 8.41 already installed
Error: pkg-config 0.29.2 already installed
Error: protobuf 3.5.0 already installed
Error: protobuf-c 1.3.0_2 already installed
Error: qt 5.9.3 already installed
Error: qwt 6.1.3_4 already installed
Error: osrf/simulation/sdformat5 5.3.0 already installed
Error: osrf/simulation/simbody 3.5.4_2 already installed
Error: tbb 2018_U1 already installed
[... 135 lines cut ...]
/usr/local/Homebrew/Library/Homebrew/formula.rb:1734:in `system'
/usr/local/Homebrew/Library/Taps/osrf/homebrew-simulation/gazebo8.rb:76:in `block in <class:Gazebo8>'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1640:in `block (3 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/utils.rb:554:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1639:in `block (2 levels) in run_test'
/usr/local/Homebrew/Library/Homebrew/formula.rb:834:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1638:in `block in run_test'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:14:in `block in mktemp'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `block in run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `chdir'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:74:in `run'
/usr/local/Homebrew/Library/Homebrew/extend/fileutils.rb:13:in `mktemp'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1632:in `run_test'
/usr/local/Homebrew/Library/Homebrew/test.rb:28:in `block in <main>'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:91:in `block in timeout'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `block in catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:33:in `catch'
/usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.3.3/lib/ruby/2.3.0/timeout.rb:106:in `timeout'
/usr/local/Homebrew/Library/Homebrew/test.rb:27:in `<main>'
```
**"brew upgrade protobuf":**
Error: protobuf 3.5.0 already installed
**" brew upgrade ignition-msgs0":**
Error: osrf/simulation/ignition-msgs0 0.7.0_5 already installed

scpeters, 2017-12-08:
I can't reproduce the problem. I just installed the `gazebo8` bottle on our 10.12 machine and it was able to run the test successfully.

I have seen the "File already exists in database" protobuf errors before, and I don't remember exactly what happened, but I think my environment and/or environment variables were at fault.

scpeters, 2018-01-23:
I just remembered one way to reproduce the "File already exists in database" protobuf error. If you compile gazebo but have a different set of gazebo libraries earlier in your `DYLD_LIBRARY_PATH`, then you might see this error. The key is making sure the correct libraries are loaded.

For example, I just noticed this error after `brew install gazebo8`, then compiling gazebo from source without installing, and running a test. After `brew unlink gazebo8`, I was able to run the test successfully.

scpeters, 2018-01-25:
I'm going to close this for now; please re-open if the issue persists.

closed 2018-01-25 by scpeters