osrf/homebrew-simulation #334: LibTinyXML2 Version Error

Animanny, 2017-12-12:
Tried running 2 command install. Install worked but now I receive this error. I checked and found that the version of libtinyxml2 is 2.6 but I don't know what I have to do to solve it :/

~~~
Anis-MacBook-Pro:~ Ani$ gazebo
dyld: Library not loaded: /usr/local/opt/tinyxml2/lib/libtinyxml2.5.dylib
  Referenced from: /usr/local/bin/gazebo
  Reason: image not found
Abort trap: 6

~~~

scpeters, 2017-12-12:
Thanks for the report. A new version of tinyxml2 was released, which broke our bottles. I'm building new ones as we speak for gazebo8 in #330, and gazebo7 was already fixed in #331

scpeters, 2017-12-12:
should be fixed by new bottles in #330 and #331

closed 2017-12-12 by scpeters