osrf/homebrew-simulation #342: Should all version 0 formulas end with 0?

chapulina, 2017-12-26:
I see some version 0 formulas end with zero, i.e.:

~~~
ignition-cmake0.rb
ignition-common0.rb
ignition-fuel-tools0.rb
ignition-gui0.rb
ignition-msgs0.rb
~~~

But other don't, such as:

~~~
ignition-rendering.rb
ignition-rndf.rb
ignition-sensors.rb
ignition-tools.rb
~~~

I think it would make sense to be consistent.

j-rivero, 2017-12-26:
Probably @scpeters has more information about this. There is no big problem at this point for keeping both but I agree that we should try to standardize the naming. I would prefer the formulas without the 0 but I have no strong preference.

scpeters, 2017-12-26:
As [documented here](https://docs.brew.sh/Versions.html), homebrew originally named versioned formulae with the major number at the end of the formula name, like `gazebo7`, but they now use a different pattern: `gazebo@7`. I try to remember to use [aliases](https://github.com/osrf/homebrew-simulation/blob/master/Aliases/gazebo%407) to make sure both names are available. I recently started pointing the unversioned name to the most recent stable release for some formulae, but it's not universal yet (ie. [gazebo -> gazebo8](https://github.com/osrf/homebrew-simulation/blob/master/Aliases/gazebo)).

As you pointed out, some of the formulae are still named without a major version, but I've been trying to add them, especially before we build bottles for them, since renaming a formula requires a new bottle build.

tl;dr yes let's put major versions in the formula names

scpeters, 2020-09-22:
I don't think we'll change names of existing formulae, but we can follow this pattern for new formulae

closed 2020-09-22 by scpeters