osrf/homebrew-simulation #383: Consider adding Xacro for model preprocessing

starthal, 2018-02-22:
Would it be out of scope to add a [xacro](https://github.com/ros/xacro) formula to this tap? We would like to use xacro independently of ROS to generate SDF robot models on macOS.

scpeters, 2018-02-22:
xacro depends on catkin, so we'd have to package that first; I'd have to talk to @wjwwood to get advice on how to accomplish this

an alternative is to use embedded python or embedded ruby to make model templates (see examples in [this tutorial](http://gazebosim.org/tutorials?tut=kinematic_loop&branch=kinematic_loop))

wjwwood, 2018-02-22:
Looks like the patches for ROS 2 support considered making it a pure python package, which would be easier to package and/or you could just use pip to install it:

https://github.com/ros/xacro/pull/154#issuecomment-367835306

Removing catkin is also possible, and perhaps easy depending on how `xacro` uses it. I recommend opening an issue on the xacro repo to discuss it, @codebot actually maintains it with help from some community members. I don't have much to do with it these days.

I don't think packaging catkin in homebrew just to package xacro would be a problem, but it also wouldn't be my first approach.

starthal, 2018-04-09:
Looks like this is in progress on that PR. Thanks for the info.

closed 2018-04-09 by starthal